wolfSSL/wolfssl #2380: wolfSSL and ARDUINO IDE using TLS 1.3?

torntrousers, 2019-07-27:
Helo. I've followed your [Arduino IDE README](https://github.com/wolfSSL/wolfssl/tree/master/IDE/ARDUINO) and managed to get an ESP8266 client talking to an ESP32 server using TLS 1.2 with the provided wolfssl_client / wolfssl_server sketches.

What do I need to change to be able to use TLS 1.3?

Changing [this line](https://github.com/wolfSSL/wolfssl/blob/master/IDE/ARDUINO/sketches/wolfssl_client/wolfssl_client.ino#L44) to have wolfTLSv1_3_client_method gives a compile error: 'wolfTLSv1_3_client_method' was not declared in this scope

Googling It sounds like I need to set the flag --enable-tls13 somewhere but I can't find how or where. 

Any hints?

I'm using wolfSSL Release 4.1.0 (07/22/2019)

kaleb-himes, 2019-07-31:
@torntrousers,

Can you try adding these to the wolfSSL/user_settings.h and let us know if this gets you going with TLS 1.3?

```
#define WOLFSSL_TLS13
#define HAVE_TLS_EXTENSIONS
#define HAVE_SUPPORTED_CURVES
#define HAVE_ECC
#define HAVE_HKDF
#define HAVE_FFDHE_8192 // or one of the other supported FFDHE sizes [2048, 3072, 4096, 6144, 8192]
#define WC_RSA_PSS
```

- KH

torntrousers, 2019-07-31:
Trying with those settings gives some compile errors:
```
C:\Arduino\ESP8266\190705\arduino-1.8.9\libraries\wolfSSL\tls.c: In function 'TLSX_PopulateSupportedGroups':

C:\Arduino\ESP8266\190705\arduino-1.8.9\libraries\wolfSSL\tls.c:9447:39: error: 'Options' has no member named 'minDhKeySz'

             if (2048/8 >= ssl->options.minDhKeySz &&

                                       ^

C:\Arduino\ESP8266\190705\arduino-1.8.9\libraries\wolfSSL\tls.c:9448:67: error: 'Options' has no member named 'maxDhKeySz'

                                             2048/8 <= ssl->options.maxDhKeySz) {

                                                                   ^

exit status 1
C:\Arduino\ESP8266\190705\arduino-1.8.9\libraries\wolfSSL\tls13.c: In function 'EncryptTls13':

C:\Arduino\ESP8266\190705\arduino-1.8.9\libraries\wolfSSL\tls13.c:1677:63: error: 'Keys' has no member named 'aead_enc_imp_IV'

             BuildTls13Nonce(ssl, ssl->encrypt.nonce, ssl->keys.aead_enc_imp_IV,

                                                               ^

C:\Arduino\ESP8266\190705\arduino-1.8.9\libraries\wolfSSL\tls13.c: In function 'DecryptTls13':

C:\Arduino\ESP8266\190705\arduino-1.8.9\libraries\wolfSSL\tls13.c:1924:63: error: 'Keys' has no member named 'aead_dec_imp_IV'

             BuildTls13Nonce(ssl, ssl->decrypt.nonce, ssl->keys.aead_dec_imp_IV,

                                                               ^

Error compiling for board NodeMCU 1.0 (ESP-12E Module).
```

kaleb-himes, 2019-08-06:
This looks related to some of the assembly optimizations. Are you turning on any assembly support in your settings? If not can you try adding the WOLFSSL_NO_ASM and TFM_NO_ASM settings to user_settings.h and let us know if the issues persist?

- K

dgarske, 2019-08-06:
Hi @torntrousers,

Those errors would happen if `NO_DH` was defined and `HAVE_AEAD` was not defined. In your application make sure you follow the instruction for setting `WOLFSSL_USER_SETTINGS` and using user_settings.h here: https://github.com/wolfSSL/wolfssl/tree/master/IDE/ARDUINO

Its important the wolfssl.h header generated has `WOLFSSL_USER_SETTINGS` defined in it and you are including `#include <wolfssl/wolfcrypt/settings.h>`.

Thanks,
David Garske, wolfSSL

torntrousers, 2019-08-07:
Thanks to your colleague I have this working now. The trick with the Arduino environment is to remove the define for HAVE_FFDHE.

If it helps anyone, this is what my user_settings.h looks like:
```
/* Generated wolfSSL user_settings.h file for Arduino */
#ifndef ARDUINO_USER_SETTINGS_H
#define ARDUINO_USER_SETTINGS_H

/* Platform */
#define WOLFSSL_ARDUINO

/* Math library (remove this to use normal math)*/
#define USE_FAST_MATH
#define TFM_NO_ASM

/* RNG DEFAULT !!FOR TESTING ONLY!! */
/* comment out the error below to get started w/ bad entropy source
 * This will need fixed before distribution but is OK to test with */
/* #error "needs solved, see: https://www.wolfssl.com/docs/porting-guide/" */
#define WOLFSSL_GENSEED_FORTEST

/* The remaining added by me: */

#define NO_ASN_TIME
#define SINGLE_THREADED 

#define XTIME fnSecondsSinceEpoch
#define XGMTIME
#define ALT_ECC_SIZE

#define WOLFSSL_TLS13
#define HAVE_TLS_EXTENSIONS
#define HAVE_SUPPORTED_CURVES
#define HAVE_ECC
#define HAVE_HKDF
#define WC_RSA_PSS
#define NO_DH 

#define HAVE_AESGCM

#endif /* ARDUINO_USER_SETTINGS_H */
```

closed 2019-08-07 by torntrousers

oaugustocezar, 2021-05-26:
Have anyone an example of TLS 1.3 on esp8266 ? 
I'm trying, but my IO Callback functions look like no work

elms, 2021-05-27:
@oaugustocezar Can you open a new issue with details on the settings you are using and what you are seeing? Feel free to mention this issue to relate them

dgarske, 2021-06-01:
> Have anyone an example of TLS 1.3 on esp8266 ?
> I'm trying, but my IO Callback functions look like no work

Have you seen the IO callback examples in our wolfssl-examples repository here?
https://github.com/wolfSSL/wolfssl-examples/blob/master/tls/client-tls-callback.c

The most common issue is the IO read callback first asks for 5 bytes then the remainder of that TLS packet. The 5 bytes is the TLS header, which indicates the size of the TLS packet. Some modems require receiving a full TCP frame, which will require you to cache the remainder of the packet till the next IO callback. This utasker example does this (for example): https://github.com/wolfSSL/wolfssl-examples/blob/master/utasker/wolfSSLClientTask.c#L210

Thanks,
David Garske, wolfSSL