wolfSSL/wolfssl #2392: can't build wolfssl with --enable-devcrypto --enable-aesccm

cotequeiroz, 2019-08-02:
If both options are enabled, build fails

```
$ ./configure --enable-aesccm --enable-devcrypto
```
```
---
Configuration summary for wolfssl version 4.1.0

   * Installation prefix:        /usr/local
   * System type:                pc-linux-gnu
   * Host CPU:                   x86_64
   * C Compiler:                 gcc
   * C Flags:                     -Werror -Wno-pragmas -Wall -Wno-strict-aliasing -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmaybe-uninitialized -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wnormalized=id -Woverride-init -Wpointer-arith -Wpointer-sign -Wredundant-decls -Wshadow -Wsign-compare -Wstrict-overflow=1 -Wswitch-enum -Wundef -Wunused -Wunused-result -Wunused-variable -Wwrite-strings -fwrapv
   * C++ Compiler:
   * C++ Flags:
   * CPP Flags:                   -fvisibility=hidden
   * CCAS Flags:
   * LIB Flags:                   -pie -z relro -z now -Werror
   * Debug enabled:              no
   * Coverage enabled:
   * Warnings as failure:        yes
   * make -j:                    5
   * VCS checkout:               yes

   Features
   * Single threaded:            no
   * Filesystem:                 yes
   * OpenSSH Build:              no
   * OpenSSL Extra API:          no
   * OpenSSL Coexist:            no
   * Old Names:                  yes
   * Max Strength Build:         no
   * Distro Build:               no
   * fastmath:                   yes
   * Assembly Allowed:           yes
   * sniffer:                    no
   * snifftest:                  no
   * ARC4:                       no
   * AES:                        yes
   * AES-NI:                     no
   * AES-CBC:                    yes
   * AES-GCM:                    yes
   * AES-CCM:                    yes
   * AES-CTR:                    no
   * DES3:                       no
   * IDEA:                       no
   * Camellia:                   no
   * NULL Cipher:                no
   * MD5:                        yes
   * RIPEMD:                     no
   * SHA:                        yes
   * SHA-224:                    no
   * SHA-384:                    yes
   * SHA-512:                    yes
   * SHA3:                       yes
   * BLAKE2:                     no
   * CMAC:                       no
   * keygen:                     no
   * certgen:                    no
   * certreq:                    no
   * certext:                    no
   * certgencache:               no
   * HC-128:                     no
   * RABBIT:                     no
   * CHACHA:                     yes
   * Hash DRBG:                  yes
   * PWDBASED:                   no
   * scrypt:                     no
   * wolfCrypt Only:             no
[... 99 lines cut ...]
make -j5  all-am
make[1]: Entering directory '/home/equeiroz/src/wolfssl'
/bin/sh ./libtool  --tag=CC   --mode=compile gcc -DHAVE_CONFIG_H -I.  -DBUILDING_WOLFSSL -fvisibility=hidden -fvisibility=hidden -DBUILDING_WOLFSSL -DHAVE_FFDHE_2048 -D_POSIX_THREADS  -DHAVE_THREAD_LS -DNDEBUG -pthread -DTFM_TIMING_RESISTANT -DECC_TIMING_RESISTANT -DWC_RSA_BLINDING -DHAVE_AESGCM -DHAVE_AESCCM -DWOLFSSL_DEVCRYPTO -DWOLFSSL_DEVCRYPTO_CBC -DWOLFSSL_DEVCRYPTO_AES -DWOLFSSL_DEVCRYPTO_HASH -DWOLFSSL_NO_HASH_RAW -DWOLFSSL_SHA512 -DWOLFSSL_SHA384 -DNO_DSA -DHAVE_ECC -DTFM_ECC256 -DECC_SHAMIR -DWOLFSSL_BASE64_ENCODE -DNO_RC4 -DNO_HC128 -DNO_RABBIT -DWOLFSSL_SHA3 -DHAVE_POLY1305 -DHAVE_ONE_TIME_AUTH -DHAVE_CHACHA -DHAVE_HASHDRBG -DHAVE_TLS_EXTENSIONS -DHAVE_SUPPORTED_CURVES -DHAVE_EXTENDED_MASTER -DNO_PSK -DNO_MD4 -DNO_PWDBASED -DWOLFSSL_DEVCRYPTO_HASH_KEEP -DUSE_FAST_MATH -DWOLFSSL_X86_64_BUILD -DWC_NO_ASYNC_THREADING -DNO_DES3 -Wall -Wno-unused -O2 -fomit-frame-pointer -DHAVE___UINT128_T=1 -Werror -Wno-pragmas -Wall -Wno-strict-aliasing -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmaybe-uninitialized -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wnormalized=id -Woverride-init -Wpointer-arith -Wpointer-sign -Wredundant-decls -Wshadow -Wsign-compare -Wstrict-overflow=1 -Wswitch-enum -Wundef -Wunused -Wunused-result -Wunused-variable -Wwrite-strings -fwrapv  -MT wolfcrypt/src/src_libwolfssl_la-aes.lo -MD -MP -MF wolfcrypt/src/.deps/src_libwolfssl_la-aes.Tpo -c -o wolfcrypt/src/src_libwolfssl_la-aes.lo `test -f 'wolfcrypt/src/aes.c' || echo './'`wolfcrypt/src/aes.c
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -DBUILDING_WOLFSSL -fvisibility=hidden -fvisibility=hidden -DBUILDING_WOLFSSL -DHAVE_FFDHE_2048 -D_POSIX_THREADS -DHAVE_THREAD_LS -DNDEBUG -pthread -DTFM_TIMING_RESISTANT -DECC_TIMING_RESISTANT -DWC_RSA_BLINDING -DHAVE_AESGCM -DHAVE_AESCCM -DWOLFSSL_DEVCRYPTO -DWOLFSSL_DEVCRYPTO_CBC -DWOLFSSL_DEVCRYPTO_AES -DWOLFSSL_DEVCRYPTO_HASH -DWOLFSSL_NO_HASH_RAW -DWOLFSSL_SHA512 -DWOLFSSL_SHA384 -DNO_DSA -DHAVE_ECC -DTFM_ECC256 -DECC_SHAMIR -DWOLFSSL_BASE64_ENCODE -DNO_RC4 -DNO_HC128 -DNO_RABBIT -DWOLFSSL_SHA3 -DHAVE_POLY1305 -DHAVE_ONE_TIME_AUTH -DHAVE_CHACHA -DHAVE_HASHDRBG -DHAVE_TLS_EXTENSIONS -DHAVE_SUPPORTED_CURVES -DHAVE_EXTENDED_MASTER -DNO_PSK -DNO_MD4 -DNO_PWDBASED -DWOLFSSL_DEVCRYPTO_HASH_KEEP -DUSE_FAST_MATH -DWOLFSSL_X86_64_BUILD -DWC_NO_ASYNC_THREADING -DNO_DES3 -Wall -Wno-unused -O2 -fomit-frame-pointer -DHAVE___UINT128_T=1 -Werror -Wno-pragmas -Wall -Wno-strict-aliasing -Wextra -Wunknown-pragmas --param=ssp-buffer-size=1 -Waddress -Warray-bounds -Wbad-function-cast -Wchar-subscripts -Wcomment -Wfloat-equal -Wformat-security -Wformat=2 -Wmaybe-uninitialized -Wmissing-field-initializers -Wmissing-noreturn -Wmissing-prototypes -Wnested-externs -Wnormalized=id -Woverride-init -Wpointer-arith -Wpointer-sign -Wredundant-decls -Wshadow -Wsign-compare -Wstrict-overflow=1 -Wswitch-enum -Wundef -Wunused -Wunused-result -Wunused-variable -Wwrite-strings -fwrapv -MT wolfcrypt/src/src_libwolfssl_la-aes.lo -MD -MP -MF wolfcrypt/src/.deps/src_libwolfssl_la-aes.Tpo -c wolfcrypt/src/aes.c  -fPIC -DPIC -o wolfcrypt/src/.libs/src_libwolfssl_la-aes.o
wolfcrypt/src/aes.c: In function ‘roll_x’:
wolfcrypt/src/aes.c:6378:9: error: implicit declaration of function ‘wc_AesEncrypt’; did you mean ‘wc_AesCcmEncrypt’? [-Werror=implicit-function-declaration]
         wc_AesEncrypt(aes, out, out);
         ^~~~~~~~~~~~~
         wc_AesCcmEncrypt
wolfcrypt/src/aes.c:6378:9: error: nested extern declaration of ‘wc_AesEncrypt’ [-Werror=nested-externs]
cc1: all warnings being treated as errors
make[1]: *** [Makefile:4103: wolfcrypt/src/src_libwolfssl_la-aes.lo] Error 1
make[1]: Leaving directory '/home/equeiroz/src/wolfssl'
make: *** [Makefile:2882: all] Error 2
```
If I add `-Wno-error=nested-externs`, then it fails to link, as that function is not compiled anywhere:
```
libtool: link: gcc -DHAVE_FFDHE_2048 -D_POSIX_THREADS -DHAVE_THREAD_LS -DNDEBUG -pthread -DTFM_TIMING_RESISTANT -DECC_TIMING_RESISTANT -DWC_RSA_BLINDING -DHAVE_AESGCM -DHAVE_AESCCM -DWOLFSSL_DEVCRYPTO -DWOLFSSL_DEVCRYPTO_CBC -DWOLFSSL_DEVCRYPTO_AES -DWOLFSSL_DEVCRYPTO_HASH -DWOLFSSL_NO_HASH_RAW -DWOLFSSL_SHA512 -DWOLFSSL_SHA384 -DNO_DSA -DHAVE_ECC -DTFM_ECC256 -DECC_SHAMIR -DWOLFSSL_BASE64_ENCODE -DNO_RC4 -DNO_HC128 -DNO_RABBIT -DWOLFSSL_SHA3 -DHAVE_POLY1305 -DHAVE_ONE_TIME_AUTH -DHAVE_CHACHA -DHAVE_HASHDRBG -DHAVE_TLS_EXTENSIONS -DHAVE_SUPPORTED_CURVES -DHAVE_EXTENDED_MASTER -DNO_PSK -DNO_MD4 -DNO_PWDBASED -DWOLFSSL_DEVCRYPTO_HASH_KEEP -DUSE_FAST_MATH -DWOLFSSL_X86_64_BUILD -DWC_NO_ASYNC_THREADING -DNO_DES3 -Wall -Wno-unused -O2 -fomit-frame-pointer -DHAVE___UINT128_T=1 -Wno-error=nested-externs -o examples/client/.libs/client examples/client/client.o  src/.libs/libwolfssl.so -lm -pthread
/usr/lib/gcc/x86_64-pc-linux-gnu/8.3.0/../../../../x86_64-pc-linux-gnu/bin/ld: src/.libs/libwolfssl.so: undefined reference to `wc_AesEncrypt'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:3477: examples/client/client] Error 1
make[1]: Leaving directory '/home/equeiroz/src/wolfssl'
make: *** [Makefile:2882: all] Error 2
```
The same thing happens if you use `--enable-afalg` instead of `--enable-devcrypto`.  As for the `--enable-devcrypto` parameters, using `--enable-devcrypto=hash` works; `aes`, `cbc`, and `all` fail.

JacobBarthelmeh, 2019-08-16:
Hi @cotequeiroz 
A change was merged for this in PR https://github.com/wolfSSL/wolfssl/pull/2403
Can you tell us some about the project use case with devcrypto?
Regards,
Jacob

cotequeiroz, 2019-08-16:
Thanks for the reply.
I use devcrypto with openwrt.  There are some routers with hw-crypto capabilities supported.  I personally use a Linksys WRT3200ACM, but--in my particular case--hw-crypto gets little use in userspace because none of the AEAD ciphers are currently supported.

I've tested your change, using openwrt with uhttpd running under wolfssl, and fetching a file with curl, forcing the use of TLS 1.2 and `AES128-SHA`.  In order to verify the use of crypto hardware, I check the process open fds and the `/proc/interrupts` counts:
```
Every 2.0s: egrep crypto /proc/interrupts; ls -l /proc/`pidof uhttpd`/fd     Fri Aug 16 16:47:47 2019

 39:     232414          0     GIC-0  51 Level     f1090000.crypto
 40:          0          0     GIC-0  52 Level     f1090000.crypto
lr-x------    1 root     root            64 Aug 16 16:43 0 -> /dev/null
l-wx------    1 root     root            64 Aug 16 16:43 1 -> /dev/null
lrwx------    1 root     root            64 Aug 16 16:44 10 -> socket:[226266]
lrwx------    1 root     root            64 Aug 16 16:44 11 -> /dev/crypto
lr-x------    1 root     root            64 Aug 16 16:44 12 -> /www/image6.bin
lrwx------    1 root     root            64 Aug 16 16:44 13 -> /dev/crypto
lrwx------    1 root     root            64 Aug 16 16:44 14 -> /dev/crypto
l-wx------    1 root     root            64 Aug 16 16:43 2 -> pipe:[73718]
lrwx------    1 root     root            64 Aug 16 16:43 3 -> socket:[73723]
lrwx------    1 root     root            64 Aug 16 16:43 4 -> socket:[73724]
lrwx------    1 root     root            64 Aug 16 16:43 5 -> socket:[73725]
lrwx------    1 root     root            64 Aug 16 16:43 6 -> socket:[73726]
lrwx------    1 root     root            64 Aug 16 16:43 7 -> anon_inode:[eventpoll]
lr-x------    1 root     root            64 Aug 16 16:43 8 -> pipe:[73727]
l-wx------    1 root     root            64 Aug 16 16:43 9 -> pipe:[73727]
```

It works with `--enable-devcrypto=yes`, and `--enable-devcrypto=aes`, but not with `--enable-devcrypto=cbc`.  The latter builds if the option is selected all by itself, but it fails to build when combined with `--enable-wpas` (I singled that one out, but it fails with other options as well).  Here's the error message:
```
$ ./configure --enable-aesccm --enable-devcrypto=cbc --enable-wpas; make
```
```
  CC       wolfcrypt/src/src_libwolfssl_la-rsa.lo
  CC       wolfcrypt/src/src_libwolfssl_la-aes.lo
  CC       wolfcrypt/src/src_libwolfssl_la-cmac.lo
  CC       wolfcrypt/src/src_libwolfssl_la-des3.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha.lo
wolfcrypt/src/aes.c: In function ‘wc_AesDecryptDirect’:
wolfcrypt/src/aes.c:2442:13: error: implicit declaration of function ‘wc_AesDecrypt’; did you mean ‘wc_AesEncrypt’? [-Werror=implicit-function-declaration]
             wc_AesDecrypt(aes, in, out);
             ^~~~~~~~~~~~~
             wc_AesEncrypt
wolfcrypt/src/aes.c:2442:13: error: nested extern declaration of ‘wc_AesDecrypt’ [-Werror=nested-externs]
At top level:
wolfcrypt/src/aes.c:1326:19: error: ‘Td4’ defined but not used [-Werror=unused-const-variable=]
 static const byte Td4[256] =
                   ^~~
```
Even when successfully built, I can't see the "persistent" `/dev/crypto` fd, and curl fails to connect using `AES128-CBC`, but it works with `ECDHE-RSA-AES128-SHA` (it's probably some hardening option that `--enable-wpas` disables, isn't it?).  Notice that unlike previous output, this has only two open `/dev/crypto` fds, and when the connection is closed, no `/dev/crypto` fd is left open.  With the other options, it leaves one `/dev/crytpo` fd open (fd 11 in the previous example).  I'm not sure this is relevant or not.

```
Every 2.0s: egrep crypto /proc/interrupts; ls -l /proc/`pidof uhttpd`/fd                                                          aptv: Fri Aug 16 17:34:03 2019

 39:     384799          0     GIC-0  51 Level     f1090000.crypto
 40:          0          0     GIC-0  52 Level     f1090000.crypto
lr-x------    1 root     root            64 Aug 16 17:26 0 -> /dev/null
l-wx------    1 root     root            64 Aug 16 17:26 1 -> /dev/null
lrwx------    1 root     root            64 Aug 16 17:26 10 -> socket:[402790]
lrwx------    1 root     root            64 Aug 16 17:27 11 -> /dev/crypto
lrwx------    1 root     root            64 Aug 16 17:33 12 -> /dev/crypto
lr-x------    1 root     root            64 Aug 16 17:33 13 -> /www/image6.bin
l-wx------    1 root     root            64 Aug 16 17:26 2 -> pipe:[399412]
lrwx------    1 root     root            64 Aug 16 17:26 3 -> socket:[399417]
lrwx------    1 root     root            64 Aug 16 17:26 4 -> socket:[399418]
lrwx------    1 root     root            64 Aug 16 17:26 5 -> socket:[399419]
lrwx------    1 root     root            64 Aug 16 17:26 6 -> socket:[399420]
lrwx------    1 root     root            64 Aug 16 17:26 7 -> anon_inode:[eventpoll]
lr-x------    1 root     root            64 Aug 16 17:26 8 -> pipe:[399421]
l-wx------    1 root     root            64 Aug 16 17:26 9 -> pipe:[399421]
```

[... 35 lines cut ...]
```
Every 2.0s: egrep crypto /proc/interrupts; ls -l /proc/`pidof uhttpd`/fd     Fri Aug 16 16:52:23 2019

 39:     287495          0     GIC-0  51 Level     f1090000.crypto
 40:          0          0     GIC-0  52 Level     f1090000.crypto
lr-x------    1 root     root            64 Aug 16 16:51 0 -> /dev/null
l-wx------    1 root     root            64 Aug 16 16:51 1 -> /dev/null
[... cut ...]
While at it, please review the two comments in `configure.ac` indicating that either [`--devcrypto=aes`](https://github.com/wolfSSL/wolfssl/blob/dea4f2fb1aea75a58197118720c57347f15a304f/configure.ac#L1109) and [`--devcrypto=cbc`](https://github.com/wolfSSL/wolfssl/blob/dea4f2fb1aea75a58197118720c57347f15a304f/configure.ac#L1121) will `#enable only AES-CBC algorithm support`.  I'm assuming that `--devcrypto=aes` enables all AES modes, while `devcrypto=cbc` is the one that enables only `AES-CBC`.

cotequeiroz, 2019-09-06:
I've tested 4.1.0 with #2403 and #2442 applied, using curl/uhttpd, using builds with: (1) afalg, (2) devcrypto=full, (3) devcrypto=aes, & (4) devcrypto=cbc, and they all appear to work fine.  So I'm closing this.  Thanks for the fix.

closed 2019-09-06 by cotequeiroz