wolfSSL/wolfssl #2398: SRTP AES_CM_128 is different from FFMPEG version

maximkulkin, 2019-08-08:
Hi. I'm trying to use WolfSSL to implement SRTP AES_CM_128 encryption profile. First step in there is to derive various encryption keys from a master key (see [this](https://github.com/FFmpeg/FFmpeg/blob/master/libavformat/srtp.c#L100-L102)). My version looks something like the one below. Basically, it initialized AES with master key, then takes 14 byte salt, appends two zero bytes and does AES encrypt for the result and stores it.

```c
static void srtp_encrypt_counter(Aes *aes, uint8_t *iv, uint8_t *outbuf, int outlen) {
    int i, j, outpos;
    for (i = 0, outpos = 0; outpos < outlen; i++) {
        uint8_t keystream[16];
        iv[15] = i & 0xff;
        iv[14] = (i >> 8) & 0xff;

        wc_AesEncrypt(aes, iv, keystream);
        for (j = 0; j < 16 && outpos < outlen; j++, outpos++)
            outbuf[outpos] ^= keystream[j];
    }
}

static void srtp_derive_key(Aes *aes, const uint8_t *salt, int label, uint8_t *out, int outlen) {
    uint8_t input[16] = { 0 };
    memcpy(input, salt, 14);
    input[14 - 7] ^= label;
    memset(out, 0, outlen);
    srtp_encrypt_counter(aes, input, out, outlen);
}

void init() {
    // skip

    Aes *aes = (Aes*) malloc(sizeof(Aes));
    wc_AesInit(aes, NULL, INVALID_DEVID);
    wc_AesSetKey(
        aes,
        srtp_master_key, 16,
        NULL, AES_ENCRYPTION
    );

    srtp_derive_key(aes, srtp_master_salt, 0x00, rtp_key, sizeof(rtp_key));
}
```
But result is different from FFMPEG version and encrypted payloads can not be decrypted by SRTP player (while FFMPEG version works fine). I've looked at algorithms that are used to initialize round keys in [WolfSSL](https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/aes.c#L2119-L2136) and [FFMPEG AES](https://github.com/FFmpeg/FFmpeg/blob/master/libavutil/aes.c#L236-L251) and they seem to be quite different. What am I missing?

dgarske, 2019-08-09:
Hi @maximkulkin,

Have you tried using `wc_AesCbcEncrypt` instead of `wc_AesEncrypt`?

For reference the FFmpeg AES code is here:
https://github.com/FFmpeg/FFmpeg/blob/master/libavutil/aes.c

If that doesn't help let us know.

Thanks,
David Garske, wolfSSL

maximkulkin, 2019-08-09:
What's the benefit of using wc_AesCbcEncrypt? From source code I see that it internally calls wc_AesEncrypt, as far as I understand the whole point of Cbc version is how you dance around wc_AesEncrypt. In my case it's only a single call to wc_AesEncrypt and it already different. The code I showed above can be simplified into:

```c
Aes *aes = (Aes*) malloc(sizeof(Aes));
wc_AesInit(aes, NULL, INVALID_DEVID);
wc_AesSetKey(
    aes,
    srtp_master_key, 16,
    NULL, AES_ENCRYPTION
);

uint8_t input[16] = { 0 };
memcpy(input, srtp_master_salt, 14);
wc_AesEncrypt(aes, input, rtp_key);
```

Yes, I'm very aware where FFMPEG implementation is, I was reading it through very thoroughly. I was trying to figure out why they are different but failed so far.

maximkulkin, 2019-08-09:
The algorithms actually look similar, although one thing I do not understand. [This thing](https://github.com/FFmpeg/FFmpeg/blob/master/libavutil/aes.c#L262-L263). Why FFMPEG version does reverse round keys for encryption (while WolfSSL does not do that) ?

dgarske, 2019-08-09:
Hi @maximkulkin,

To me it doesn't look like the FFPEG AES is fully standard. I see what you mean about the init doing something different for encrypt than we do. We've taken this aes.c module through multiple FIPS certifications and NIST testing, so I do not believe its an error in our implementation. Instead its more likely FFMPEG is using a slight variation of AES. Based on their documentation it looks like if an IV is provided CBC mode is used otherwise ECB.

What is the processor you are running on? Have you tried changing the byte order (endianness)? By default its little endian, to switch to big define `BIG_ENDIAN_ORDER`.

Have you confirmed the `srtp_master_key` is the same? The FFMPEG code uses a Base64 decode on the input buffer.

If you end up making changes to aes.c and would like to make a contribution that we be fine with a signed contributor agreement. Make sure the feature is enabled using a build-macro (not enabled by default).

Thanks,
David Garske, wolfSSL

maximkulkin, 2019-08-09:
Well, regarding FFMPEG AES not being standard compliant: I would take that but the problem is that on the other end is iOS running video decoding and understanding FFMPEG's coding. I really doubt Apple uses broken AES implementation.

> Have you confirmed the srtp_master_key is the same? The FFMPEG code uses a Base64 decode on the input buffer.

Yes, I am certain. To be 100% sure I actually running modified FFMPEG and dumping bytes right where they are passed inside AES (and all other required places).

dgarske, 2020-01-23:
Hi @maximkulkin ,

I wanted to check in and see how things are going. Looking back at this issue and noticed the title. Perhaps you were looking for AES CCM not CBC? Have you tried the CCM API's?
See https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/aes.c#L6506

Thanks,
David Garske, wolfSSL

maximkulkin, 2020-01-23:
Thanks for checking, David.
I was able to figure that one out. Turns out I needed to use raw AES to implement CM mode.

closed 2020-01-23 by maximkulkin

dgarske, 2020-01-23:
Thanks excellent! Did you end up pushing any FFMPEG work upstream or have a fork/branch to review?

maximkulkin, 2020-01-23:
No, this was for my [ESP32 CAMERA](https://github.com/maximkulkin/esp32-homekit-camera) project, FFMPEG was used as a reference implementation.