wolfSSL/wolfssl #2406: Signature verification always succeeds on a provided PKCS7 DER encoded detached signature with signature attributes

csenanda, 2019-08-14:
Hello,
Signature verification using wc_PKCS7_VerifySignedData always succeeds regardless of the content data of pkcs7 structure, when passing a detached signature with signature attributes.

I used wolfssl-example code to write pkcs7-verify_detached.c.  Please see the output of this code for detached signatures with/without signature attributes.

$cd wolfssl
$ git describe 
v3.15.7-stable
$./configure --enable-pkcs7 && make && make install

$ cat content.txt 
Hello World
$ cat InvalidContent.txt 
Hello

**Signature verification with no signed attributes in detached** 
$ ./pkcs7-verify_detached signedData_detached_noattrs.der content.txt 
Der 1766
data 11
PKCS7 Verify Success
$ ./pkcs7-verify_detached signedData_detached_noattrs.der InvalidContent.txt 
Der 1766
data 6
RC=-229 >: Signature verification fails here

**Signature verification with signed attributes in the detached signature**
$ ./pkcs7-verify_detached signedData_detached_attrs.der InvalidContent.txt 
Der 1893
data 6
PKCS7 Verify Success >  It should not succeed here
$ ./pkcs7-verify_detached signedData_detached_attrs.der content.txt 
Der 1893
data 11
PKCS7 Verify Success 

**openssl was able to correctly verify the signature for the above cases**
$ openssl smime -verify -content content.txt -in signedData_detached_attrs.der -inform DER -noverify
Verification successful
Hello World
$ openssl smime -verify -content InvalidContent.txt -in signedData_detached_attrs.der -inform DER -noverify
Hello
**Verification failure** 
140186295396000:error:21071065:PKCS7 routines:PKCS7_signatureVerify:digest failure:pk7_doit.c:1170:
140186295396000:error:21075069:PKCS7 routines:PKCS7_verify:signature failure:pk7_smime.c:410:
$ openssl smime -verify -content content.txt -in signedData_detached_noattrs.der -inform DER -noverify
Verification successful
Hello World
$ openssl smime -verify -content InvalidContent.txt -in signedData_detached_noattrs.der -inform DER -noverify
Hello
**Verification failure**
139876027033248:error:04091068:rsa routines:INT_RSA_VERIFY:bad signature:rsa_sign.c:291:
139876027033248:error:21071069:PKCS7 routines:PKCS7_signatureVerify:signature failure:pk7_doit.c:1205:
139876027033248:error:21075069:PKCS7 routines:PKCS7_verify:signature failure:pk7_smime.c:410:
$ openssl version 
OpenSSL 1.0.1f 6 Jan 2014

[pkcs7_detached_sign_verify_example.zip](https://github.com/wolfSSL/wolfssl/files/3501393/pkcs7_detached_sign_verify_example.zip)

kaleb-himes, 2019-08-20:
@cconlon,

Can you or Jacob review.

Thanks!

- KH

cconlon, 2019-08-23:
Hi @csenanda,

Thanks for letting us know about this.  The following PR, which is undergoing testing and peer review, should fix this issue.

https://github.com/wolfSSL/wolfssl/pull/2429

Thanks,
Chris

cconlon, 2019-08-29:
Hi @csenanda 

PR #2429 has been merged into our master branch.  I'm going to mark this as resolved, but feel free to re-open if needed.

Thanks,
Chris

closed 2019-08-29 by cconlon