wolfSSL/wolfssl #2409: How to implement X509_sign?

zero-rp, 2019-08-15:
How to implement X509_sign?

dgarske, 2019-08-15:
Hi @zero-rp ,

We have the `X509_sign` compatibility API in a branch being developed for Apache. You can currently find this here: https://github.com/dgarske/wolfssl/tree/apache_httpd

These changes will be made into a public wolfSSL PR soon. @cariepointer : can you provide some additional details?

Thanks,
David Garske, wolfSSL

cariepointer, 2019-08-21:
HI @zero-rp,

As David mentioned, the compatibility API for `X509_sign` will be made available on master soon. If you wish to work from the development branch, an example use case is noted below:
```
WOLFSSL_EVP_PKEY* priv;
WOLFSSL_EVP_PKEY* pub;
WOLFSSL_X509_NAME *name = wolfSSL_X509_NAME_new();
WOLFSSL_X509* x509 = wolfSSL_X509_new();

// Subject name
wolfSSL_X509_NAME_add_entry_by_txt(name, "country", MBSTRING_UTF8, (byte*)"US", 2, -1, 0);
wolfSSL_X509_NAME_add_entry_by_txt(name, "commonName", MBSTRING_UTF8,(byte*)"wolfssl.com", 11, -1, 0);
wolfSSL_X509_NAME_add_entry_by_txt(name, "emailAddress", MBSTRING_UTF8, (byte*)"support@wolfssl.com", 19, -1, 0);
// ..etc

// Set EVP_PKEY from DER formatted rsaPriv and rsaPub
priv = wolfSSL_d2i_PrivateKey(EVP_PKEY_RSA, NULL, &rsaPriv, (long)priv_key_der_size); 
pub = wolfSSL_d2i_PUBKEY(NULL, &rsaPub, (long)pub_key_der_size);

// Set subject name, add pubkey, then sign
wolfSSL_set_subject_name(x509, name);
wolfSSL_set_pubkey(x509, pub);
// ..etc
wolfSSL_X509_sign(x509, priv, EVP_sha256());
```
Hope this helps! Let us know if you have any additional questions.

Thanks,
Carie Pointer
wolfSSL