wolfSSL/wolfssl #2415: Replace inline asm with compiler built-in intrinsic functions

dr-m, 2019-08-19:
When I experimented with [MemorySanitizer](https://github.com/google/sanitizers/wiki/MemorySanitizerLibcxxHowTo) in MariaDB Server using `clang-8` today (see [MDEV-20377](https://jira.mariadb.org/browse/MDEV-20377) for details), I learned that inline assembler leads to false positives (‘uninitialized value’). Could you please consider replacing the code with intrinsic functions not only on Windows, but across all platforms? I successfully tested with the following:
```diff
diff --git a/wolfcrypt/src/random.c b/wolfcrypt/src/random.c
index 6b0d5dafc..817619d16 100644
--- a/wolfcrypt/src/random.c
+++ b/wolfcrypt/src/random.c
@@ -173,7 +173,7 @@ int wc_RNG_GenerateByte(WC_RNG* rng, byte* b)
     static int wc_GenerateRand_IntelRD(OS_Seed* os, byte* output, word32 sz);
     #endif
 
-#ifdef USE_WINDOWS_API
+#if 1 /*def USE_WINDOWS_API */
     #include <immintrin.h>
 #endif /* USE_WINDOWS_API */
 #endif
@@ -1282,7 +1282,7 @@ int wc_FreeNetRandom(void)
 
 #ifdef HAVE_INTEL_RDSEED
 
-#ifndef USE_WINDOWS_API
+#if 0/*ndef USE_WINDOWS_API*/
 
     /* return 0 on success */
     static WC_INLINE int IntelRDseed64(word64* seed)
@@ -1302,7 +1302,7 @@ int wc_FreeNetRandom(void)
     {
         int ok;
 
-        ok = _rdseed64_step(seed);
+        ok = _rdseed64_step((unsigned long long*) seed);
         return (ok) ? 0 : -1;
     }
 
```
The last hunk was needed in order to silence a warning about type mismatch between `word64` (which was defined as `unsigned long`) and `unsigned long long`. Both should be 64 bits, because I was compiling for AMD64.

Note: this may require some change to the build configuration as well. Within MariaDB, I got a compilation error. I was able to work around that problem by adding `-march=native` to the `CMAKE_C_FLAGS`.

dr-m, 2020-08-31:
MariaDB/server@98fc6b923f69cd054ef715f918f72da72555bad3 works around this problem by disabling the features altogether in {{-fsanitize=memory}} builds.

In MariaDB/server@31e6c96b0449761dc15f548c28ded671d1b7219b I found out that the Intel header files actually work just fine with GCC 5 or newer. I did not check the oldest good `clang` version, but already `clang-4` worked fine. For GCC and Clang, the trick is to enable the extra instructions by function attributes like `__attribute__((target("sse4.2")))`, in code that is executed if the `cpuid` check passed. I tested that it works in GCC 4.8.2.

SparkiDev, 2020-09-03:
For legal reasons we can't accept pull requests without a Contributor Agreement.
Our process for doing this involves the contributor sending a request to support@wolfss.com asking to sign a Contributor Agreement in relation to a specific piece of work.
In this case quote the title of the pull request and include a link to it in the text of the message. 

Could I ask you to do this please?

Thanks,
Sean

dr-m, 2020-09-23:
@SparkiDev Sorry, I missed your request here, but in the end I signed the Contributor Agreement and my pull request #3268 was merged today. Thank you!

closed 2020-09-23 by dr-m