wolfSSL/wolfssl #2448: Android termux build error

patrickkh7788, 2019-09-04:
I try build on Android https://termux.com/ aarch64 shell,  with options `--enable-asm --enable-armasm`.

get this error:

```sh
^
wolfcrypt/src/port/arm/armv8-poly1305.c:261:54: error: unknown token in expression
        "LD4        { v10.4S-v13.4S }, [%[m]], #64 \n\t"
                                                     ^
<inline asm>:58:26: note: instantiated into assembly here
        SUB        x29, x29, #4*#16 
                                ^
  CC       wolfcrypt/src/src_libwolfssl_la-hmac.lo
wolfcrypt/src/port/arm/armv8-poly1305.c:405:54: error: unknown token in expression
        "LD4        { v10.4S-v13.4S }, [%[m]], #64 \n\t"
                                                     ^
<inline asm>:180:26: note: instantiated into assembly here
        SUB        x29, x29, #4*#16 
                                ^
wolfcrypt/src/port/arm/armv8-poly1305.c:463:44: error: unknown token in expression
        "MOV        v5.D[0], v5.D[1]     \n\t"
                                           ^
<inline asm>:228:26: note: instantiated into assembly here
        SUB        x29, x29, #2*#16 
                                ^
wolfcrypt/src/port/arm/armv8-poly1305.c:562:54: error: unknown token in expression
        "LD2        { v10.2D-v11.2D }, [%[m]], #32 \n\t"
                                                     ^
<inline asm>:309:26: note: instantiated into assembly here
        SUB        x29, x29, #2*#16 
                                ^
wolfcrypt/src/port/arm/armv8-poly1305.c:573:49: error: invalid operand for instruction
        "AND        v10.16B, v10.16B, v27.16B \n\t"
                                                ^
<inline asm>:320:19: note: instantiated into assembly here
        MOV        v10.2S[1], v10.2S[2] 
                         ^
wolfcrypt/src/port/arm/armv8-poly1305.c:574:44: error: invalid operand for instruction
        "MOV        v10.2S[1], v10.2S[2] \n\t"
                                           ^
<inline asm>:321:19: note: instantiated into assembly here
        MOV        v11.2S[1], v11.2S[2] 
                         ^
wolfcrypt/src/port/arm/armv8-poly1305.c:575:44: error: invalid operand for instruction
        "MOV        v11.2S[1], v11.2S[2] \n\t"
                                           ^
<inline asm>:322:19: note: instantiated into assembly here
        MOV        v12.2S[1], v12.2S[2] 
                         ^
wolfcrypt/src/port/arm/armv8-poly1305.c:576:44: error: invalid operand for instruction
        "MOV        v12.2S[1], v12.2S[2] \n\t"
                                           ^
<inline asm>:323:19: note: instantiated into assembly here
        MOV        v13.2S[1], v13.2S[2] 
                         ^
wolfcrypt/src/port/arm/armv8-poly1305.c:577:44: error: invalid operand for instruction
        "MOV        v13.2S[1], v13.2S[2] \n\t"
                                           ^
<inline asm>:324:19: note: instantiated into assembly here
        MOV        v14.2S[1], v14.2S[2] 
                         ^
wolfcrypt/src/port/arm/armv8-poly1305.c:655:45: error: unknown token in expression
        "ADD        x10, x10, x9, LSR #26 \n\t"
                                            ^
<inline asm>:386:26: note: instantiated into assembly here
        SUB        x29, x29, #2*#16 
                                ^
wolfcrypt/src/port/arm/armv8-poly1305.c:732:44: error: invalid operand for instruction
        "MUL        x11, x20, x27        \n\t"
                                           ^
<inline asm>:449:33: note: instantiated into assembly here
        MUL        v5.2S, v0.2S, v28.2S[0] 
                                       ^
wolfcrypt/src/port/arm/armv8-poly1305.c:735:44: error: invalid operand for instruction
        "MUL        x12, x20, x28        \n\t"
                                           ^
<inline asm>:451:33: note: instantiated into assembly here
        MUL        v6.2S, v1.2S, v28.2S[0] 
                                       ^
wolfcrypt/src/port/arm/armv8-poly1305.c:737:44: error: invalid operand for instruction
        "MUL        x13, x20, x30        \n\t"
                                           ^
<inline asm>:453:33: note: instantiated into assembly here
        MUL        v7.2S, v2.2S, v28.2S[0] 
                                       ^
wolfcrypt/src/port/arm/armv8-poly1305.c:739:44: error: invalid operand for instruction
        "MADD       x9, x21, x19, x9     \n\t"
                                           ^
<inline asm>:455:33: note: instantiated into assembly here
        MUL        v8.2S, v3.2S, v28.2S[0] 
                                       ^
wolfcrypt/src/port/arm/armv8-poly1305.c:741:44: error: invalid operand for instruction
        "MADD       x10, x21, x25, x10   \n\t"

```

JacobBarthelmeh, 2019-09-04:
Hi @patrickkh7788 , There is a PR here https://github.com/wolfSSL/wolfssl/pull/2438 for poly1305 clang build fixes. Does this PR also address this issue seen when compiling?
Regards,
Jacob

patrickkh7788, 2019-09-05:
thanks for reply, #2438 fix the problem.

julek-wolfssl, 2019-09-05:
Hi @patrickkh7788 ,
it appears that some operating systems reserve the w18/x18 register for system calls. Windows uses it as a "Platform register: in kernel mode, points to KPCR (Kernel Processor Control Region) for the current processor; in user mode, points to TEB (Thread Environment Block)". I'll look into trying to move the assembly code from using this register.
Sincerely
Juliusz

patrickkh7788, 2019-09-07:
thanks for explain.

patrickkh7788, 2019-09-09:
Should I disable  the arm asm for android until w18/x18 register removed from wolfSSL ?

julek-wolfssl, 2019-09-23:
Hi @patrickkh7788 ,
sorry for the delay. To solve your issue, we removed the use of register x18 from our ARM optimizations (pull requests #2453 and #2454). These changes are merged and present in our master branch. Could you please check and confirm that this solves your issue?
Sincerely
Juliusz

patrickkh7788, 2019-09-25:
hi @julek-wolfssl 

I confirm it build ok on android, and passed my owned chacha20 test case.

closed 2019-09-25 by patrickkh7788