wolfSSL/wolfssl #2457: macOSX build error

patrickkh7788, 2019-09-07:
try build x25519 on macOSX:

```sh
wolfcrypt/src/fe_x25519_asm.S:1044:2: error: unknown directive
 .type cpuFlagsSet, @object
 ^
wolfcrypt/src/fe_x25519_asm.S:1045:2: error: unknown directive
 .size cpuFlagsSet,4
 ^
wolfcrypt/src/fe_x25519_asm.S:1049:2: error: unknown directive
 .type intelFlags, @object
 ^
wolfcrypt/src/fe_x25519_asm.S:1050:2: error: unknown directive
 .size intelFlags,4
 ^
wolfcrypt/src/fe_x25519_asm.S:1054:2: error: unknown directive
 .type fe_mul_p, @object
 ^
wolfcrypt/src/fe_x25519_asm.S:1055:2: error: unknown directive
 .size fe_mul_p,8
 ^
wolfcrypt/src/fe_x25519_asm.S:1059:2: error: unknown directive
 .type fe_sq_p, @object
 ^
wolfcrypt/src/fe_x25519_asm.S:1060:2: error: unknown directive
 .size fe_sq_p,8
 ^
wolfcrypt/src/fe_x25519_asm.S:1064:2: error: unknown directive
 .type fe_mul121666_p, @object
 ^
wolfcrypt/src/fe_x25519_asm.S:1065:2: error: unknown directive
 .size fe_mul121666_p,8
 ^
wolfcrypt/src/fe_x25519_asm.S:1069:2: error: unknown directive
 .type fe_sq2_p, @object
 ^
wolfcrypt/src/fe_x25519_asm.S:1070:2: error: unknown directive
 .size fe_sq2_p,8
 ^
wolfcrypt/src/fe_x25519_asm.S:1074:2: error: unknown directive
 .type fe_invert_p, @object
 ^
wolfcrypt/src/fe_x25519_asm.S:1075:2: error: unknown directive
 .size fe_invert_p,8
 ^
wolfcrypt/src/fe_x25519_asm.S:1079:2: error: unknown directive
 .type curve25519_p, @object
 ^
wolfcrypt/src/fe_x25519_asm.S:1080:2: error: unknown directive
 .size curve25519_p,8
 ^
wolfcrypt/src/fe_x25519_asm.S:1084:2: error: unknown directive
 .type fe_pow22523_p, @object
 ^
wolfcrypt/src/fe_x25519_asm.S:1085:2: error: unknown directive
 .size fe_pow22523_p,8
 ^
wolfcrypt/src/fe_x25519_asm.S:1089:2: error: unknown directive
 .type fe_ge_to_p2_p, @object
 ^
wolfcrypt/src/fe_x25519_asm.S:1090:2: error: unknown directive
 .size fe_ge_to_p2_p,8
 ^
wolfcrypt/src/fe_x25519_asm.S:1094:2: error: unknown directive
 .type fe_ge_to_p3_p, @object
 ^
wolfcrypt/src/fe_x25519_asm.S:1095:2: error: unknown directive
 .size fe_ge_to_p3_p,8
 ^
wolfcrypt/src/fe_x25519_asm.S:1099:2: error: unknown directive
 .type fe_ge_dbl_p, @object
 ^
wolfcrypt/src/fe_x25519_asm.S:1100:2: error: unknown directive
 .size fe_ge_dbl_p,8
 ^
wolfcrypt/src/fe_x25519_asm.S:1104:2: error: unknown directive
 .type fe_ge_madd_p, @object
 ^
wolfcrypt/src/fe_x25519_asm.S:1105:2: error: unknown directive
 .size fe_ge_madd_p,8
 ^
wolfcrypt/src/fe_x25519_asm.S:1109:2: error: unknown directive
 .type fe_ge_msub_p, @object
 ^
wolfcrypt/src/fe_x25519_asm.S:1110:2: error: unknown directive
 .size fe_ge_msub_p,8
 ^
wolfcrypt/src/fe_x25519_asm.S:1114:2: error: unknown directive
 .type fe_ge_add_p, @object
 ^
wolfcrypt/src/fe_x25519_asm.S:1115:2: error: unknown directive
 .size fe_ge_add_p,8
 ^
wolfcrypt/src/fe_x25519_asm.S:1119:2: error: unknown directive
 .type fe_ge_sub_p, @object
 ^
wolfcrypt/src/fe_x25519_asm.S:1120:2: error: unknown directive
 .size fe_ge_sub_p,8
```

SparkiDev, 2019-09-08:
Hi Patrick,

This was fixed in pull request #2423. (https://github.com/wolfSSL/wolfssl/pull/2423)
Let me know if it doesn't work for you.

Sean

--
Sean Parkinson, wolfSSL
Senior Software Engineer

patrickkh7788, 2019-09-09:
Thanks, they work.

closed 2019-09-09 by patrickkh7788