wolfSSL/wolfssl #2459: Heap-overread bug in checking cert signature

flyroom, 2019-09-10:
Hi, another heap-overread bug which could be triggerd remotely
tested on wolfSSL 4.1.0 / ubuntu 14.04/ clang /gcc / asan / generate Makefile with ./configure --enable-lowresource
Missing sanity checks before CheckCertSignature_ex wolfcrypt/src/asn.c:8041
```
if (ret == 0) {
                    if (cert[extIdx] == ASN_BOOLEAN) {
                        if (GetBoolean(cert, &extIdx, certSz) < 0)
                            ret = ASN_PARSE_E;
                    }
}
```

You could trigger it through following steps:
1. start wolfss example server program under directory "examples/server" by invoking: ./server -b -p 4433
2. start py script client_raw.py  which sends crafted messages to local port 4433 [client_raw.zip](https://github.com/wolfSSL/wolfssl/files/3595168/client_raw.zip)

ASAN report ( if examples/server is compiled with asan ):
```
=================================================================
==12298==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x619000000e94 at pc 0x7f483b36f170 bp 0x7ffc7d33a9a0 sp 0x7ffc7d33a998
READ of size 1 at 0x619000000e94 thread T0
    #0 0x7f483b36f16f in CheckCertSignature_ex wolfcrypt/src/asn.c:8041
    #1 0x7f483b36fabb in CheckCertSignature wolfcrypt/src/asn.c:8158
    #2 0x7f483b3a80f7 in ProcessPeerCertParse src/internal.c:9506
    #3 0x7f483b3a9c0b in ProcessPeerCerts src/internal.c:10111
    #4 0x7f483b3ab4d7 in DoCertificate src/internal.c:10704
    #5 0x7f483b3adec5 in DoHandShakeMsgType src/internal.c:11542
    #6 0x7f483b3aed7b in DoHandShakeMsg src/internal.c:11733
    #7 0x7f483b3b6534 in ProcessReply src/internal.c:14189
    #8 0x7f483b330784 in wolfSSL_accept src/ssl.c:11083
    #9 0x7f483b321f65 in server_test examples/server/server.c:2113
    #10 0x7f483b3226e1 in main examples/server/server.c:2413
    #11 0x7f4839e53f44 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21f44)
    #12 0x7f483b31d748  (/opt/disk/binary_analysis/fuzz/wolfssl/wolfssl_github/examples/server/server+0x13748)

0x619000000e94 is located 0 bytes to the right of 1044-byte region [0x619000000a80,0x619000000e94)
allocated by thread T0 here:
    #0 0x7f483a7d9479 in __interceptor_malloc /home/nwani/m3/conda-bld/compilers_linux-64_1560109574129/work/.build/x86_64-conda_cos6-linux-gnu/src/gcc/libsanitizer/asan/asan_malloc_linux.cc:144
    #1 0x7f483b3558fd in wolfSSL_Malloc wolfcrypt/src/memory.c:140

SUMMARY: AddressSanitizer: heap-buffer-overflow wolfcrypt/src/asn.c:8041 in CheckCertSignature_ex
Shadow bytes around the buggy address:
  0x0c327fff8180: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c327fff8190: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c327fff81a0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c327fff81b0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c327fff81c0: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0c327fff81d0: 00 00[04]fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c327fff81e0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c327fff81f0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c327fff8200: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c327fff8210: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c327fff8220: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==12298==ABORTING
```

JacobBarthelmeh, 2019-09-11:
Hi @flyroom, Thank you for the report!

JacobBarthelmeh, 2019-09-22:
@flyroom PR#2465 was merged in to fix the reported issue.

closed 2019-09-22 by JacobBarthelmeh