wolfSSL/wolfssl #2500: Debian packages

zabbal, 2019-10-07:
Are there official .deb packages for Debian/Ubuntu available somewhere?

kaleb-himes, 2019-10-07:
Hi @zabbal,

Thank you for reaching out to wolfSSL support. On Debian/Ubuntu you can use ```apt-cache search wolfssl``` to check if it is available on your release. If available you should see the below two options:

```
libwolfssl-dev - Development files for the wolfSSL encryption library
libwolfssl15 - wolfSSL encryption library
```

libwolfssl-dev contains all the SSL/TLS and crypto. libwolfssl15 contains just the crypto without SSL/TLS API's. Let us know if there is anything else!

Warm Regards,

K

danielinux, 2019-10-07:
more details can be found straight from the official repositories.

Links below:

Debian:
https://packages.debian.org/search?keywords=libwolfssl

Ubuntu:
https://packages.ubuntu.com/search?keywords=libwolfssl

zabbal, 2019-10-08:
My bad, missed the obvious.

closed 2019-10-08 by zabbal