wolfSSL/wolfssl #2518: several possible null pointer dereferences found by cppcheck

chipitsine, 2019-10-18:
```
[IDE/MDK-ARM/MDK-ARM/wolfSSL/shell.c:439] -> [IDE/MDK-ARM/MDK-ARM/wolfSSL/shell.c:440]: (warning) Either the condition 'args==NULL' is redundant or there is possible null pointer dereference: args.
[IDE/MDK5-ARM/Projects/wolfSSL-Full/shell.c:480] -> [IDE/MDK5-ARM/Projects/wolfSSL-Full/shell.c:481]: (warning) Either the condition 'args==NULL' is redundant or there is possible null pointer dereference: args.
[examples/benchmark/tls_bench.c:1282] -> [examples/benchmark/tls_bench.c:1258]: (warning) Either the condition 'if(args)' is redundant or there is possible null pointer dereference: args.
[examples/benchmark/tls_bench.c:1282] -> [examples/benchmark/tls_bench.c:1259]: (warning) Either the condition 'if(args)' is redundant or there is possible null pointer dereference: args.
[src/bio.c:459] -> [src/bio.c:539]: (warning) Either the condition 'bio' is redundant or there is possible null pointer dereference: bio.
[src/internal.c:15001] -> [src/internal.c:14993]: (warning) Either the condition 'ssl==NULL' is redundant or there is possible null pointer dereference: ssl.
[src/internal.c:20708] -> [src/internal.c:20704]: (warning) Either the condition 'ssl==NULL' is redundant or there is possible null pointer dereference: ssl.
[src/internal.c:20796] -> [src/internal.c:20794]: (warning) Either the condition 'ssl==NULL' is redundant or there is possible null pointer dereference: ssl.
[src/ssl.c:5227] -> [src/ssl.c:5216]: (warning) Either the condition 'if(ctx)' is redundant or there is possible null pointer dereference: ctx.
[src/ssl.c:5286] -> [src/ssl.c:5273]: (warning) Either the condition 'if(ctx)' is redundant or there is possible null pointer dereference: ctx.
[src/ssl.c:37828] -> [src/ssl.c:37798]: (warning) Either the condition 'key!=NULL' is redundant or there is possible null pointer dereference: key.
[src/tls.c:5566] -> [src/tls.c:5561]: (warning) Either the condition 'extensions==NULL' is redundant or there is possible null pointer dereference: extensions.
[tests/api.c:2183] -> [tests/api.c:2236]: (warning) Either the condition 'args' is redundant or there is possible null pointer dereference: args.
[wolfcrypt/src/asn.c:9396] -> [wolfcrypt/src/asn.c:9317]: (warning) Either the condition 'if(pBuffer)' is redundant or there is possible null pointer dereference: pBuffer.
[wolfcrypt/src/ecc.c:4040] -> [wolfcrypt/src/ecc.c:4038]: (warning) Either the condition 'key==NULL' is redundant or there is possible null pointer dereference: key.
[wolfcrypt/src/ecc.c:7223] -> [wolfcrypt/src/ecc.c:7220]: (warning) Either the condition 'key==NULL' is redundant or there is possible null pointer dereference: key.
[wolfcrypt/src/pkcs7.c:2447] -> [wolfcrypt/src/pkcs7.c:2440]: (warning) Either the condition 'if(output2&&output2Sz)' is redundant or there is pointer arithmetic with NULL pointer.
[wolfcrypt/src/pkcs7.c:2447] -> [wolfcrypt/src/pkcs7.c:2442]: (warning) Either the condition 'if(output2&&output2Sz)' is redundant or there is pointer arithmetic with NULL pointer.
[wolfcrypt/src/pkcs7.c:2447] -> [wolfcrypt/src/pkcs7.c:2444]: (warning) Either the condition 'if(output2&&output2Sz)' is redundant or there is pointer arithmetic with NULL pointer.
[wolfcrypt/src/pkcs7.c:8383] -> [wolfcrypt/src/pkcs7.c:8381]: (warning) Either the condition 'idx==NULL' is redundant or there is possible null pointer dereference: idx.
[wolfcrypt/src/random.c:2349] -> [wolfcrypt/src/random.c:2376]: (warning) Either the condition 'os!=NULL' is redundant or there is possible null pointer dereference: os.
[wolfcrypt/src/random.c:2349] -> [wolfcrypt/src/random.c:2377]: (warning) Either the condition 'os!=NULL' is redundant or there is possible null pointer dereference: os.
[wolfcrypt/src/sp_int.c:1734] -> [wolfcrypt/src/sp_int.c:1821]: (warning) Either the condition 'result==NULL' is redundant or there is possible null pointer dereference: result.
[wolfcrypt/user-crypto/src/rsa.c:696] -> [wolfcrypt/user-crypto/src/rsa.c:695]: (warning) Either the condition 'mont==NULL' is redundant or there is possible null pointer dereference: mont.

```

JacobBarthelmeh, 2019-10-18:
Thanks for the report:

Most of these are due to redundant checks we have to make extra certain that no code changes will in the future cause a NULL dereference. There was a couple cases I saw and have listed here.

WOLFSSL_CRYPTOCELL build
[wolfcrypt/src/ecc.c:4040] -> [wolfcrypt/src/ecc.c:4038]: (warning) Either the condition 'key==NULL' is redundant or there is possible null pointer dereference: key.
[wolfcrypt/src/ecc.c:7223] -> [wolfcrypt/src/ecc.c:7220]: (warning) Either the condition 'key==NULL' is redundant or there is possible null pointer dereference: key.

QSH build
[src/internal.c:20708] -> [src/internal.c:20704]: (warning) Either the condition 'ssl==NULL' is redundant or there is possible null pointer dereference: ssl.
[src/internal.c:20796] -> [src/internal.c:20794]: (warning) Either the condition 'ssl==NULL' is redundant or there is possible null pointer dereference: ssl.

Regards,
Jacob

JacobBarthelmeh, 2019-11-01:
Hi @chipitsine , we opened a PR for the QSH and CRYPTOCELL potential null dereferences. Could you tell us a little about your project?
Regards, Jacob

chipitsine, 2019-11-01:
Pardon, what do you mean by project?

JacobBarthelmeh, 2019-11-04:
The use case with wolfSSL. We like to hear about peoples hobbies they used wolfSSL in, opensource projects, and projects in general out of curiosity. It also helps us to know some about what our users want and if we can potentially help out.
Regards, Jacob

chipitsine, 2019-11-04:
I participiate in haproxy, SE VPN, openvpn.

chipitsine, 2019-11-04:
if we will come with some pull request to SE VPN, most probably it will be accepted

JacobBarthelmeh, 2019-11-04:
@chipitsine thank you for the feedback on the project, and the note about SE VPN!

closed 2019-12-07 by JacobBarthelmeh