wolfSSL/wolfssl #2528: Add OpenSSL compatibility API’s  for picotls support!

patrickkh7788, 2019-10-27:
the missing api:

EC_POINT_oct2point
EVP_PKEY_CTX_set_rsa_pss_saltlen
EVP_PKEY_CTX_set_rsa_mgf1_md
HMAC_size
CRYPTO_add
CRYPTO_LOCK_X509_STORE
CRYPTO_LOCK_EVP_PKEY
X509_FILETYPE_DEFAULT
X509_STORE_up_ref
X509_PURPOSE_SSL_SERVER

tmael, 2019-10-29:
Hi @patrickkh7788,

Thanks for contacting wolfSSL with your request!
 
You are correct that those APIs are not available yet. 
We are constantly expanding the OpenSSL compatibility layer based on customer needs and the missing APIs didn't come up before. 

Any good reasons why you are building our library with another TLS? 

Thanks,
Tesfa

patrickkh7788, 2019-11-01:
Thanks for quick replay.

picotls provide nice api to control TLS ticket, EDNS and others.  It also allow user to switch backend(Openssl, Windows native, minicrypto).  

I already create a async api with picotls,  but with openssl & Picotls the binary  size is huge. that is why I try with wolfSSL.  

wolfSSL provide nice build options to control binary size, but lake support EDNS.

patrickkh7788, 2020-02-22:
hi @tmael 

Today I build wolfssl with picotls,  find that `EVP_chacha20` is missing, so I add them to the list and remove some already exists apis.

Can you give some tips about how to add `EVP_chacha20` ? I can add patch to make picotls work without rest apis, but chach20 is a must have feature, and I guess other projects also could depends on `EVP_chacha20`  since it is the most frequency mobile encryption stream cipher.

Thanks,  
Patrick

bgermann, 2020-09-19:
I second EVP_PKEY_CTX_set_rsa_pss_saltlen for SWUpdate: https://groups.google.com/forum/#!topic/swupdate/HQYFvx8T8Ao

tmael, 2021-01-13:
Closing because we are tracking this feature request internally. 
If you would like to expedite the request, please contact us at support@wolfssl.com

closed 2021-01-13 by tmael