wolfSSL/wolfssl #2546: There is a coding-miss of MACRO in pkcs7.c

guangqing, 2019-11-01:
in wc_PKCS7_DecodeAuthEnvelopedData(PKCS7* pkcs7, byte* in,
                                                 word32 inSz, byte* output,
                                                 word32 outputSz)

#if defined(HAVE_AESGCM) || defined(HAVE_AESCC)
should be 
#if defined(HAVE_AESGCM) || defined(HAVE_AESCCM)

guangqing, 2019-11-01:
in asn.h

#ifndef IGNORE_NAME_CONSTRAINT
should be 
#ifndef IGNORE_NAME_CONSTRAINTS

JacobBarthelmeh, 2019-11-01:
Hi @guangqing ,

Thank you for the report, I reviewed it and it appears accurate. Will be opening a PR with a fix shortly.

Regards,
Jacob

JacobBarthelmeh, 2019-11-01:
Out of curiosity was a tool used to find these or a build that you are using that ran into it?

closed 2019-11-11 by JacobBarthelmeh