wolfSSL/wolfssl #2552: wolfssl 4.2.0 does not compile on Windows, with MSVC compiler, and with MariaDB

vaintroub, 2019-11-02:
WolfSSL 4.2.0 has a number of issues with MariaDB, thus it cannot replace 4.1.0 yet . There are 3 issues with the new version, which I list below

- First issue (major)

From MariaDB build attempt at http://buildbot.askmonty.org/buildbot/builders/winx64-packages/builds/15860/steps/build/logs/stdio 

 25>D:\winx64-packages\build\src\extra\wolfssl\wolfssl\src\ssl.c(8141,16): error C2057:  expected constant expression [D:\winx64-packages\build\extra\wolfssl\wolfssl.vcxproj]
    25>D:\winx64-packages\build\src\extra\wolfssl\wolfssl\src\ssl.c(8141,16): error C2466:  cannot allocate an array of constant size 0 [D:\winx64-

Inspecting source code at the mentioned line, we have

    const int sz = CTC_NAME_SIZE*2;
    int rc = WOLFSSL_FAILURE;
    char tmp[sz];

I do not think you can do that in C portably,  in C++ it will work probably, and with gcc it will work, but in standard C, you need to have a compile time constant for the size of tmp array, like 

    char tmp[CTC_NAME_SIZE*2];

- Second issue (which is maybe arguable) is the introduction of the new Timeval struct, it is included through standard headers via ssl.h . 

This brings compile errors such as

35>D:\winx64-packages\build\src\extra\wolfssl\wolfssl\wolfssl/callbacks.h(62,7): error C2371:  'Timeval': redefinition; different basic types [D:\winx64-packages\build\sql\sql_sequence.vcxproj]

from the buildbot log link

The reason is that MariaDB defines its own Timeval elsewhere. I'd like to suggest to rename the own Timeval into something that has a "namespace", e.g WolfSSL_Timeval, to prevent clashes with something that its users also define, or to avoid inclusion of new custom structs via standard headers.

- Third issue (which is minor), from this attempt to build is 

D:\winx64-packages\build\src\vio\viosslfactories.c(276,31): warning C4013:  'SSL_CTX_set_ciphersuites' undefined; assuming extern returning int [D:\winx64-packages\build\vio\vio.vcxproj]

The problem here is following code in MariaDB 's header

#ifndef TLS1_3_VERSION
#define SSL_CTX_set_ciphersuites(X,Y) 0
#endif

So, it assumes SSL_CTX_set_ciphersuites is there if TLS1_3_VERSION is defined. And TLS1_3_VERSION is now defined, but the function, unlike in OpenSSL, is njot there.

kaleb-himes, 2019-11-07:
@vaintroub,

Just an update, my windows test bed died and I am waiting for a new one to arrive so I can reproduce and work on this issue. Apologies for the delay.

Cheers,

K

kaleb-himes, 2019-12-18:
@vaintroub,

I have a new test bed but can't seem to reproduce these same issues on wolfSSL master. I'm sure it has something to do with the settings used to build.

Can you send me your user_settings.h so I can test in our Visual Studio solution projects?

Warm Regards,

K

vaintroub, 2019-12-18:
Sure
https://github.com/MariaDB/server/blob/10.5/extra/wolfssl/user_settings.h.in

vaintroub, 2019-12-18:
it is a cmake file, which is configured, after configuration step I get

#ifndef WOLFSSL_USER_SETTINGS_H
#define WOLFSSL_USER_SETTINGS_H

#define HAVE_CRL
#define WOLFSSL_MYSQL_COMPATIBLE
#define HAVE_ECC
#define ECC_TIMING_RESISTANT
#define HAVE_HASHDRBG
#define WOLFSSL_AES_DIRECT
#define WOLFSSL_SHA384
#define WOLFSSL_SHA512
#define WOLFSSL_SHA224
#define SESSION_CERT
#define KEEP_OUR_CERT
#define WOLFSSL_STATIC_RSA
#define WOLFSSL_USER_IO
#define WC_RSA_BLINDING
#define HAVE_TLS_EXTENSIONS
#define HAVE_AES_ECB
#define WOLFSSL_AES_COUNTER
#define NO_WOLFSSL_STUB
#define OPENSSL_ALL
#define WOLFSSL_ALLOW_TLSV10

#define WOLFSSL_AESNI
#define USE_FAST_MATH
#define TFM_TIMING_RESISTANT
#define HAVE_INTEL_RDSEED
/* #undef USE_INTEL_SPEEDUP */
#define FP_MAX_BITS 16384
#define USE_FAST_MATH
#define WOLFSSL_X86_64_BUILD

#endif  /* WOLFSSL_USER_SETTINGS_H */

kaleb-himes, 2019-12-18:
Thank you so much @vaintroub! I can reproduce the issues with those settings. I'll work on getting a PR up shortly!

- K

kaleb-himes, 2019-12-19:
OK Fix:

1) I will address the non-const in array declaration.
2) This has already been fixed in master
3) Is it possible that the wolfssl/openssl/ headers are NOT being included? (https://github.com/wolfSSL/wolfssl/blob/master/wolfssl/openssl/ssl.h#L260)

- K

kaleb-himes, 2019-12-19:
@vaintroub,

> Is it possible that the wolfssl/openssl/ headers are NOT being included? (https://github.com/wolfSSL/wolfssl/blob/master/wolfssl/openssl/ssl.h#L261)

In other words, yes there will be no API SSL_CTX_set_ciphersuites but if it is properly being re-mapped to wolfSSL_CTX_set_cipher_list is should resolve. If the actual openssl headers are in the include path and included in preference to the wolfssl/openssl headers then the re-mapping would fail.

Can you double check in this case that the wolfssl/openssl header are preferred to actual openssl headers to resolve item 3?

Regards,

-K

vaintroub, 2019-12-19:
It complained about SSL_CTX_set_ciphersuites, not SSL_CTX_set_cipher_list. 
The "suites" is now there, but it seems to be a relatively new addition (it is in master branch, but not in 4.2.0-stable about which I complained)
https://github.com/wolfSSL/wolfssl/blob/v4.2.0-stable/wolfssl/openssl/ssl.h

vaintroub, 2019-12-19:
I think you can resolve 3.
Thanks very much for your help!

kaleb-himes, 2020-01-08:
@vaintroub,

Not a problem, thank you so much for the report. Happy New Year!

- K

closed 2020-01-08 by kaleb-himes