wolfSSL/wolfssl #2555: Heap based buffer overflow while parsing crafted X.509 certificates

cve-reporting, 2019-11-04:
WolfSSL in versions 4.1.0 and 4.2.0 incorrectly handles X.509 certificates leading to a heap-buffer overflow inside the DecodedCert structure, overwriting a NULL pointer and as a result crash during memory deallocation.
This vulnerability affects both client and server in two supported protocols: TLS and DTLS.

During processing of a crafted certificate, WolfSSL incorrectly handles the loc buffer in the DecodedName structure.
In the following line the count variable reaches value 21, while the loc table has fixed size 19:
wolfcrypt/src/asn.c:5121:
>
>    dName->loc[count++] = id;
>

Declarations:
wolfssl/wolfcrypt/asn.h:
> 
> #define DOMAIN_COMPONENT_MAX 10
> #define DN_NAMES_MAX 9
> 
> 
> struct DecodedName {
>     char*   fullName;
> ...
>     int     loc[DOMAIN_COMPONENT_MAX + DN_NAMES_MAX];
>     int     locSz;
> };
> 
> struct DecodedCert {
> ...
>     DecodedName issuerName;
>     DecodedName subjectName;
> 
 

Overflow of the loc table in the issuerName leads to overwritting of the fullName pointer in the subjectName.
During deallocation in the FreeDecodedCert check for cert->subjectName.fullName is ineffective, leading to execution of the free() function on the overwritten pointer.

Crashing function:
wolfcrypt/src/asn.c:4505:

> void FreeDecodedCert(DecodedCert* cert)
> {
> ...
>     if (cert->subjectName.fullName != NULL)
>         XFREE(cert->subjectName.fullName, cert->heap, DYNAMIC_TYPE_X509);
> 

-------------------------------------

Proposed CVSS 3.0 score:

7.5 (High)
CVSS:3.0/AV:N/AC:L/PR:N/UI:N/S:U/C:N/I:N/A:H

-----------------------------------------------------------

Crash on DTLS client connection (parsing cert from incoming network packets):

> ./examples/client/client -u -h 1xx.1xx.1xx.1xx -p 4433 -g
> ASAN:SIGSEGV
> 
> ==21175==ERROR: AddressSanitizer: SEGV on unknown address 0xfffffffffffffffa (pc 0x7fe754d335fc bp 0xfffffffffffffffa sp 0x7ffc731221f0 T0)
>     #0 0x7fe754d335fb  (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x215fb)
>     #1 0x7fe754daa29d in __interceptor_free (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x9829d)
>     #2 0x7fe75455ea98 in FreeDecodedCert wolfcrypt/src/asn.c:4532
>     #3 0x7fe7547a50ec in FreeProcPeerCertArgs src/internal.c:9548
>     #4 0x7fe7547a50ec in ProcessPeerCerts src/internal.c:10807
>     #5 0x7fe75480d92c in DoCertificate src/internal.c:10843
>     #6 0x7fe75480d92c in DoHandShakeMsgType src/internal.c:11681
>     #7 0x7fe754825b6c in DoDtlsHandShakeMsg src/internal.c:12305
>     #8 0x7fe754825b6c in ProcessReply src/internal.c:14334
>     #9 0x7fe754888c07 in wolfSSL_connect src/ssl.c:11172
>     #10 0x413527 in client_test examples/client/client.c:2805
[... 110 lines cut ...]
> ==14502==ERROR: AddressSanitizer: SEGV on unknown address 0xfffffffffffffffa (pc 0x7fb7139575fc bp 0xfffffffffffffffa sp 0x7ffe8c8d8180 T0)
>     #0 0x7fb7139575fb  (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x215fb)
>     #1 0x7fb7139ce29d in __interceptor_free (/usr/lib/x86_64-linux-gnu/libasan.so.2+0x9829d)
>     #2 0x7fb713182a98 in FreeDecodedCert wolfcrypt/src/asn.c:4532
>     #3 0x7fb713492d2d in ProcessBuffer src/ssl.c:5398
>     #4 0x7fb713498ad3 in ProcessFile src/ssl.c:6339
>     #5 0x7fb71349ef4e in wolfSSL_CTX_use_certificate_chain_file src/ssl.c:6912
>     #6 0x40dc34 in server_test examples/server/server.c:1610
>     #7 0x40585e in main examples/server/server.c:2434
>     #8 0x7fb712a8f82f in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2082f)
>     #9 0x405a38 in _start (wolfssl-4.2.0c_ASAN/examples/server/.libs/lt-server+0x405a38)
> 
> AddressSanitizer can not provide additional info.
> SUMMARY: AddressSanitizer: SEGV ??:0 ??
> ==14502==ABORTING
> 

-----------------------------------------------------------

Reproduction:

Download and compile WolfSSL 4.2.0 or 4.1.0.
Unzip and use attached crafted certificate with above commands (./example/server/server -c ... or ./example/client/client -c ...).

[crash_000_FreeDecodedCert.zip](https://github.com/wolfSSL/wolfssl/files/3805130/crash_000_FreeDecodedCert.zip)

kaleb-himes, 2019-11-05:
@cve-reporting,

Thank you so much for the report. We will work on reproducing the reported issue and update accordingly.

- The wolfSSL team

kaleb-himes, 2019-11-05:
@cve-reporting,

Just so you are aware you are most welcome to report these via support@wolfssl.com so we have time to address the issue before it is public. It is most responsible to coordinate with an effected vendor before making details public to avoid 0-day exploits.

We appreciate all your hard work please keep the reports coming but please send them to support@wolfssl.com first if it is not too much to ask.

Thanks and Regards,

the wolfSSL team

kaleb-himes, 2019-11-05:
@cve-reporting,

A fix has been opened here: https://github.com/wolfSSL/wolfssl/pull/2563

Can you shoot us an email at support@wolfssl.com once you have had a chance to review the fix please?

Warm Regards,

K

closed 2019-11-05 by kaleb-himes

cve-reporting, 2019-11-07:
@kaleb-himes 

Thank you for a quick confirmation of this issue and I really appreciate you efforts to quickly fix it.
Next time I will first sent you information via an email, as you have proposed.

In past we (our team that shares this account for reporting) used to try disclose issues more responsibly, however recently our notifications were sometimes ignored, so we switched to direct reporting in an issue tracker.

kaleb-himes, 2019-11-07:
@cve-reporting,

> In past we (our team that shares this account for reporting) used to try disclose issues more responsibly, however recently our notifications were sometimes ignored, so we switched to direct reporting in an issue tracker.

Totally understandable! I can confidently say that with wolfSSL you will never experience this! Here is a blog post on how our team responds to any vulnerability report: https://www.wolfssl.com/everything-wanted-know-wolfssl-support-handles-vulnerability-reports-afraid-ask/

Thanks again for your teams efforts and hard work and willingness to coordinate with us on all future reports.

Sincerest Regards,

- Kaleb and the wolfSSL Team

cve-reporting, 2019-11-09:
Registered as: CVE-2019-18840