wolfSSL/wolfssl #2559: Crash caused by double free in SSL and DTLS servers

cve-reporting, 2019-11-05:
WolfSSL example server in all versions starting from 3.12.0 (including current 4.2.0c) incorrectly handles error when server certificate cannot be read from a given path.
This vulnerability affects the server in both supported protocols: TLS and DTLS.

Crashing function:
examples/server/server.c:2165-2197

>     {
>         byte*  rnd;
>         byte*  pt;
>         size_t size;
> 
>         /* get size of buffer then print */
>         size = wolfSSL_get_server_random(NULL, NULL, 0);
>         if (size == 0) {
>             err_sys_ex(runWithErrors, "error getting server random buffer size");
>         }
> 
>         rnd = (byte*)XMALLOC(size, NULL, DYNAMIC_TYPE_TMP_BUFFER);
>         if (rnd == NULL) {
>             err_sys_ex(runWithErrors, "error creating server random buffer");
>         }
> 
>         size = wolfSSL_get_server_random(ssl, rnd, size);
>         if (size == 0) {
>     2183:   XFREE(rnd, NULL, DYNAMIC_TYPE_TMP_BUFFER);
>             err_sys_ex(runWithErrors, "error getting server random buffer");
>         }
> 
>         printf("Server Random : ");
>         pt = rnd;
>         if (pt != NULL) {
>             for (pt = rnd; pt < rnd + size; pt++) printf("%02X", *pt);
>             printf("\n");
>         } else {
>             err_sys_ex(runWithErrors, "error: attempted to dereference null "
>                                                                    "pointer");
>         }
>     2196:XFREE(rnd, NULL, DYNAMIC_TYPE_TMP_BUFFER);
>     }

Buffer rnd is freed at line 2183 and then freed again on line 2196.

-------------------------------------

Crash message on SSL server (with ASAN):

> ./examples/server/server -p 4433 -b -x -c s
> wolfSSL error: can't load server cert file, check file and run from wolfSSL home dir
> Continuing server execution...
> 
> wolfSSL error: unable to get SSL
> Continuing server execution...
> 
> wolfSSL error: unable to enable CRL
> Continuing server execution...
> 
> wolfSSL error: unable to load CRL
> Continuing server execution...
> 
> wolfSSL error: unable to set CRL callback url
> Continuing server execution...
> 
> wolfSSL error: error in setting fd
> Continuing server execution...
> 
> SSL_accept error -173, Bad function argument
> wolfSSL error: SSL_accept failed
> Continuing server execution...
> 
> peer has no cert!
[... 249 lines cut ...]
> 7f8321c01000-7f8321c10000 rw-p 00000000 00:00 0 
> 7f8321c26000-7f8321c28000 rw-p 00000000 00:00 0 
> 7f8321c28000-7f8321c29000 r--p 00025000 fd:01 28169                      /lib/x86_64-linux-gnu/ld-2.23.so
> 7f8321c29000-7f8321c2a000 rw-p 00026000 fd:01 28169                      /lib/x86_64-linux-gnu/ld-2.23.so
> 7f8321c2a000-7f8321c2b000 rw-p 00000000 00:00 0 
> 7ffc81616000-7ffc81637000 rw-p 00000000 00:00 0                          [stack]
> 7ffc817b5000-7ffc817b8000 r--p 00000000 00:00 0                          [vvar]
> 7ffc817b8000-7ffc817ba000 r-xp 00000000 00:00 0                          [vdso]
> ffffffffff600000-ffffffffff601000 r-xp 00000000 00:00 0                  [vsyscall]
> Aborted

-----------------------------------------------------------

Reproduction:

1. Download and compile WolfSSL in version 3.12.0 or later.
2. Execute one of following commands (there should be no file cert.pem in the current directory):

(for SSL server)
./examples/server/server -p 4433 -b -x -c cert.pem

(for DTLS server)
./examples/server/server -u -p 4433 -b -x -c cert.pem

3. Connect to SSL or DTLS server from another device

kaleb-himes, 2019-11-05:
Thank you for the report regarding our example application! We do want our examples to be well composed!

This would not be considered a vulnerability nor CVE worthy (@cve-reporting) if it does not effect the internal SSL/TLS or Cryptographic library. We are more than happy to fix a bug in our example applications though, keep up the good work!

- the wolfSSL team

kaleb-himes, 2019-11-05:
@cve-reporting,

Just so you are aware you are most welcome to report these via support@wolfssl.com so we have time to address the issue before it is public. It is most responsible to coordinate with an effected vendor before making details public to avoid 0-day exploits.

We appreciate all your hard work please keep the reports coming but please send them to support@wolfssl.com first if it is not too much to ask.

Thanks and Regards,

- the wolfSSL team

cve-reporting, 2019-11-07:
@kaleb-himes 
This issue has almost zero potential to cause any damage to the server, because it requires wrong certificate path to be provided by administrator, so I agree with You that this issue should not be reported to the CVE database.
However, example applications are very often reused in production ones, so I decided to report it.

kaleb-himes, 2019-11-07:
> However, example applications are very often reused in production ones, so I decided to report it.

A valid point! Again thanks for the report. PR is up to address: https://github.com/wolfSSL/wolfssl/pull/2572

kaleb-himes, 2019-11-08:
@cve-reporting,

PR https://github.com/wolfSSL/wolfssl/pull/2572 has been merged resolving this issue, please let us know if you find anything else and thank you again for the report!

P.S. We appreciate reports like this so much that we are happy to send SWAG (items with the wolfSSL logo, stickers, mugs, whatever our HR department has on hand) to research teams performing this valuable work. If your team would like to receive some SWAG for your efforts please contact us via support@wolfssl.com and we will be happy to connect you with our HR department to receive some export-friendly wolfSSL logo items for your effort!

Regards,

- Kaleb and the wolfSSL Team.

closed 2019-11-08 by kaleb-himes