wolfSSL/wolfssl #2570: cppcheck findings

chipitsine, 2019-11-06:
```
[IDE/IAR-EWARM/Projects/common/minimum-startup.c:42]: (error) syntax error: ".intvec" =
[IDE/LPCXPRESSO/wolf_example/src/lpc_18xx_startup.c:31] -> [IDE/LPCXPRESSO/wolf_example/src/lpc_18xx_startup.c:87] -> [IDE/LPCXPRESSO/wolf_example/src/lpc_18xx_startup.c:32] -> [IDE/LPCXPRESSO/wolf_example/src/lpc_18xx_startup.c:89] -> [IDE/LPCXPRESSO/wolf_example/src/lpc_18xx_startup.c:89]: (error) Comparing pointers that point to different objects
[IDE/zephyr/lib/user_settings.h:8]: (error) No header in #include
[src/internal.c:27338]: (error) Shifting signed 32-bit value by 31 bits is undefined behaviour
[wolfcrypt/src/fe_operations.c:159]: (error) Uninitialized variable: x1
[wolfcrypt/src/fe_operations.c:160]: (error) Uninitialized variable: x2
[wolfcrypt/src/fe_operations.c:161]: (error) Uninitialized variable: z2
[wolfcrypt/src/fe_operations.c:162]: (error) Uninitialized variable: x3
[wolfcrypt/src/fe_operations.c:163]: (error) Uninitialized variable: z3
[wolfcrypt/src/fe_operations.c:196]: (error) Uninitialized variable: tmp0
[wolfcrypt/src/fe_operations.c:196]: (error) Uninitialized variable: tmp1
[wolfcrypt/src/fe_operations.c:655]: (error) Uninitialized variable: t0
[wolfcrypt/src/fe_operations.c:656]: (error) Uninitialized variable: t1
[wolfcrypt/src/fe_operations.c:659]: (error) Uninitialized variable: t2
[wolfcrypt/src/fe_operations.c:665]: (error) Uninitialized variable: t3
[wolfcrypt/src/fe_operations.c:1271]: (error) Uninitialized variable: t0
[wolfcrypt/src/fe_operations.c:1272]: (error) Uninitialized variable: t1
[wolfcrypt/src/fe_operations.c:1281]: (error) Uninitialized variable: t2
[wolfcrypt/src/integer.c:2324]: (error) Uninitialized variable: redux
[wolfcrypt/src/integer.c:2352]: (error) Uninitialized variable: redux
[wolfcrypt/src/pkcs7.c:2638]: (error) Address of local auto-variable assigned to a function parameter.
[wolfcrypt/src/pkcs7.c:2750]: (error) Address of local auto-variable assigned to a function parameter.
[wolfcrypt/src/pkcs7.c:2851]: (error) Address of local auto-variable assigned to a function parameter.
[wolfcrypt/src/pkcs7.c:2990]: (error) Address of local auto-variable assigned to a function parameter.
[wolfcrypt/src/sp_arm32.c:22639]: (error) Shifting by a negative value is undefined behaviour
[wolfcrypt/src/sp_armthumb.c:12961]: (error) Shifting by a negative value is undefined behaviour
[wolfcrypt/src/sp_cortexm.c:13148]: (error) Shifting by a negative value is undefined behaviour
[wolfcrypt/src/wc_port.c:1313]: (error) Uninitialized variable: p
[wolfcrypt/src/wc_port.c:1407]: (error) Uninitialized variable: p
```

```
[IDE/Espressif/ESP-IDF/test/test_wolfssl.c:283]: (warning) %llu in format string (no. 1) requires 'unsigned long long' but the argument type is 'unsigned long'.
[IDE/Espressif/ESP-IDF/test/test_wolfssl.c:286]: (warning) %llu in format string (no. 1) requires 'unsigned long long' but the argument type is 'unsigned long'.
[IDE/Espressif/ESP-IDF/test/test_wolfssl.c:289]: (warning) %llu in format string (no. 1) requires 'unsigned long long' but the argument type is 'unsigned long'.
[IDE/GCC-ARM/Source/armtarget.c:145]: (warning) %lx in format string (no. 1) requires 'unsigned long' but the argument type is 'unsigned int'.
[IDE/GCC-ARM/Source/armtarget.c:146]: (warning) %lx in format string (no. 1) requires 'unsigned long' but the argument type is 'unsigned int'.
[IDE/GCC-ARM/Source/armtarget.c:147]: (warning) %lx in format string (no. 1) requires 'unsigned long' but the argument type is 'unsigned int'.
[IDE/GCC-ARM/Source/armtarget.c:148]: (warning) %lx in format string (no. 1) requires 'unsigned long' but the argument type is 'unsigned int'.
[IDE/GCC-ARM/Source/armtarget.c:149]: (warning) %lx in format string (no. 1) requires 'unsigned long' but the argument type is 'unsigned int'.
[IDE/GCC-ARM/Source/armtarget.c:150]: (warning) %lx in format string (no. 1) requires 'unsigned long' but the argument type is 'unsigned int'.
[IDE/GCC-ARM/Source/armtarget.c:151]: (warning) %lx in format string (no. 1) requires 'unsigned long' but the argument type is 'unsigned int'.
[IDE/GCC-ARM/Source/armtarget.c:152]: (warning) %lx in format string (no. 1) requires 'unsigned long' but the argument type is 'unsigned int'.
[IDE/GCC-ARM/Source/armtarget.c:153]: (warning) %lx in format string (no. 1) requires 'unsigned long' but the argument type is 'unsigned int'.
[IDE/GCC-ARM/Source/armtarget.c:154]: (warning) %lx in format string (no. 1) requires 'unsigned long' but the argument type is 'unsigned int'.
[IDE/GCC-ARM/Source/armtarget.c:155]: (warning) %lx in format string (no. 1) requires 'unsigned long' but the argument type is 'unsigned int'.
[IDE/GCC-ARM/Source/armtarget.c:156]: (warning) %lx in format string (no. 1) requires 'unsigned long' but the argument type is 'unsigned int'.
[IDE/GCC-ARM/Source/armtarget.c:157]: (warning) %lx in format string (no. 1) requires 'unsigned long' but the argument type is 'unsigned int'.
[IDE/GCC-ARM/Source/armtarget.c:158]: (warning) %lx in format string (no. 1) requires 'unsigned long' but the argument type is 'unsigned int'.
[IDE/IAR-EWARM/Projects/benchmark/current_time.c:47]: (warning) %d in format string (no. 1) requires 'int' but the argument type is 'unsigned int'.
[IDE/MDK-ARM/MDK-ARM/wolfSSL/shell.c:439] -> [IDE/MDK-ARM/MDK-ARM/wolfSSL/shell.c:440]: (warning) Either the condition 'args==NULL' is redundant or there is possible null pointer dereference: args.
[IDE/MDK-ARM/MDK-ARM/wolfSSL/wolfssl_MDK_ARM.c:72]: (warning) %d in format string (no. 1) requires 'int *' but the argument type is 'unsigned int *'.
[IDE/MDK-ARM/MDK-ARM/wolfSSL/wolfssl_MDK_ARM.c:72]: (warning) %d in format string (no. 2) requires 'int *' but the argument type is 'unsigned int *'.
[IDE/MDK-ARM/MDK-ARM/wolfSSL/wolfssl_MDK_ARM.c:72]: (warning) %d in format string (no. 3) requires 'int *' but the argument type is 'unsigned int *'.
[IDE/MDK-ARM/MDK-ARM/wolfSSL/wolfssl_MDK_ARM.c:72]: (warning) %d in format string (no. 4) requires 'int *' but the argument type is 'unsigned int *'.
[IDE/MDK5-ARM/Projects/wolfSSL-Full/shell.c:480] -> [IDE/MDK5-ARM/Projects/wolfSSL-Full/shell.c:481]: (warning) Either the condition 'args==NULL' is redundant or there is possible null pointer dereference: args.
[IDE/RISCV/SIFIVE-HIFIVE1/main.c:146]: (warning) %d in format string (no. 1) requires 'int' but the argument type is 'unsigned long'.
[IDE/RISCV/SIFIVE-HIFIVE1/main.c:151]: (warning) %d in format string (no. 1) requires 'int' but the argument type is 'signed long'.
[examples/benchmark/tls_bench.c:1282] -> [examples/benchmark/tls_bench.c:1258]: (warning) Either the condition 'if(args)' is redundant or there is possible null pointer dereference: args.
[examples/benchmark/tls_bench.c:1282] -> [examples/benchmark/tls_bench.c:1259]: (warning) Either the condition 'if(args)' is redundant or there is possible null pointer dereference: args.
[examples/client/client.c:318]: (warning) Assignment of function parameter has no effect outside the function. Did you forget dereferencing it?
[examples/client/client.c:319]: (warning) Assignment of function parameter has no effect outside the function. Did you forget dereferencing it?
[examples/server/server.c:997] -> [examples/server/server.c:1000] -> [examples/server/server.c:1961] -> [examples/server/server.c:1962]: (warning) Opposite inner 'if' condition leads to a dead code block.
[examples/server/server.c:888] -> [examples/server/server.c:889] -> [examples/server/server.c:1608]: (warning) Logical disjunction always evaluates to true: !(usePsk) || usePskPlus.
[examples/server/server.c:888] -> [examples/server/server.c:889] -> [examples/server/server.c:1649]: (warning) Logical disjunction always evaluates to true: !(usePsk) || usePskPlus.
[examples/server/server.c:1726]: (warning) Logical disjunction always evaluates to true: usePsk == 0 || usePskPlus.
[examples/server/server.c:888] -> [examples/server/server.c:889] -> [examples/server/server.c:1842]: (warning) Logical disjunction always evaluates to true: !(usePsk) || usePskPlus.
[examples/server/server.c:1842]: (warning) Logical conjunction always evaluates to false: !useAnon && loadCertKeyIntoSSLObj.
[examples/server/server.c:888] -> [examples/server/server.c:889] -> [examples/server/server.c:1854]: (warning) Logical disjunction always evaluates to true: !(usePsk) || usePskPlus.
[... 51 lines cut ...]
[... cut ...]
[wolfcrypt/src/tfm.c:2826]: (style) Condition 'cy' is always false
[wolfcrypt/src/wc_pkcs11.c:855] -> [wolfcrypt/src/wc_pkcs11.c:872]: (style) The expression 'ret == 0' is always true.
[wolfcrypt/src/wc_pkcs11.c:935] -> [wolfcrypt/src/wc_pkcs11.c:944]: (style) The expression 'ret == 0' is always true.
[wolfcrypt/test/test.c:14493]: (style) Condition 'openssl_aes_test()!=0' is always false
[wolfcrypt/test/test.c:24204] -> [wolfcrypt/test/test.c:24289]: (style) The expression 'ret >= 0' is always true.

```

tmael, 2019-11-08:
Hi @chipitsine, 

Thanks for sharing your findings! 

Could you please let us know how you generated the list of warnings with cppcheck? 
I run cppcheck against the master and I'm seeing some differences.  

Below is the command I used to run it in my Linux environment. I also tried '--enable=all' but I didn't see all of the warnings you reported. 

~/wolfssl$ cppcheck --enable=warning . 2>&1 | tee cppcheck.log.txt

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.2 LTS
Release:	18.04
Codename:	bionic

$ cppcheck --version
Cppcheck 1.82

chipitsine, 2019-11-09:
I'm running Fedora 31: `cppcheck-1.88-4.fc31.x86_64`

my cppcheck run is 15 commits behind current master, things might have changed.

here's how I run it:

script
cppcheck --force --enable=all .
Ctrl-D
grep '(error' typescript
grep '(warning' typescript
grep '(style' typescript

tmael, 2019-11-11:
@chipitsine, Thanks for providing the info on your environment and how you run cppcheck! 
It now makes sense why I'm seeing some differences. You are correct that things have changed since we have been fixing some cppcheck related issues via #2554 #2518 and #2538.  Will work on fixing the remaining issues where it makes sense.

PR #2748 (2020-02-05): Fix cppcheck
This PR fixes errors and warnings found with cppcheck.

$ cd ~/wolfssl
$ cppcheck --force --enable=all . 2>&1 | tee cppcheck.enable-all.txt
$ grep error cppcheck.enable-all.txt
$ grep warning cppcheck.enable-all.txt

Fixes #2570
closed 2020-02-05 by JacobBarthelmeh