wolfSSL/wolfssl #2574: Potential undefined behavior

ManSoSec, 2019-11-08:
Hi

It seems null checking is missed here:

https://github.com/wolfSSL/wolfssl/blob/0fe5d40507e42945d5bc9aefb653ebb0b9f32cf3/wolfcrypt/src/random.c#L370-L377

While it is done here:

https://github.com/wolfSSL/wolfssl/blob/0fe5d40507e42945d5bc9aefb653ebb0b9f32cf3/wolfcrypt/src/random.c#L500-L510

kaleb-himes, 2019-11-08:
@ManSoSec,

Thank you for the report, we will review and provide feedback soon!

- K

kaleb-himes, 2019-11-08:
After a quick code review:

Case 1) Line 500-510 where NULL sanity check is performed in ```Hash_gen()```.
Call Stack:

```
wc_RNG_GenerateBlock(WC_RNG* rng, byte* output, word32 sz)
Invokes with params:
Hash_DRBG_Generate(rng->drbg, output, sz)

Hash_DRBG_Generate(DRBG* drbg, byte* out, word32 outSz)
Invokes with params:
Hash_gen(drbg, out, outSz, drbg->V)

Hash_gen(DRBG* drbg, byte* out, word32 outSz, const byte* V)
Sanity Check performed on line 500 on user-provided buffer
```

Path summary: external applications may invoke non-opaque API wc_RNG_GenerateBlock() passing in an output buffer to store the result and a sz of desired block to be filled. The output buffer may or may not be NULL and the output sz may or may not be zero depending on user-application and is subject to user input error. We want to sanity check in this case that buffer is not null and sz is not zero.

Case 2) Line 370 - 377 where NULL sanity check is NOT performed in ```Hash_df()```.
Call Stack:

```
Hash_DRBG_Reseed(DRBG* drbg, const byte* seed, word32 seedSz)
Invokes with params:
Hash_df(drbg, newV, sizeof(newV), drbgReseed, drbg->V, sizeof(drbg->V), seed, seedSz)
Note newV is a static buffer local to wolfSSL internal and can never be NULL.
See declaration on line 393 in Hash_DRBG_Reseed().

Determination: Sanity check not necessary in this call stack.
```

Alternate call stack:

```
wc_RNG_DRBG_Reseed(WC_RNG* rng, const byte* seed, word32 seedSz)
Invokes with params:
Hash_DRBG_Reseed(rng->drbg, seed, seedSz)
NOTE: rng struct sanity check on line 419, rng->drbg will not be null if rng null check passes.

Hash_DRBG_Reseed(DRBG* drbg, const byte* seed, word32 seedSz)
Invokes with params:
Hash_df(drbg, drbg->C, sizeof(drbg->C), drbgInitC, drbg->V, sizeof(drbg->V), NULL, 0)

Determination: drbg->C and sizeof(drbg->C) will always be valid if at this point since RNG structure was already checked in wc_RNG_DRBG_Reseed, another check is not necessary.
```

There is a third call stack involving wc_RNG_GenerateBlock() which again passes the rng->drbg, again the rng is checked for validity first and in that case also a redundant check in Hash_df would not be necessary.

All that being said, I will present your report to the team in our next technical review, additional sanity checks are never a bad thing and it would help to guard against future additions or code paths that may end up passing in end-user application variables.

I'll send any notes following our meeting next week on Tuesday.

Cheers,

K

kaleb-himes, 2019-12-18:
Hi @ManSoSec,

I sincerely apologize, I forgot to send the followup. Our team went over this and concluded the checks were all properly in place higher in the call stack and there was no path to code section in question without prior sanity checks. No action was taken on this.

The wolfSSL team and I would like to extend our gratitude for your efforts in reviewing our code and our best wishes to you this holiday season. Thank you for your evaluation of our product, let us know if you find any other potential issues that should be reviewed!

Warmest Regards,

K

closed 2019-12-18 by kaleb-himes

neheb, 2020-12-29:
Does this issue explain https://github.com/transmission/transmission/issues/1537 ?