wolfSSL/wolfssl #2576: Missing Check

ManSoSec, 2019-11-08:
StoreECC_DSA_Sig might return BUFFER_E -132 while there is no check on the return value when it is called:
https://github.com/wolfSSL/wolfssl/blob/e329431bc114142178275b1f758a20b71d3ae239/wolfcrypt/src/ecc.c#L4564

No the same, but similar cases have been checked here:
https://github.com/wolfSSL/wolfssl/blob/e329431bc114142178275b1f758a20b71d3ae239/wolfcrypt/src/ecc.c#L5473-L5474

Thanks!

kaleb-himes, 2019-11-12:
@ManSoSec,

The return value is checked immediately upon exiting the switch statement. If you're tool is concerned about R and S they are checked prior and calling mp_clear on them before checking err is fine. Also BUFFER_E is in relation to outLen being less than the SUM OF: determined size of r (```mp_unsigned_bin_size(r);```) plus the determined size of s (```mp_unsigned_bin_size(s);```) plus the header sz + 2 additional bytes + the leading zero bytes on both r and s.

There is no issue with this one.

```
            err = StoreECC_DSA_Sig(out, outlen, r, s);

            /* done with R/S */
            mp_clear(r);
            mp_clear(s);
            break;

        default:
            err = BAD_STATE_E;
            break;
    }

    /* if async pending then return and skip done cleanup below */
    if (err == WC_PENDING_E) { // <----- Return checked upon exiting switch statement
        key->state++;
        return err;
    }
```

- K

closed 2019-11-12 by kaleb-himes

SparkiDev, 2019-11-12:
@ManSoSec 
Note that outlen comes from outside wc_ecc_sign_hash_async and the error must be and is returned back to the caller.

Sean