wolfSSL/wolfssl #2579: Potential leak

ManSoSec, 2019-11-08:
Digest is not zeroed out in InitSha224

https://github.com/wolfSSL/wolfssl/blob/e329431bc114142178275b1f758a20b71d3ae239/wolfcrypt/src/sha256.c#L1220-L1231

While it is done here in InitSha256:

https://github.com/wolfSSL/wolfssl/blob/e329431bc114142178275b1f758a20b71d3ae239/wolfcrypt/src/sha256.c#L173-L184

kaleb-himes, 2019-11-10:
Note 1) Not zeroing on initialization would not lead to leakage, maybe an uninitialized variable warning in some compilers.

Note 2) Not zeroing on over-writing (as in re-seeding) would not lead to leakage, over-writing with zeroes or with new bytes of entropy, or both? Double-write seems unnecessary.

Will review with the team.

ejohnstown, 2019-11-12:
SHA-224 vs SHA-256: The digests are getting initialized to constants. They don't need to be zeroed out before being set. Why is SHA-256 doing it? Who knows. It is redundant and doesn't hurt or help.

Hash_DRBG Instantiate vs Reseed: We do want to zero out the whole DRBG structure on instantiation. Not all the members are set to zero on instantiate. The XMEMSET might be redundant, or setting lastBlock and matchCount are. We wouldn't on Reseed because part of the DRBG's initial state is always used and we don't want to break reseeding by zeroing it out.

ejohnstown, 2019-11-27:
Are we in agreement on these details? May I close this issue?

closed 2019-12-03 by ManSoSec