wolfSSL/wolfssl #2580: WolfSSL /w hardware acceleration cannot be used in multithreaded environment

roel0, 2019-11-10:
"wolfSSL (formerly CyaSSL) is thread safe by design"

It might be thread safe, but when you enable hardware acceleration it becomes unusable. (Ex. STM32F2_CRYPTO)

This leads to ex. bad_record_mac when two threads are doing some decrypting/encrypting at the same time.

roel0, 2019-11-10:
I think the fault lies in the fact that the __HAL_LOCK API from the STM32 HAL library is misunderstood by the developer who added the STM32 support to wolfSSL

`     #define __HAL_LOCK(__HANDLE__)                                           \                                                         
                                   do{                                        \     
                                       if((__HANDLE__)->Lock == HAL_LOCKED)   \     
                                       {                                      \     
                                          return HAL_BUSY;                    \     
                                       }                                      \     
                                       else                                   \     
                                       {                                      \     
                                          (__HANDLE__)->Lock = HAL_LOCKED;    \     
                                       }                                      \     
                                     }while (0U)     
`

kaleb-himes, 2019-11-10:
@roel0,

Thank you for your notes, we will review and get back to you soon with any notes.

Warmest Regards,

- K

dgarske, 2019-11-13:
HI @roel0,

Thanks for reporting the threading mutex issue with STM32. I've enabled our hardware mutex protection around the STM32 hardware calls. See PR #2586. Looking forward to your feedback on it.

Thanks,
David Garske, wolfSSL

dgarske, 2019-11-18:
HI @roel0,

The changes to support mutex locking on the STM32 hardware have finished testing on our side.  See PR #2586.

Thanks,
David Garske, wolfSSL

PR #2586 (2019-11-18): STM32 Crypto hardware fixes and improvements
* Make the `wc_Stm32_Aes_Init` API always available for STM32. ZD 9503
* Fix for SHA256 hash context to only be included for `STM32_HASH_SHA2`. ZD 9503
* Add mutex protection on STM32 RNG, AES/DES and Hashing crypto hardware calls for thread safety. Fixes #2580
* Fix for AES GCM Decrypt auth tag. ZD 9507 (new issue starting https://github.com/wolfSSL/wolfssl/pull/2375)
* Fix for AEC CTR mode with STM32 and CubeMX (new issue starting https://github.com/wolfSSL/wolfssl/pull/2375)
* Cleanup for invalid use of `NO_SHA512` in api.c unit test.
* Improveme [... cut]
commit 27d95d1d (2019-11-18): Add mutex protection on STM32 RNG, AES/DES and Hashing crypto hardware calls for thread safety. Fixes #2580
closed 2019-11-18 by toddouska