wolfSSL/wolfssl #2590: Several instances of undefined behavior

guidovranken, 2019-11-14:
On these two lines:

https://github.com/wolfSSL/wolfssl/blob/22042830d64e05a8916184bad4980b906432eacc/src/internal.c#L12105
https://github.com/wolfSSL/wolfssl/blob/22042830d64e05a8916184bad4980b906432eacc/src/internal.c#L12212

```newDiff``` can be 0, so ```1 << (newDiff - 1)``` resolves to ```(int)1 << 4294967295```, which is undefined behavior (left shift of signed integer too many places).

On this line:

https://github.com/wolfSSL/wolfssl/blob/22042830d64e05a8916184bad4980b906432eacc/src/internal.c#L12234

the expression ```oldWindow[i-idx] >> (DTLS_WORD_BITS - newDiff)``` causes a 32 bit right shift on a 32 bit unsigned integer if ```newDiff``` is 0, which is also undefined behavior, see: https://stackoverflow.com/questions/18918256/is-right-shift-undefined-behavior-if-the-count-is-larger-than-the-width-of-the-t

One more potential issue is the following:

https://github.com/wolfSSL/wolfssl/blob/22042830d64e05a8916184bad4980b906432eacc/wolfcrypt/src/tfm.c#L3184

Using the TLS server, I've come across this warning by Undefined Behavior sanitizer:

```
wolfcrypt/src/tfm.c:3184:18: runtime error: index -1 out of bounds for type 'fp_digit [136]'
```

guidovranken, 2019-11-19:
Also:

```
runtime error: shift exponent 256 is too large for 32-bit type 'int'
```
at https://github.com/wolfSSL/wolfssl/blob/1a3455110e0c658ce1aa95993f1d7305c5345fd9/src/tls.c#L4271

ejohnstown, 2019-11-27:
The DTLS and the TFM issues should be fixed with PR #2610. Sean fixed the curve issue in PR #2604.

closed 2019-12-05 by ejohnstown