wolfSSL/wolfssl #2600: Problems building against wolfSSL v4.2.0-stable in Visual Studio

jay, 2019-11-18:
I built wolfSSL v4.2.0-stable using a [Windows build script that we offer users as part of curl](https://github.com/curl/curl/blob/821171d/projects/build-wolfssl.bat) to build wolfSSL's Visual Studio solution and prepare the file locations to then build [curl's Visual Studio solutions](https://github.com/curl/curl/blob/821171d/projects/), which are basically Visual Studio projects to build curl in some common basic configurations, as an alternative to cmake or other build systems. I tried to build one of the wolfSSL configurations 'DLL Debug - DLL wolfSSL' but I got an error `bad math long / long long settings` from wolfcrypt/types.h. It looks like the header expects SIZEOF_LONG_LONG for CheckRunTimeSettings:

https://github.com/wolfSSL/wolfssl/blob/48c4b2fedcee4d61b6a76c5ce9e33ab212d6ab4a/wolfssl/wolfcrypt/types.h#L736-L757

Earlier in the file there's an attempt to set SIZEOF_LONG_LONG if it is not set but it excludes MSVC:

https://github.com/wolfSSL/wolfssl/blob/48c4b2fedcee4d61b6a76c5ce9e33ab212d6ab4a/wolfssl/wolfcrypt/types.h#L60-L75

So I guess SIZEOF_LONG_LONG is defined somewhere during wolfSSL build process but not when building a different library that uses its headers. In Visual Studio type long has always been 4 bytes, and long long (__int64) has always been 8 bytes. That's not to say they won't change. Barring a true runtime sizeof(long) type check, a more cautious approach could be:

~~~diff
diff --git a/wolfssl/wolfcrypt/types.h b/wolfssl/wolfcrypt/types.h
index 7c8cc1d..92e0742 100644
--- a/wolfssl/wolfcrypt/types.h
+++ b/wolfssl/wolfcrypt/types.h
@@ -58,7 +58,19 @@
 
 
     /* try to set SIZEOF_LONG or LONG_LONG if user didn't */
-    #if !defined(_MSC_VER) && !defined(__BCPLUSPLUS__) && !defined(__EMSCRIPTEN__)
+    #ifdef _MSC_VER
+        #include <limits.h>
+        #ifdef ULONG_MAX
+            #if (ULONG_MAX == 0xffffffffUL)
+                #define SIZEOF_LONG 4
+            #endif
+        #endif
+        #ifdef ULLONG_MAX
+            #if (ULLONG_MAX == 0xffffffffffffffffui64)
+                #define SIZEOF_LONG_LONG 8
+            #endif
+        #endif
+    #elif !defined(__BCPLUSPLUS__) && !defined(__EMSCRIPTEN__)
         #if !defined(SIZEOF_LONG_LONG) && !defined(SIZEOF_LONG)
             #if (defined(__alpha__) || defined(__ia64__) || \
                 defined(_ARCH_PPC64) || defined(__mips64) || \
~~~

---

Second issue is I get a lot of "C4214: nonstandard extension used" warnings from the wolfSSL headers due to bitfield types, for example:
~~~
X:\j\curl\wolfssl\wolfssl/wolfcrypt/asn_public.h(193): warning C4214: nonstandard extension used : bit field types other than int
X:\j\curl\wolfssl\wolfssl/ssl.h(477): warning C4214: nonstandard extension used : bit field types other than int
X:\j\curl\wolfssl\wolfssl/ssl.h(478): warning C4214: nonstandard extension used : bit field types other than int
X:\j\curl\wolfssl\wolfssl/wolfcrypt/asn.h(886): warning C4214: nonstandard extension used : bit field types other than int
X:\j\curl\wolfssl\wolfssl/wolfcrypt/asn.h(887): warning C4214: nonstandard extension used : bit field types other than int
...
~~~

We recently faced a similar situation in libcurl and @bagder fixed it with `typedef bool bit;` for MSVC and then use bit, which works in curl because for pre-c99 [bool is an int](https://github.com/curl/curl/blob/curl-7_67_0/lib/curl_setup_once.h#L291-L313). This appears to be a nuisance warning and the build is working ok so far. (curl 7.67.1-DEV (i386-pc-win32) libcurl/7.67.1-DEV wolfSSL/4.2.0 nghttp2/1.40.0).

kaleb-himes, 2020-01-14:
Hi Jay,

So sorry it's been this long without an update, we had a lot going the end of 2019. I will try to get time to test these suggested changes in the next day or two and let you know the results. If all checks out would you prefer I open the pull request (PR) on your behalf or would you want to open the PR yourself?

Warm Regards,

K

jay, 2020-01-15:
No problem Kaleb, whatever is easier.

PR #2747 (2020-01-17): Improvements for detection of 64-bit support
* Adds support for IBM s390x. Fixes #2745.
* Improves detection on Windows. Fixes #2600
* Adds check for `HAVE_LIMITS_H` to use limits.h to detect sizeof long.
commit 541bf3e6 (2020-01-17): Improvements for detection of 64-bit support. Adds support for IBM s390x. Improves detection on Windows. Adds new `WC_USE_LIMITS_FOR_SIZEOF` option to use limits.h to detect sizeof long. Fixes #2600 and Fixes #2745.
closed 2020-01-17 by JacobBarthelmeh