wolfSSL/wolfssl #2616: wolfssl 4.2.0  make check not pass : ./build-aux/test-driver: line 107: ./testsuite/testsuite.test: cannot execute binary file

YuanDaYu002, 2019-11-21:
=====================================
   wolfssl 4.2.0: ./test-suite.log
=====================================

# TOTAL: 7
# PASS:  5
# SKIP:  0
# XFAIL: 0
# FAIL:  2
# XPASS: 0
# ERROR: 0

.. contents:: :depth: 2

FAIL: testsuite/testsuite
=========================

./build-aux/test-driver: line 107: ./testsuite/testsuite.test: cannot execute binary file
FAIL testsuite/testsuite.test (exit status: 126)

FAIL: tests/unit
================

./build-aux/test-driver: line 107: ./tests/unit.test: cannot execute binary file
FAIL tests/unit.test (exit status: 126)

YuanDaYu002, 2019-11-21:
My compile parameters are as follows:

./configure --host=arm-himix100-linux --prefix=/home/yuanda.yu/svndebug/Hi3518E_V300_LiteOS/3rdlibs_src_code/wolfssl-4.2.0-stable/out --enable-static --disable-shared --enable-32bit --enable-opensslall   --disable-asm   --enable-aeskeywrap --enable-dtls --enable-debug

then：
make

ejohnstown, 2019-11-27:
Are you running configure, make, and make check on the same machine you are expecting to run your executable on? Is the build machine an arm-himix100-linux box, or a regular desktop PC running linux?

ejohnstown, 2019-12-05:
I'm going to assume that was your issue, and consider this issue closed. If you have any other questions regarding this, you may reopen the ticket.

closed 2019-12-05 by ejohnstown