wolfSSL/wolfssl #2639: Specific set of config options leads to build failure

guidovranken, 2019-12-04:
```
./configure --enable-static --enable-tls13 --enable-ocsp --enable-dtls --enable-sni --enable-blake2 --enable-blake2s --enable-curve25519 --enable-session-ticket --enable-nullcipher --enable-crl --enable-ed25519 --enable-psk --enable-earlydata --enable-postauth --enable-hrrcookie --enable-opensslextra --enable-certext --disable-examples --enable-tlsx --enable-oldtls --enable-tlsv10 --disable-crypttests --enable-indef --enable-psk
```

leads to build error:

```
src/.libs/src_libwolfssl_la-ssl.o: In function `wolfSSL_OBJ_txt2nid':
/src/fuzzers/wolfssl/wolfssl/src/ssl.c:37155: undefined reference to `EncodePolicyOID'
/usr/bin/ld: src/.libs/libwolfssl.so.23.0.0: hidden symbol `EncodePolicyOID' isn't defined
/usr/bin/ld: final link failed: Bad value
clang-10: error: linker command failed with exit code 1 (use -v to see invocation)
```

ejohnstown, 2019-12-04:
I'm going to add a fix for this to pull request #2624.

guidovranken, 2019-12-04:
Thanks! Your commit fixes it.

closed 2019-12-04 by ejohnstown