wolfSSL/wolfssl #2644: WolfSSL client/server behaviour in case of NAT changes port number in the middle og the session

AButenko, 2019-12-05:
Version: wolfssl-3.13.0-stable

Hi all,

I have WolfSSL client connected to some DTLS server with NAT in between. If something happens with NAT - it can change the client port number (which is the only way to identify current DTLS session). 
What should be server and client behaviour in such case?

In my implementation:
At the moment server sends DTLS fatal alert "Decrypt Error" and this is reasonable except one thing - server already lost all the context about the session and DTLS alert is sent with Epoch=0, Sequence number=0.
On the other hand, WolfSSL client receives that kind of DTLS Alert and silently drops the packet (it has the same  Epoch=0, Sequence number=0 as fist "Hello Verify Request" sent by server):

My view there should be some setting/API/etc. to get notification about such kind of DTLS alert on application side (WolfSSL user). Can someone, please, help me to find such kind of setting/API/etc?

ejohnstown, 2019-12-05:
We have an option to turn on the tracking of some statistics for DTLS packet dropping. We will count replay drops and MAC failure drops. The totals can be polled on the session. Adding `-DWOLFSSL_DTLS_DROP_STATS` to the CPPFLAGS for configure will enable that. The function `wolfSSL_dtls_get_drop_stats()` will give you these statistics.

AButenko, 2019-12-05:
Thank you for the response!

But is there any way to identify was it an DTLS alert dropped or real duplicated message?
Can you imagine any other (better) way to handle such kind of situation?

Thanks

ejohnstown, 2019-12-05:
Is there a way to identify it as an alert instead of a duplicated message? You can write your own receive callback and look at the data yourself.

A better way? I don't know. The threat model used in the design of DTLS was that it was trivial to send forged UDP datagrams and that a UDP attack can look like normal behavior.

What does your application protocol do when your client doesn't hear from the server for a period of time?

AButenko, 2019-12-05:
Application protocol retries up to 16 times and then makes 300s pause afterwards.

But using receive callback is a good idea I obviously missed. DTLS Alert message is not encrypted and I can easily parse it on application side. 

Thank you very much for your help. I think, the solution is to use  receive callback and I can close this question.

closed 2019-12-05 by AButenko