wolfSSL/wolfssl #2657: wolfssl 4.0.0 accepts a certificate with an invalid time format

llqll, 2019-12-10:
I recently created a certificate chain [rootCA,intermediate certificate,leaf certificate], which leaf certificate has an invalid notbefore fields with  " #0101010I00Z" . Clearly, the string include non-digits. Meanwhile, the chain can still pass certificate verification with wolfssl-4.0.0.0 version.

Does wolfssl have a bug here?
(Or do I have some misunderstandings on wolfssl in its parsing or
verification procedure?) Will it cause any further problems in
certificate verification?

The code I used is(wolfssl.c):
```
#include <wolfssl/options.h>
#include <wolfssl/ssl.h>
#include <wolfssl/wolfcrypt/error-crypt.h>
int wolfssl_verify_cert(char * cert_file,char *ca_file)
{
    int ret = 0;
    WOLFSSL_CERT_MANAGER* cm;
    cm = wolfSSL_CertManagerNew();//new a certmanager to verify certificate
    if(cm == NULL) {
        printf("error creating new cert manager\n");
    }
    ret = wolfSSL_CertManagerLoadCA(cm,ca_file, NULL);
    if (ret != SSL_SUCCESS) {
        printf ("error loading CA certs into cert manager\n");
    }
    ret = wolfSSL_CertManagerVerify(cm, cert_file,SSL_FILETYPE_PEM);
    if (ret != SSL_SUCCESS) {
       int a=ret;
        wolfSSL_ERR_reason_error_string(-144);
        printf("wolfSSL_CertManagerVerify() failed (%d): %s\n",ret, 
        wolfSSL_ERR_reason_error_string(ret));
    }
    else printf("Verification Successful!\n");
    wolfSSL_CertManagerFree(cm);
    return ret;
}
int main(int argc,char **argv){
   char * cert_file=argv[1];
   char *ca_file=argv[2];
   wolfssl_verify_cert(cert_file,ca_file);
   return 0;
}
```
The command is :
```
gcc -g wolfssl.c -o wolfssl -lwolfssl
wolfssl   leaf.pem  1.pem
```
The verification returns
```
Verification Successful!
```
1.pem (it contains two certificates inside):
```
-----BEGIN CERTIFICATE-----
MIIGCDCCA/CgAwIBAgIQY8Mi35RmHbQSpWR8XD7V+zANBgkqhkiG9w0BAQsFADBt
MQswCQYDVQQGEwJDTjEMMAoGA1UECAwDVEo1MQwwCgYDVQQKDANUSlUxFDASBgNV
BAsMC2JlaXlhbmd5dWFuMQswCQYDVQQDDAJDUzEfMB0GCSqGSIb3DQEJARYQbGpm
cG93ZXJAMTYzLmNvbTAgFw0wMDAxMDEwMTAwMDFaGA82NTY2MDMyMzEyMTIzM1ow
ajELMAkGA1UEBhMCQ04xDDAKBgNVBAgMA1RKNTEMMAoGA1UECgwDVEpVMRQwEgYD
VQQLDAtiZWl5YW5neXVhbjENMAsGA1UEAwwEYjMyNjEaMBgGCSqGSIb3DQEJARYL
bGkxQDE2My5jb20wggIiMA0GCSqGSIb3DQEBAQUAA4ICDwAwggIKAoICAQDUWAVE
VHGqn3tPc+kJTGwXpsiD+pwu287ibcwa7nlcQ8KyrwbS/7dnhK3Mpz3jjkbk9Zqw
Ju8R5ku9hEsSX3ZW7KQYj+jqVWVnLNlp5j0a1G2fdB7vn0ORtj9GgFAbKn37cXqo
6G2EyQ0NXhpOiwUtQXSnhbMUUJal2jMSaSGSKyyex9lDrZfSzQ164VIvMKz49kPB
Z6EupA0E6QkwZ1a8wGthdhQ3tJrHt0jcmBVpJ5mo9zlvX7ErsK4prXgJvBQR/IRc
YhqYHxsKLq/mgjezNqy/WoPN313HxDG8YETy8m9BKWI5OLBHIr0kahmBFumttlGa
a4rW+w2NZz8jtrnkM8sFSEoegO7xA8JZdO6O3mSedWOiA2zEuT8hQqkSYDSdZxOd
J1u/mdyumLErXquenaMTAHb0lviNc7llZqDKMJ8yfROZwv9PDCs3OBGOttr3MMRT
JHN5f4ZStqx6unV90Rx8QIh8wstG3c/QrJ4lBS+c72A6bMmxLpiTg1+CjG9ntgvC
mspMbVlu710Y7JHcAuq9RSnR0Nv31AGjOZEpKAGpUfzoVf47GYV38VpLskgy0tiA
Tesse5g8rUE9ozwgj6B34qfNdPxCmv6UkLYxU/CLpw2cRKT8hShAO8zDfgmU9262
ctTdrVU3PsSwMs7F8SlG/9kWq6HgqaBPadCsRwIDAQABo4GkMIGhMB0GA1UdDgQW
BBSSPopRSpZMfPAxCvUPCu4TZmh38DAfBgNVHSMEGDAWgBRyFaB24RFh9c9zf0+D
YA01twtiWjASBgNVHRMBAf8ECDAGAQH/AgEAMA4GA1UdDwEB/wQEAwIBhjA7BgNV
HREENDAyggdhYmMuY29tggkqLmFiYy5jb22CB3h5ei5jb22CDXd3dy5iYWlkdS5j
b22HBH8AAAEwDQYJKoZIhvcNAQELBQADggIBAJwtzZT7z1eImP8a7GTnfbPYu8k4
kdbGnWSyrEr8x6UjZQLCa1DXdxKkms84yCW1QM5vdKody/Sz1lvETPeTgpXRLlcO
i/75L+Knz1asfz3D+SO/YCSc/VF27GnkKyjFlt7LUmHuFUQoprpCi12wJ0IJP5D6
AQarnWuS2AA4op0exLrK1+BonYyqH//QDt5jhUJFEKQVgckHOtVOklHmazplr8bu
JzHz0+C7mDtZbLXoBSgZIFaVCSk4uxsf98QWOxKQURUv8gAhHLOo/QlkyqiiFCaN
1Se0Zp16pegTxs0qS8qY1pLgw4AO56ifG+LcOmYminbAZtApmiOvtxf8JAw5Twc8
6gLRlq2cv/bY55hZde4uvUzC/Te/zENu9rlv7qQqQ9jS5tiWZjZVqhEt275KymBT
4855pB+8oGb5Xznl6/AzmxUbOmRX1q5bbv+11ZscRtUp3XD3gA5Y5UYBF5UVICcb
zTVUNDgaUjyuXIiF/ZFtbcxX57PfIqKHP3A2XseUhpN3qFSWb29BsTAa7E59s8pL
0m/aftSXF1g/8q0IsHFuZRv4l+eyYWJhwtQTY9TTHnjYJbljcwGtVjYuAfMB+eec
beH0LdKLVbOKlMPySiqy18cKDkwQ1wTPqoZnz5/mKRr5Hpt/RKSe997NjIeuJZl0
W0ebRMo2T0FNhUhm
-----END CERTIFICATE-----
-----BEGIN CERTIFICATE-----
MIIGCDCCA/CgAwIBAgIQY8Mi35RmHbQSpWR8XD7V+jANBgkqhkiG9w0BAQsFADBt
MQswCQYDVQQGEwJDTjEMMAoGA1UECAwDVEo1MQwwCgYDVQQKDANUSlUxFDASBgNV
BAsMC2JlaXlhbmd5dWFuMQswCQYDVQQDDAJDUzEfMB0GCSqGSIb3DQEJARYQbGpm
cG93ZXJAMTYzLmNvbTAgFw0wMDAxMDEwMTAwMDBaGA85OTk5MTIyMzExMjMzNFow
bTELMAkGA1UEBhMCQ04xDDAKBgNVBAgMA1RKNTEMMAoGA1UECgwDVEpVMRQwEgYD
VQQLDAtiZWl5YW5neXVhbjELMAkGA1UEAwwCQ1MxHzAdBgkqhkiG9w0BCQEWEGxq
ZnBvd2VyQDE2My5jb20wggIiMA0GCSqGSIb3DQEBAQUAA4ICDwAwggIKAoICAQC+
WcvCnpCA78zG1ZkRhiIPjPEmFx3PHaX5f+KYod68qvCqsRGsB4n7rQS2ljFUZ7MY
4GNWtiMZANdWMuOrnkT0sNmtQ1aXWh+6lMUKLr/690SkKMbKU1y6OTfGBntau6em
1djv9Q8fYmapdne3tr5UNTJBvqc5qivWiF98XUQdp8qGKLYfF0NOxkreD6u4Pddo
/6PR5pn+nbgCHkDFmVGL+0DtZzC+K/NQbKpmP4/Zpolf1C5wPpxWPpjDl/yRSctC
qX1G0WGyB8/w/IR94Gx3rDmA/NkZMP+4tXBFVSoz0XJpdNqCtwxCkl6NqLpMN0gp
XrU78ToNnTiUW4zoyIfKBSlXRkPd4srgB8gTO3cHqJkSmzt/gFMnbBP1gNV10R0P
KzbNuV/uIHx5wGYJIW8w9fL8hKrCYcO5Yfq3VDGy9Lr3/5QFYI36oPLIw0cZS/i+
NyPLYT1TN/o6E8dtnsz1AY+VQyriW44CB6J3tlfrGLigfP81rsaQpcGd+W+0ntyc
cWpzRKwwut3I9CJSGjRuwHfz0n6Fk+Hoj+i+Qv6h/y7+KwqjDMMHIrbieBhUwQbm
Hlyj25IwyvYc6OOBymAyy8pUByAC7QWw4KxogDol6165iAubaupDxkDQXKr/IMmj
pCcTBDmVwhStVBDCD6Lo4HhxDE5a6IA4DSxdWIV2iQIDAQABo4GhMIGeMB0GA1Ud
DgQWBBRyFaB24RFh9c9zf0+DYA01twtiWjAfBgNVHSMEGDAWgBRyFaB24RFh9c9z
f0+DYA01twtiWjAPBgNVHRMBAf8EBTADAQH/MA4GA1UdDwEB/wQEAwIBhjA7BgNV
HREENDAyggdhYmMuY29tggkqLmFiYy5jb22CB3h5ei5jb22CDXd3dy5iYWlkdS5j
b22HBH8AAAEwDQYJKoZIhvcNAQELBQADggIBAFYRDs+WyMwr8rPCkzFHnMK0ePfD
cWc1O1L02foAePXEicrqQwv7JnsikBsx28E0T+mjqFU+7IIq7K+T0ndlEfax96Gi
j3H8zfwAG10JBFMjsFtdo8Hq6Q4CeMu1D83NPhQacZ1lOdCp/ZUdRvlcveeBx5VX
hFel6erfsR+6GX6I0b2Z9qIBKwmpxLcsPkY60RuazvkSf7xAd4eNJ18vzdo55J1c
x6mJK+c5J63a/IW6rjEd2v6URwwlbOyuRSurXoETMxYwuxs7pBnxA3MRU/OWIaCy
fAO+2ao4qn4WNo4oGo1BJBaX+mQJa+NwCw2F+sRqGZ+3ooSq2bjjXrLxiytr4b+o
fUBiCzhZLOGaRubJXlWp39dgLf6mo3ajjYPhTUtlqv0ZfX97C7xEXitNY3Dy9aqe
NnQn2+u2dkzEMTc+zW5i+xkByRhoSXY5AhYDdyd0Qtuk1T8sRs38TJmavr6/H6hv
6FGrmgqFypmsVy1LdRAn80yVBce1t3eWcgVnTND+wSS8mEj9rHS4th4sZbwwpVWJ
Z0cJSFnqSLMh7ZrDyzcKFUhgdU7GxuaACxIbBt3f5pCp1QDKffb3kVG333l/OLqN
2qYOTP6iFf3JpKttNvaSA9Q+GNk4t/8ozZW6lfyz+uDfmQecEgAv/u1s1brMgQo7
TQ/vJrJvgyxVSgOH
-----END CERTIFICATE-----
```
the leaf.pem:
```
-----BEGIN CERTIFICATE-----
MIIFQDCCAyigAwIBAgIRAPABuQ6DmexEq0k9QQaewMUwDQYJKoZIhvcNAQELBQAw
ajELMAkGA1UEBhMCQ04xDDAKBgNVBAgMA1RKNTEMMAoGA1UECgwDVEpVMRQwEgYD
VQQLDAtiZWl5YW5neXVhbjENMAsGA1UEAwwEYjMyNjEaMBgGCSqGSIb3DQEJARYL
bGkxQDE2My5jb20wHhcNICMwMTAxMDEwSTAwWhcNMzUxMjIzMTEyMzM0WjB7MQsw
CQYDVQQGEwJDTjEMMAoGA1UECAwDVEo1MQswCQYDVQQHDAJUSjEMMAoGA1UECgwD
VEpVMRQwEgYDVQQLDAtiZWl5YW5neXVhbjEMMAoGA1UEAwwDTExRMR8wHQYJKoZI
hvcNAQkBFhBsamZwb3dlckAxNjMuY29tMIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8A
MIIBCgKCAQEArXvIlHbQRwFvnLFz2dsnbPBgE8WIDRpIIpRTWJL+pdi/duUcE5Xn
VRNA0lnlYBOl8igItyFudUC4o45xa0Q9Htd8hisbdaHpRpdTRUUpljH9rOOWOyY0
aqRJ0RrU2ayhJslTH9OBBg1ZaatMYxI2u8Bz1MJrtsCUcvymScT59QAYI17ZAzI5
ouqUsn3F5BgiU53kdm4ubfKts2su/sUvM9BN03+/p2o/50FanBVrRMHAUs2p65FM
yFtNwqT77ZpO9BZdEOV3KSRJRLbZbELoanMQ0txztznWI6PULTenf8eR24dQscqX
N38Qk+SGwp/lu/6qLN916oY2WFTRGrnCcwIDAQABo4HPMIHMMAkGA1UdEwQCMAAw
HQYDVR0OBBYEFI0Gz1ruYze8+EmA4MZ06BPU0AsiMB8GA1UdIwQYMBaAFJI+ilFK
lkx88DEK9Q8K7hNmaHfwMA4GA1UdDwEB/wQEAwIFoDAdBgNVHSUEFjAUBggrBgEF
BQcDAQYIKwYBBQUHAwQwKAYDVR0RBCEwH4IFYS5jb22CCCouYWMuY29tggZ4ei5j
b22HBH8AAAEwGAYDVR0SBBEwD4INYWJjZXd3cnd0LmNvbTAMBgNVHSQEBTADgAED
MA0GCSqGSIb3DQEBCwUAA4ICAQBvx+Z8r/YjdhvkV5XbnRan25H7afvfg3aFHDGW
q2WxNEKynxvdM9TEQtbQXJrWRj9sXXRohaYxObuAic+gYdTOrsYtk48gENG6GH4s
fxdX92XeWm2wUr0KXKOu+Mvtj/egk0bEMQloZe/tkjeOLAGzrJetXyGtxgIA+/XI
E/AyyNULHYFATZWx/XD0Q1s/VOZttPn6FG4qi5UogM/XqqCQbZ8C/DSj9RltQi02
IGmr+CCS4Y9ACHq3HT9YfSFMPV+7OwC/fegLadsd2Bk6TwAi+WNs/48M/LATAsnH
MxFV61T/qHuabPNfmlirpe/ooMWEIAoKKvxght4CztYRK5QZA01BBgqePur4wqAw
JeAp0M+bFWEDvt6xRiijmb261WRTM2C5mqnlQFJSdZ6h9MzerBph4Zvl3USXzEMb
hXPSeIIA7TwEWeH3whqP6w6NnmcS94jCgXnmvv9uInSc/CAKz5h2HElLQroP1Mmh
+KnKOAiSQrr0vyuGjZyxebu7E5RWWS//G2FJrG+2WyOAq0rml8HcjWtZu0I54xYq
rk0SKXpUBAvLbXky9rmAM5MinasHDBnUe7zTjlNuathI+5SPJ83PK/d+0HF6zzud
nvjqR+fWa4N/3jZ0DRquE1gEUWkK7jLegPalIZiLW064nfi6j2q6HP8eiyHarnA0
Mnwt2Q==
-----END CERTIFICATE-----
```

kaleb-himes, 2019-12-10:
@llqll,

Thank you so much for your report and for including a re-producible test case! We will review and if we have any questions we will let you know.

Warmest Regards,

K

SparkiDev, 2019-12-11:
@llqll,

We have looked into this issue and come up with a fix!
The pull request has gone up and is awaiting review:
https://github.com/wolfSSL/wolfssl/pull/2658

It will be merged into master soon, but you can try it out if you want.

Sean

kaleb-himes, 2019-12-11:
Thanks @SparkiDev!

kaleb-himes, 2020-01-08:
Fix merged, closing this out. Thank you again @llqll for the report!

Regards,

K

closed 2020-01-08 by kaleb-himes