wolfSSL/wolfssl #2659: wolfssl-4.0.0 accepts a certificate with authority key identifier extension field  that do not match issuer

llqll, 2019-12-11:
I recently created a certificate chain [rootCA,intermediate certificate,leaf certificate], where the key identifier field of the authority key identifier extension of leaf certificate is not the same as subject key identifier of issuer (intermediate certificate). Clearly, leaf certificate violate Section 4.2.1.2, RFC5280: “In conforming CA certificates, the value of the subject key identifier MUST be the value placed in the key identifier field of the authority key identifier extension (Section 4.2.1.1) of certificates issued by the subject of this certificate.”. Meanwhile, the chain can still pass certificate verification with wolfssl-4.0.0. however, the chain was rejected in openssl. 

the  key identifier field of subject key identifier of intermediate certificate is:
```
92:3E:8A:51:4A:96:4C:7C:F0:31:0A:F5:0F:0A:EE:13:66:68:77:F0
```
the  key identifier field of authority key identifier of leaf certificate is:
```
92:3E:8A:51:4A:96:6D:6F:F0:1D:0A:F5:0F:0A:EE:FF:66:68:77:F0
```
**Steps to reproduce**  
The code I used is(wolfssl.c):
```
#include <wolfssl/options.h>
#include <wolfssl/ssl.h>
#include <wolfssl/wolfcrypt/error-crypt.h>
int wolfssl_verify_cert(char * cert_file,char *ca_file)
{
    int ret = 0;
    WOLFSSL_CERT_MANAGER* cm;
    cm = wolfSSL_CertManagerNew();//new a certmanager to verify certificate
    if(cm == NULL) {
        printf("error creating new cert manager\n");
    }
    ret = wolfSSL_CertManagerLoadCA(cm,ca_file, NULL);
    if (ret != SSL_SUCCESS) {
        printf ("error loading CA certs into cert manager\n");
    }
    ret = wolfSSL_CertManagerVerify(cm, cert_file,SSL_FILETYPE_PEM);
    if (ret != SSL_SUCCESS) {
       int a=ret;
        wolfSSL_ERR_reason_error_string(-144);
        printf("wolfSSL_CertManagerVerify() failed (%d): %s\n",ret, 
        wolfSSL_ERR_reason_error_string(ret));
    }
    else printf("Verification Successful!\n");
    wolfSSL_CertManagerFree(cm);
    return ret;
}
int main(int argc,char **argv){
   char * cert_file=argv[1];
   char *ca_file=argv[2];
   wolfssl_verify_cert(cert_file,ca_file);
   return 0;
}
```
The command is :
```
gcc -g wolfssl.c -o wolfssl -lwolfssl
wolfssl   leaf.pem  1.pem
```
**Expected result**   
Verification failed
**Actual result**  
Verification Successful!

**Reason**  
Through debugging, I found that wolfssl looks up the CA certificate through the authority key identifier. If it doesn't find it, it will use the issuer field to find it again. In this case, In this case, if the issuer field matches the subject of the CA certificate, the verification result will be successful. 
However, openssl will perform more checks after finding CA certificate by issuer name.
the wolfssl code:
[image]
the openssl code:
[image]

**certificate** 
1.pem (it contains two certificates inside):
```
-----BEGIN CERTIFICATE-----
MIIGCDCCA/CgAwIBAgIQY8Mi35RmHbQSpWR8XD7V+zANBgkqhkiG9w0BAQsFADBt
MQswCQYDVQQGEwJDTjEMMAoGA1UECAwDVEo1MQwwCgYDVQQKDANUSlUxFDASBgNV
[... 80 lines cut ...]
03+/p2o/50FanBVrRMHAUs2p65FMyFtNwqT77ZpO9BZdEOV3KSRJRLbZbELoanMQ
0txztznWI6PULTenf8eR24dQscqXN38Qk+SGwp/lu/6qLN916oY2WFTRGrnCcwID
AQABo4IBwTCCAb0wCQYDVR0TBAIwADAdBgNVHQ4EFgQUjQbPWu5jN7z4SYDgxnTo
E9TQCyIwHwYDVR0jBBgwFoAUkj6KUUqWbW/wHQr1Dwru/2Zod/AwDgYDVR0PAQH/
BAQDAgWgMB0GA1UdJQQWMBQGCCsGAQUFBwMBBggrBgEFBQcDBDAoBgNVHREEITAf
ggVhLmNvbYIIKi5hYy5jb22CBnh6LmNvbYcEfwAAATAYBgNVHRIEETAPgg1hYmNl
d3dyd3QuY29tMDAGCCsGAQUFBwEBBCQwIjAgBggrBgEFBQcwAoYUaHR0cDovL215
LmNhL2NhLmh0bWwwgbAGA1UdIASBqDCBpTAFBgMqAwQwBgYELQYHCDCBkwYDKwUI
MIGLMCEGCCsGAQUFBwIBFhVodHRwOi8vbXkuaG9zdC5uYW1lLzswIQYIKwYBBQUH
AgEWFWh0dHA6Ly9teS55b3VyLm5hbWUvOzBDBggrBgEFBQcCAjA3MCEWEU9yZ2Fu
aXNhdGlvbiBOYW1lMAwCAQECAQICAQMCAQQaEkV4cGxpY2l0IFRleHQgSGVyZTAM
BgNVHSQEBTADgAEDMAoGA1UdNgQDAgECMA0GCSqGSIb3DQEBDQUAA4ICAQB4/6th
xs0tdxXEgxnY8kjU6+OzDTAc7oTJ1ok6Y8slE6zOoN/ZGclNrIfabE10uZKaKNLm
Y/GptkgjjUyv043/uKlLQWtJ9rc0ROzrfKtIN5UdYt5LfKfRRrZKLwDCC8lBFROn
kRl8J6Pggd0Y+kffF9SeDJSF2rbATUzxFCEHxztJhas1KZVjNMVU1kOk2R0b2L6j
yE6UCxgt8QMMTSPUvLaTAfw3G1xev9+Gga5r4RissX60+VDvolpkhObNs8MDIqSl
ll4o9LXUFHY8+a48pKXlcDrCSIWggJOaC3oGzkTMnSs8V4gQeTuA3eZJgiGSdSXc
apH1h7LWwSNIT3BedlZFpFiUyXvlZOWvd1zCqydO7RuME4pJ4d0GXH3h5tQCs3uE
7f2YiLZY46iWBLG5qSvtLddiaCx7h8SY5qLLMHIjADxckzPkL55FwfNJtOqNTDO1
psJGFXcAft8H/LqqqN0XxFCSr/QGlmsCJF9hNEg8K4L4Zx/hZdODxGLi7XrvETOu
tGIH/VaciBsaScXbAdVMLN3jpG/FcVaXHQM6ehLFbqQM+hdHO9g/e/wDgFKjSbv7
IvESPedVJj1fb/UdeXIhSWKdCUwi64ggCOdocXDEtbx0J1Sks3dtR9igWixYhCCp
q4dG931dO/VL42b1z0a1G9EBGvirI0vkTii20g==
-----END CERTIFICATE-----
```

SparkiDev, 2019-12-12:
@llqll 

The code has moved on from wolfSSL 4.0.0 and there is now a pull request up with a fix amongst others:
https://github.com/wolfSSL/wolfssl/pull/2661

The pull request has not yet been through complete testing or code review.

In the meantime, I have a patch for wolfSSL 4.0.0:

diff --git a/wolfcrypt/src/asn.c b/wolfcrypt/src/asn.c
index eef4035b9..162c2ef71 100644
--- a/wolfcrypt/src/asn.c
+++ b/wolfcrypt/src/asn.c
@@ -8088,8 +8088,13 @@ int ParseCertRelative(DecodedCert* cert, int type, int verify, void* cm)
                 cert->ca = GetCA(cm, cert->extAuthKeyId);
             if (cert->ca == NULL && cert->extSubjKeyIdSet)
                 cert->ca = GetCA(cm, cert->extSubjKeyId);
-            if (cert->ca == NULL)
+            if (cert->ca == NULL) {
                 cert->ca = GetCAByName(cm, cert->issuerHash);
+                if (cert->ca && cert->extAuthKeyIdSet) {
+                    WOLFSSL_MSG("CA SKID doesn't match AKID");
+                    cert->ca = NULL;
+                }
+            }

             /* OCSP Only: alt lookup using subject and pub key w/o sig check */
         #ifdef WOLFSSL_NO_TRUSTED_CERTS_VERIFY

Let me know if you are happy with this.

Sean

llqll, 2019-12-12:
I am very happy, thank you.

SparkiDev, 2019-12-12:
Your welcome :-)

Sean

closed 2019-12-12 by SparkiDev