wolfSSL/wolfssl #2674: Cookie exchange in Session-Resuming Handshake

probas, 2019-12-16:
Hi,

As per Figure 2 from [Datagram Transport Layer Security Version 1.2](https://tools.ietf.org/html/rfc6347) there are only 3 flights for the handshake that employs session resumption:
[image]

However, when I perform resumption from examples using client and server as I see an extra flight with cookie exchange at the very beginning caused by HelloVerifyRequest from the server, e.g packets 52-54:
[image]

Questions are: can cookie exchange in the lib examples be explained by section 4.2.1. Denial-of-Service Countermeasures of the above mentioned RFC? Are there other reasons of having HelloVerifyRequest during session resumption?

Usage example of client/server built from the latest sources:
server -s -u -r
client -s -u -r

kaleb-himes, 2019-12-16:
@probas,

Thank you so much for reaching out to wolfSSL. I will check with our resident DTLS expert and get you an answer as soon as possible!

Regards,

K

ejohnstown, 2019-12-18:
Yes, we are opting to perform a cookie exchange in line with section 4.2.1 of the RFC. The hello verify transaction is still providing a filter before the server bothers to allocate resources towards performing the shortened resume handshake.

ejohnstown, 2019-12-30:
Is this something you want to have changed?

probas, 2020-01-13:
Sorry for delay.
My question was rather a clarification, as I faced the issue with my WolfSSL-based client failing to complete the handshake if the server did not send HelloVerifyRequest.
The issue had been already resolved through WolfSSL support. Thank you!

closed 2020-01-13 by probas