wolfSSL/wolfssl #2680: wolfssl4.2.0 accepts a certificate whose issuer not matching the subject of CA certificate

llqll, 2019-12-17:
I recently created a certificate chain [rootCA,intermediate certificate,leaf certificate], where the issuer distinguished name of leaf certificate is not the same as subject distinguished name of CA certificate  (intermediate certificate). 

Clearly, leaf certificate violate Section 4.1.2.6, RFC5280: “If the subject is a CA (e.g., the basic constraints extension, as discussed in Section 4.2.1.9, is present and the value of cA is TRUE), then the subject field MUST be populated with a non-empty distinguished name matching the contents of the issuer field (Section 4.1.2.4) in all certificates issued by the subject CA.”. 

Meanwhile, the chain can still pass certificate verification with wolfssl-4.2.0. however, the chain was rejected in openssl. 

the  subject distinguished name of intermediate certificate is:
```
 C = CN, ST = TJ5, O = TJU, OU = beiyangyuan, CN = b326, emailAddress = li1@163.com
```
the  issuer distinguished name of leaf certificate is:
```
 0.36.36.36.36.14752.36.36.36.36.50.16292.36.36.36.100.69.36.36.36.36.36.36.36.3 = CN, ST = TJ5, O = TJU, OU = beiyangyuan, CN = b326, emailAddress = li1@163.com
```
**Steps to reproduce**  
The code I used is(wolfssl.c):
```
#include <wolfssl/options.h>
#include <wolfssl/ssl.h>
#include <wolfssl/wolfcrypt/error-crypt.h>
int wolfssl_verify_cert(char * cert_file,char *ca_file)
{
    int ret = 0;
    WOLFSSL_CERT_MANAGER* cm;
    cm = wolfSSL_CertManagerNew();//new a certmanager to verify certificate
    if(cm == NULL) {
        printf("error creating new cert manager\n");
    }
    ret = wolfSSL_CertManagerLoadCA(cm,ca_file, NULL);
    if (ret != SSL_SUCCESS) {
        printf ("error loading CA certs into cert manager\n");
    }
    ret = wolfSSL_CertManagerVerify(cm, cert_file,SSL_FILETYPE_PEM);
    if (ret != SSL_SUCCESS) {
       int a=ret;
        wolfSSL_ERR_reason_error_string(-144);
        printf("wolfSSL_CertManagerVerify() failed (%d): %s\n",ret, 
        wolfSSL_ERR_reason_error_string(ret));
    }
    else printf("Verification Successful!\n");
    wolfSSL_CertManagerFree(cm);
    return ret;
}

int main(int argc,char **argv){
   char * cert_file=argv[1];
   char *ca_file=argv[2];
   wolfssl_verify_cert(cert_file,ca_file);
   return 0;
}
```
The command is :
```
gcc -g wolfssl.c -o wolfssl -lwolfssl
wolfssl   leaf.pem  1.pem
```
**Expected result**   
Verification failed
**Actual result**  
Verification Successful!

**Reason**  
Through debugging, I found that wolfssl looks up the CA certificate through the authority key identifier. If it doesn't find it, it will use the issuer field to find it again. However, if it find CA certificate through the authority key identifier, direct access to the CA certificate and will not do check to the issuer .  In this case, the Authority Key Identifier field of leaf certificate matches the Subject Key Identifier of the CA certificate, the verification result will be successful. 

However, openssl will perform more checks after finding CA certificate by issuer name.
the wolfssl code:
[image]
the openssl code:
[image]

[... 101 lines cut ...]
bWljcm9zb2Z0LmNvbS9wa2kvbXNjb3JwL2NybC9NaWNyb3NvZnQlMjBJVCUyMFRM
UyUyMENBJTIwMi5jcmyGSWh0dHA6Ly9jcmwubWljcm9zb2Z0LmNvbS9wa2kvbXNj
b3JwL2NybC9NaWNyb3NvZnQlMjBJVCUyMFRMUyUyMENBJTIwMi5jcmwwgYUGCCsG
AQUFBwEBBHkwdzBRBggrBgEFBQcwAoZFaHR0cDovL3d3dy5taWNyb3NvZnQuY29t
L3BraS9tc2NvcnAvTWljcm9zb2Z0JTIwSVQlMjBUTFMlMjBDQSUyMDIuY3J0MCIG
CCsGAQUFBzABhhZodHRwOi8vb2NzcC5tc29jc3AuY29tMD4GCSsGAQQBgjcVBwQx
MC8GJysGAQQBgjcVCIfahnWD7tkBgsmFG4G1nmGF9OtggV2E0t9CgueTegIBZAIB
GzAdBgNVHSUEFjAUBggrBgEFBQcDAgYIKwYBBQUHAwEwCwYDVR0PBAQDAgSwME0G
A1UdIARGMEQwQgYJKwYBBAGCNyoBMDUwMwYIKwYBBQUHAgEWJ2h0dHA6Ly93d3cu
bWljcm9zb2Z0LmNvbS9wa2kvbXNjb3JwL2NwczAnBgkrBgEEAYI3FQoEGjAYMAoG
CCsGAQUFBwMCMAoGCCsGAQUFBwMBMA0GCSqGSIb3DQEBCwUAA4ICAQBITaJ7K6jS
3nu4ZG7bpUOvlvVF8A9Yboo+KGcmH9NRAagrxzDscuvYlvPFPuAems84mDeG/fBP
jcIXXUIStKhGPXZhFKrXfkO+Trq0busUwaTwVi/r/nhjpPWZJnT5ljkcg/ptR8pe
beydUhh11ShtC8WgLM3SP4eGNC/vTTakE7xMvKTTtBqvOujTST6p19y4jSJUzqZW
qgmo6O6lDmMrxBPJxsscFXYlTENjHj4vKo5YQhogTxh0AghO7RZ12JSrhkcEQeYE
bdbCd6IM3FiH0rooeHe7DxvHM9Vjcocpb40HYLMvrRJWpbNdafJr5qmt4JQH678j
62RK+gK36nbcfnvNb2DAADQbMMClnC0St4hazfjrptxRBZNf8Eg+ZAynpOrDak2K
+P+Fs2vQKKKgLf5DK9ViaHvW8zgSZRUqIxvvRV+MdYSbUcxVf2kfjgc3X9ESvzDI
0Ll5rZ8fW98gVXyPPT4+/9oNWQ2xP0E+4BBHfZXn2bn0miUqUcQuyvvelv39o76F
7iw+RgfEZk1j3vtzJvy87L7H0la3p79sL35jit9ThO+hnC7JvdQOV7NVz1ljDuh0
bMX7ashDwIsPRCEI8pLt3EyjcXqt/3OAa1orVubYRR3Wxw7F0oyu1aLv9Zo8VtBc
VdyA+XYfRoGXWrR3eRLBs8y81I/eyOfH5Q==
-----END CERTIFICATE-----

```

SparkiDev, 2019-12-18:
@llqll 

A pull request has gone up to fix master:
https://github.com/wolfSSL/wolfssl/pull/2682

An equivalent patch for wolfSSL 4.0.0 is below.

diff --git a/wolfcrypt/src/asn.c b/wolfcrypt/src/asn.c
index 162c2ef71..df1218e34 100644
--- a/wolfcrypt/src/asn.c
+++ b/wolfcrypt/src/asn.c
@@ -8088,6 +8088,10 @@ int ParseCertRelative(DecodedCert* cert, int type, int verify, void* cm)
                 cert->ca = GetCA(cm, cert->extAuthKeyId);
             if (cert->ca == NULL && cert->extSubjKeyIdSet)
                 cert->ca = GetCA(cm, cert->extSubjKeyId);
+            if (cert->ca != NULL && XMEMCMP(cert->issuerHash,
+                                  cert->ca->subjectNameHash, KEYID_SIZE) != 0) {
+                cert->ca = NULL;
+            }
             if (cert->ca == NULL) {
                 cert->ca = GetCAByName(cm, cert->issuerHash);
                 if (cert->ca && cert->extAuthKeyIdSet) {

Let me know if this fixes the issue as expected.

Sean

llqll, 2019-12-18:
@SparkiDev  
I know a pull request has gone up to fix master:
#2682

That pull request can't solve this problem. That pull request solves the problem of accepting certificates with mismatched authority key identifier fields. The current problem is caused by mismatched issuer fields.

SparkiDev, 2019-12-18:
@llqll

The code change does make your test fail.
If the AKID matches, cert->ca is not NULL, then the issuer name is check against the subject name of the CA cert. If they don't match then a look up purely by issuer name is done.

Does this make sense?

Sean

llqll, 2019-12-18:
OK, I see ,Thank you very much !

SparkiDev, 2019-12-18:
@llqll

The pull request has been merged in to master and will go out in the next release.
Thanks for raising the issue and helping us improve wolfSSL!

Sean

closed 2019-12-18 by SparkiDev