wolfSSL/wolfssl #2691: [wolfCrypt] Much slower performance compared to other libraries

davidebeatrici, 2019-12-19:
I wrote a small benchmark tool to help decide which library we will switch to in Mumble: https://github.com/mumble-voip/crypto-benchmark

Results on an Intel Core i5-4690:
```
[OpenSSL] AES-256-GCM took 2.965976 seconds for 1000000 iterations, 4096 bytes message
[OpenSSL] AES-256-OCB took 2.413906 seconds for 1000000 iterations, 4096 bytes message
[OpenSSL] ChaCha20-Poly1305 took 4.417644 seconds for 1000000 iterations, 4096 bytes message
[libsodium] AES-256-GCM took 5.480515 seconds for 1000000 iterations, 4096 bytes message
[libsodium] ChaCha20-Poly1305 took 6.412273 seconds for 1000000 iterations, 4096 bytes message
[wolfCrypt] AES-256-GCM took 131.165997 seconds for 1000000 iterations, 4096 bytes message
[wolfCrypt] ChaCha20-Poly1305 took 28.554290 seconds for 1000000 iterations, 4096 bytes message
```

Results on the ARM64 builder provided by Travis CI:
```
[OpenSSL] AES-256-GCM took 17.986986 seconds for 1000000 iterations, 4096 bytes message
[OpenSSL] AES-256-OCB took 23.557388 seconds for 1000000 iterations, 4096 bytes message
[OpenSSL] ChaCha20-Poly1305 took 28.620894 seconds for 1000000 iterations, 4096 bytes message
[libsodium] skipping AES-256-GCM benchmark due to missing requirements. SSSE3 extensions, "aesni" and "pclmul" instructions are required.
[libsodium] ChaCha20-Poly1305 took 32.326541 seconds for 1000000 iterations, 4096 bytes message
[wolfCrypt] AES-256-GCM took 233.905675 seconds for 1000000 iterations, 4096 bytes message
[wolfCrypt] ChaCha20-Poly1305 took 43.984752 seconds for 1000000 iterations, 4096 bytes message
```

As you can see, wolfCrypt is unfortunately much slower compared to libsodium and OpenSSL.

Am I using the API incorrectly?

dgarske, 2019-12-19:
Hi @davidebeatrici ,

Thanks for the question and link your benchmark tool! 

Our default ./configure options do not enable all our possible speedup options for Intel platforms.

Please try using the following extra ./configure options:
`./configure --enable-intelasm --enable-aesni --enable-sp --enable-sp-asm`

Help `./configure --help` for these entries:

```
--enable-intelasm       Enable All Intel ASM speedups (default: disabled)
--enable-aesni          Enable wolfSSL AES-NI support (default: disabled)
--enable-sp             Enable Single Precision maths implementation
--enable-sp-asm         Enable Single Precision assembly implementation
```

I expect with these options enabled you will see much better results. Looking forward to your results.

Thanks,
David Garske, wolfSSL

davidebeatrici, 2019-12-19:
Hi @dgarske,

Thank you very much for your reply!

Here are the options used in the official Debian Bullseye package (https://packages.debian.org/bullseye/libwolfssl23):
```
--enable-distro
--enable-tls13
--disable-examples
--disable-silent-rules
```

Same for the Ubuntu Bionic package (https://packages.ubuntu.com/bionic/libwolfssl15), but without `--enable-tls13`.

I'm going to compile the library with the options you suggested and let you know about the results.

davidebeatrici, 2019-12-19:
```
[OpenSSL] AES-256-GCM took 2.974256 seconds for 1000000 iterations, 4096 bytes message
[OpenSSL] AES-256-OCB took 2.419976 seconds for 1000000 iterations, 4096 bytes message
[OpenSSL] ChaCha20-Poly1305 took 4.455250 seconds for 1000000 iterations, 4096 bytes message
[libsodium] AES-256-GCM took 5.276064 seconds for 1000000 iterations, 4096 bytes message
[libsodium] ChaCha20-Poly1305 took 6.215602 seconds for 1000000 iterations, 4096 bytes message
[wolfCrypt] AES-256-GCM took 3.286140 seconds for 1000000 iterations, 4096 bytes message
[wolfCrypt] ChaCha20-Poly1305 took 4.402146 seconds for 1000000 iterations, 4096 bytes message
```

Insane!

Is there a reason why those options are disabled by default?

davidebeatrici, 2019-12-20:
I added the results achieved with an AMD CPU in the first post, the issue is present with that as well.

dgarske, 2019-12-20:
Hi @davidebeatrici,

Q: "Is there a reason why those options are disabled by default?":

A: It's because we run on so many platforms (CPU's/compilers) and we need to ensure they all continue to work properly. By default just plain c versions of the code are enabled.

For reference we also have speedups for aarch32/aarch64. Those are enabled with `--enable-armasm`.

Thanks,
David Garske, wolfSSL

dgarske, 2019-12-20:
Hi @davidebeatrici ,

Also we have a few built-in benchmark tools:

* `./wolfcrypt/benchmark/benchmark -?`
* `./examples/benchmark/tls_bench -?`
* `./scripts/benchmark.test`

Thanks,
David Garske, wolfSSL

closed 2019-12-20 by dgarske

davidebeatrici, 2019-12-20:
Hi @dgarske,

Thank you for your explanation.

> It's because we run on so many platforms (CPU's/compilers) and we need to ensure they all continue to work properly. By default just plain c versions of the code are enabled.

That's what I thought, the issue is that AES is way too slow.

ChaCha20-Poly1305 is also slow, but not nearly as much as AES.

AES also seems to be directly affected by the single-thread performance, because the FX-6300 and the ARM64 CPU are 2 times slower than the i5-4690.

toby63, 2020-04-20:
@davidebeatrici  @dgarske: 
I don't know much about this, but isn't it possible for package maintainers to only include the options mentioned above (see also below) for specific architectures?
Or is there a problem regarding some CPUs that use the same architecture?
(Regarding compilers, the maintainers only use one obviously)

To clarify:
If possible, I would inform maintainers more actively about this.

Options:
```
--enable-intelasm       Enable All Intel ASM speedups (default: disabled)
--enable-aesni          Enable wolfSSL AES-NI support (default: disabled)
--enable-sp             Enable Single Precision maths implementation
--enable-sp-asm         Enable Single Precision assembly implementation
```

@davidebeatrici : I could not find a bug report on the page of the debian package, have you spoken with the maintainer?

davidebeatrici, 2020-04-20:
Yes, maintainers can specify options for specific architectures in the build script.

I didn't open a bug report, nor spoke to the maintainer. Would you like to do that?

toby63, 2020-04-20:
@davidebeatrici 
> Would you like to do that?

Well, I don't like my emailadresses to be public everywhere (debians bug system is horrible imo).
If you want to contact him, the [debian page](https://tracker.debian.org/pkg/wolfssl) also includes the maintainers emailadress. Maybe a mail from a mumble developer would also be more convincing :wink:.

davidebeatrici, 2020-04-20:
No problem!

toby63, 2020-04-26:
@davidebeatrici 
Any answer from the maintainer yet?

davidebeatrici, 2020-05-30:
I didn't receive an answer from the maintainer yet.

davidebeatrici, 2020-07-27:
@lechner

speedyb0y, 2021-05-15:
./configure --enable-openvpn --enable-tlsx  --enable-sni --enable-poly1305 --enable-chacha  --enable-aesni --prefix=/usr

```
[wolfCrypt] running AES-256-GCM benchmark...
[wolfCrypt] 4.054036 seconds for 1000000 iterations, 4096 bytes message
[wolfCrypt] running ChaCha20-Poly1305 benchmark...
[wolfCrypt] 5.528720 seconds for 1000000 iterations, 4096 bytes message
```

./configure --enable-openvpn --enable-tlsx  --enable-sni --enable-poly1305 --enable-chacha  --enable-aesni --prefix=/usr --enable-intelrand

```
[wolfCrypt] running AES-256-GCM benchmark...
[wolfCrypt] 3.966141 seconds for 1000000 iterations, 4096 bytes message
[wolfCrypt] running ChaCha20-Poly1305 benchmark...
[wolfCrypt] Iteration 354832/1000000 (35%), elapsed time: 10.005612
[wolfCrypt] Iteration 697005/1000000 (69%), elapsed time: 20.006941
[wolfCrypt] 28.447847 seconds for 1000000 iterations, 4096 bytes message
```

Look what happens to ChaCha20-Poly1305 if you use --enable-intelrand ...

dgarske, 2021-05-17:
> ./configure --enable-openvpn --enable-tlsx --enable-sni --enable-poly1305 --enable-chacha --enable-aesni --prefix=/usr
>
> ```
> [wolfCrypt] running AES-256-GCM benchmark...
> [wolfCrypt] 4.054036 seconds [...]

Hi @speedyb0y ,

I am unable to reproduce with our internal wolfcrypt/benchmark/benchmark. Can you please provide details on the CPU being used and the intel assembly speedups available (something like `cat /proc/cpuinfo` output)? Also can you confirm where these benchmarks are coming from (guessing openvpn)?

```
% ./configure --enable-sni --enable-poly1305 --enable-chacha --enable-aesni && make
% ./wolfcrypt/benchmark/benchmark -chacha20-poly1305 -aes-gcm
------------------------------------------------------------------------------
 wolfSSL version 4.7.1
------------------------------------------------------------------------------
wolfCrypt Benchmark (block bytes 1048576, min 1.0 sec each)
AES-128-GCM-enc   3930 MB took 1.001 seconds, 3927.067 MB/s Cycles per byte =   0.75
AES-128-GCM-dec   3955 MB took 1.000 seconds, 3953.161 MB/s Cycles per byte =   0.75
AES-192-GCM-enc   3530 MB took 1.000 seconds, 3528.800 MB/s Cycles per byte =   0.84
AES-192-GCM-dec   3605 MB took 1.001 seconds, 3601.543 MB/s Cycles per byte =   0.82
AES-256-GCM-enc   3180 MB took 1.001 seconds, 3177.556 MB/s Cycles per byte =   0.93
AES-256-GCM-dec   3195 MB took 1.001 seconds, 3190.568 MB/s Cycles per byte =   0.93
GMAC Table 4-bit  1491 MB took 1.000 seconds, 1490.525 MB/s Cycles per byte =   1.98
CHA-POLY           330 MB took 1.006 seconds,  327.976 MB/s Cycles per byte =   9.00
Benchmark complete
```

```
% ./configure --enable-sni --enable-poly1305 --enable-chacha --enable-aesni --enable-intelrand && make
% ./wolfcrypt/benchmark/benchmark -chacha20-poly1305 -aes-gcm
------------------------------------------------------------------------------
 wolfSSL version 4.7.1
------------------------------------------------------------------------------
wolfCrypt Benchmark (block bytes 1048576, min 1.0 sec each)
AES-128-GCM-enc   3860 MB took 1.001 seconds, 3856.114 MB/s Cycles per byte =   0.77
AES-128-GCM-dec   4030 MB took 1.001 seconds, 4027.873 MB/s Cycles per byte =   0.73
AES-192-GCM-enc   3710 MB took 1.001 seconds, 3706.220 MB/s Cycles per byte =   0.80
AES-192-GCM-dec   3680 MB took 1.001 seconds, 3677.195 MB/s Cycles per byte =   0.80
AES-256-GCM-enc   3180 MB took 1.000 seconds, 3178.719 MB/s Cycles per byte =   0.93
AES-256-GCM-dec   3200 MB took 1.001 seconds, 3196.679 MB/s Cycles per byte =   0.92
GMAC Table 4-bit  1494 MB took 1.000 seconds, 1493.411 MB/s Cycles per byte =   1.98
CHA-POLY           330 MB took 1.007 seconds,  327.702 MB/s Cycles per byte =   9.01
Benchmark complete
```

Thanks,
David Garske, wolfSSL

SparkiDev, 2021-05-18:
Hi @speedyb0y,

I've downloaded crypto-benchmark and tested this on my machine.
There was no difference in performance.
Please try again and make sure the system is not busy.
The rdrand instruction is using chip entropy that can be exhausted. That is, if multiple programs are trying to use rdrand then it may actually return not available and will have to be executed again.
I should also say that wolfSSL does not use random or rdrand in ChaCha20 and Poly1305 code.

It looks like that the first run is using --enable-intelasm and the second isn't.
Running the configurations you have always produce slow ChaCha20-Poly1305 performance for me.
Please check the configurations.

Thanks,
Sean Parkinson, wolfSSL