wolfSSL/wolfssl #2697: AESNI Open AES-128-CBC Mode EVP_CipherInit crash

zero-rp, 2019-12-22:
>	Test.exe!AES_128_Key_Expansion()  1267	
 	Test.exe!AES_set_encrypt_key(const unsigned char * userKey, const int bits, Aes * aes)  704	C
 	Test.exe!wc_AesSetKey(Aes * aes, const unsigned char * userKey, unsigned int keylen, const unsigned char * iv, int dir)  2350	C
 	Test.exe!AesSetKey_ex(Aes * aes, const unsigned char * key, unsigned int len, const unsigned char * iv, int dir, int direct)  16277	C
 	Test.exe!wolfSSL_EVP_CipherInit(WOLFSSL_EVP_CIPHER_CTX * ctx, const char * type, const unsigned char * key, const unsigned char * iv, int enc)  16328	C

embhorn, 2019-12-23:
Hello @zero-rp 

Thank you for contacting wolfSSL Support. Could you please provide the configuration and environment used to reproduce this?

Kind regards,
Eric
wolfSSL Support

zero-rp, 2019-12-23:
#ifndef _WIN_USER_SETTINGS_H_
#define _WIN_USER_SETTINGS_H_

#define WC_RSA_BLINDING

#define OPENSSL_ALL
#define OPENSSL_EXTRA

#define WOLFSSL_RIPEMD
#define NO_PSK
#define HAVE_EXTENDED_MASTER
#define HAVE_TLS_EXTENSIONS
#define HAVE_ALPN
#define WOLFSSL_ALLOW_TLSV10
#define WOLFSSL_TLS13
#define HAVE_AES_ECB
#define HAVE_AES_KEYWRAP
#define HAVE_AESGCM
#define HAVE_AESCCM
#define WOLFSSL_AES_CFB
#define WOLFSSL_AES_DIRECT
#define WOLFSSL_AES_XTS
#define WOLFSSL_SHA384
#define WOLFSSL_SHA512
#define HAVE_ECC
#define HAVE_ECC_ENCRYPT
#define HAVE_ECC_KOBLITZ
#define ECC_SHAMIR
#define ECC_TIMING_RESISTANT
#define WOLFSSL_CUSTOM_CURVES
#define HAVE_SUPPORTED_CURVES
#define HAVE_SECURE_RENEGOTIATION
#define WC_NO_HARDEN
#define WOLFSSL_KEY_GEN
#define HAVE_COMP_KEY
#define WOLFSSL_DES_ECB
#define HAVE_FFDHE_2048
#define WC_RSA_PSS
#define HAVE_LIBZ
#define HAVE_PKCS7
#define HAVE_HKDF
#define HAVE_X963_KDF
#define WOLFSSL_CERT_EXT
#define WOLFSSL_CERT_GEN
#define WOLFSSL_CERT_REQ
#define WOLFSSL_DER_LOAD
#define WOLFSSL_DTLS
#define HAVE_CAMELLIA
#define HAVE_CHACHA
#define HAVE_POLY1305
#define HAVE_ONE_TIME_AUTH
#define HAVE_CRL
#define HAVE_CRL_IO
#define HAVE_OCSP
#define WOLFSSL_SNIFFER
#define WOLFSSL_SNIFFER_WATCH
#define WOLFSSL_SNIFFER_STATS
#define HAVE_CURVE25519

/* Optional Performance Speedups */
#if 1
    /* AESNI on x64 */
#ifdef _WIN64
#define HAVE_INTEL_RDSEED
#define WOLFSSL_AESNI
#endif

/* Single Precision Support for RSA/DH 1024/2048/3072 and ECC P-256 */
#define WOLFSSL_SP
#define WOLFSSL_HAVE_SP_ECC
#define WOLFSSL_HAVE_SP_DH
#define WOLFSSL_HAVE_SP_RSA
#endif

#if _DEBUG
#define DEBUG_WOLFSSL
#endif

#endif /* _WIN_USER_SETTINGS_H_ */

zero-rp, 2019-12-23:
wolfSSL release version 4.3.0

zero-rp, 2019-12-23:
EVP_CIPHER_CTX* evp = (EVP_CIPHER_CTX*)lua_newuserdata(L, sizeof(EVP_CIPHER_CTX));
    EVP_CIPHER_CTX_init(evp);
    EVP_CipherInit(evp, cipher , key, iv, enc)

zero-rp, 2019-12-23:
Using the space allocated by lua_newuserdata in Lua environment will cause crash

embhorn, 2019-12-23:
The memory allocated by `lua_newuserdata` is likely not accessible by the library, so the crash is the result of an invalid memory access.
If you want to customize the memory allocation for the wolfSSL library, you'll need to change the XMALLOC / XREALLOC / XFREE macros.

zero-rp, 2019-12-24:
If you don't open AES, it will be correct. What's the reason

zero-rp, 2019-12-24:
If you don't open AESNI, it will be correct. What's the reason

ejohnstown, 2019-12-26:
Is `lua_newuserdata()` returning something that is long word aligned? Some of the AES-NI processes need to be word aligned.

kaleb-himes, 2020-01-23:
Closing as question from John was never responded to, let us know if the issue persists moving forward.

closed 2020-01-23 by kaleb-himes