wolfSSL/wolfssl #2703: how can i do use --enable-libwebsockets option compile wolfssl with MSVC?

myvcgit, 2019-12-25:
i use msvc compile wolfssl ,but i can't find how to use --enable-libwebsockets option.
i try to set OPENSSL_ALL  in Preprocessor Definitions, it's the one rithg way?
if it's ritgh ,when i compile libwebsockets the have one error "websockets_static.lib(openssl-ssl.obj) :
 error LNK2019: unresolved external symbol _SSL_get_cipher_version referenced in function _lws_openssl_describe_cipher"
 
i find "SSL_get_cipher_version" and comment it use "hello" replace can finsh compile.

int lws_openssl_describe_cipher(struct lws *wsi)
{
#if !defined(LWS_WITH_NO_LOGS)
	int np = -1;
	SSL *s = wsi->tls.ssl;

	SSL_get_cipher_bits(s, &np);
	lwsl_info("%s: wsi %p: %s, %s, %d bits, %s\n", __func__, wsi,
			SSL_get_cipher_name(s), SSL_get_cipher(s), np,
			/*SSL_get_cipher_version(s)*/"hello");
#endif

	return 0;
}

but when use the official examples lws-minimal-ws-client-rx test that have error :

[2019/12/25 08:34:49:3524] D: lws_ssl_get_error: 000C6FD0 -1 -> -308 (errno 0)
[2019/12/25 08:34:49:3534] E: CLIENT_CONNECTION_ERROR: lws_ssl_client_connect1 failed

embhorn, 2019-12-26:
Hello @myvcgit 

Please try adding the following compiler flags:
`-DWOLFSSL_LIBWEBSOCKETS -DOPENSSL_NO_EC`

These are enabled with the `--enable-libwebsockets` configuration option in a linux build.

Thanks,
Eric
wolfSSL Support

myvcgit, 2019-12-27:
@embhorn Thank you for your reply.
i do that ,but that have 3 error:
2>M:\libtest\libwebsockets-master\lib\tls\openssl\openssl-ssl.c(43,21): warning C4013: 'SSL_get_cipher_bits' undefined; assuming extern returning int
2>M:\libtest\libwebsockets-master\lib\tls\openssl\openssl-ssl.c(46,1): warning C4013: 'SSL_get_cipher_version' undefined; assuming extern returning int
2>M:\libtest\libwebsockets-master\lib\tls\openssl\openssl-client.c(761,35): warning C4013: 'SSL_CTX_use_certificate_ASN1' undefined; assuming extern returning int

i can find SSL_get_cipher_bits SSL_CTX_use_certificate_ASN1 in wolfssl/openssl/ssl.h they need defined OPENSSL_ALL to open. so i think use OPENSSL_ALL is better when do that with msvc.

but i can't find SSL_get_cipher_version define, can u tell  what function to replace in wolfssl.

embhorn, 2019-12-27:
Hi @myvcgit 

According to [this ](https://github.com/wikimedia/operations-debs-openssl/blob/master/doc/ssl/SSL_get_current_cipher.pod) that function is just a macro:

```
#define SSL_get_cipher_version(s) \
               SSL_CIPHER_get_version(SSL_get_current_cipher(s))
```

So `wolfSSL_CIPHER_get_version` is defined, as is `wolfSSL_get_current_cipher`.

myvcgit, 2019-12-28:
@embhorn Thank you 

i add
 #define SSL_get_cipher_version(s) \
               wolfSSL_CIPHER_get_version(SSL_get_current_cipher(s))
is seems to be fine

closed 2019-12-28 by myvcgit