wolfSSL/wolfssl #2712: Building LINUX-SGX with advanced options fails

ShivKushwah, 2020-01-04:
I'm trying to build https://github.com/wolfSSL/wolfssl/tree/master/IDE/LINUX-SGX using the Readme instructions. This works when I run the following command 
`make -f sgx_t_static.mk all`

However, I want to build it with DEBUG options. Following the readme, I run the following
`make -f sgx_t_static.mk CFLAGS=-DDEBUG_WOLFSSL HAVE_WOLFSSL_BENCHMARK=1 HAVE_WOLFSSL_TEST=1` and fail with the following error

`
cc -DDEBUG_WOLFSSL -Wno-implicit-function-declaration -std=c11 -m64 -O2 -nostdinc -fvisibility=hidden -fpie -fstack-protector -I/home/shiv/Research/wolfssl/ -I/home/shiv/Research/wolfssl/wolfcrypt/ -I/home/shiv/Research/Intel-SGX-Installation/linux-sgx/linux/installer/bin/sgxsdk/include -I/home/shiv/Research/Intel-SGX-Installation/linux-sgx/linux/installer/bin/sgxsdk/include/tlibc -I/home/shiv/Research/Intel-SGX-Installation/linux-sgx/linux/installer/bin/sgxsdk/include/stlport -I/home/shiv/Research/wolfssl/wolfcrypt/test -I/home/shiv/Research/wolfssl/wolfcrypt/benchmark/ -fno-builtin-printf -I. -DWOLFSSL_SGX   -c -o /home/shiv/Research/wolfssl/wolfcrypt/test/test.o /home/shiv/Research/wolfssl/wolfcrypt/test/test.c
/home/shiv/Research/wolfssl/wolfcrypt/test/test.c:28:14: fatal error: wolfssl/options.h: No such file or directory
     #include <wolfssl/options.h>
              ^~~~~~~~~~~~~~~~~~~
compilation terminated.
<builtin>: recipe for target '/home/shiv/Research/wolfssl/wolfcrypt/test/test.o' failed
make: *** [/home/shiv/Research/wolfssl/wolfcrypt/test/test.o] Error 1
`

Any ideas what I can do?

JacobBarthelmeh, 2020-01-06:
Hi ShivKushwah,

If this is from a git clone of wolfSSL then touch wolfssl/options.h should be done from the root wolfssl directory (or ./autogen.sh && ./configure && ./config.status). Bundles downloaded from the wolfssl.com website should have a blank wolfssl/options.h in them. We recently added in that include to test.c, I'll update the README. Thanks for the report.

ShivKushwah, 2020-01-06:
Awesome thanks! In the meantime, what exactly do I need to do to get the options.h file included? Do I need to run the 3 commands you mentioned before I run the make command?

JacobBarthelmeh, 2020-01-07:
Easiest way is just creating an empty wolfssl/options.h file or to download the release bundle from https://www.wolfssl.com/download/

JacobBarthelmeh, 2020-01-14:
Hi @ShivKushwah did creating an empty wolfssl/options.h file resolve the issue for you?

kaleb-himes, 2020-01-23:
Inquirer never responded, please re-open if issue persists.

closed 2020-01-23 by kaleb-himes