wolfSSL/wolfssl #2744: --enable-all forcibly disables oldnames

bagder, 2020-01-16:
I build curl with wolfSSL every now and then. Today I updated wolfSSL from git master and tried this again and the curl build failed.

It failed because `RNG` doesn't exist in the wolfssl headers (anymore). Not sure when this change happened but I've built wolfSSL the same way many times over the years and used that lib to build curl with.

I run the wolfssl configure script like this:

    ./configure --prefix=[path] --enable-harden --enable-ssh --enable-all

RNG doesn't exist because the wolfssl configure summary says:

     * Old Names:                  no

Why does it say no? configure --help says `--enable-oldnames` are by default enabled.

The reason (@danielinux helped me debug), is that `--enable-all` executes line 1903:

https://github.com/wolfSSL/wolfssl/blob/6b4551c012941ebc407d843c9a6dc835da546a34/configure.ac#L1903

... which magically turns off enable oldnames even if I explicitly add `--enable-oldnames` on the command line.

I had to edit configure.ac, remove that line and then run configure again to fix the build. Then curl builds fine again with the resulting header files and wolfssl build.

(I presume I should also rather fix the curl end and replace `RNG` with `WC_RNG` to avoid future issues like this?)

dgarske, 2020-01-16:
Hi @bagder 

The history here is we originally added `--enable-distro` for builds on Linux and I suspect this option was there for the Linux distros. Later we added the `--enable-all` and moved features into "all" to simply some testing. We may want to move the the disable old names option into disto, but we'll need to review the history to confirm.

For now I recommend using `WC_RNG` if possible.

Thanks,
David Garske, wolfSSL

bagder, 2020-01-16:
Roger that. No big deal at all for me. Will switch over.

closed 2020-01-23 by kaleb-himes