wolfSSL/wolfssl #2745: Build failure on IBM Z (s390x)

tobiasbrunner, 2020-01-16:
I get the following build failure on IBM Z (s390x) on Travis CI:

```
In file included from wolfcrypt/src/hmac.c:54:0:
./wolfcrypt/src/misc.c: In function ‘xorbuf’:
./wolfcrypt/src/misc.c:185:10: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
     if (((wolfssl_word)buf | (wolfssl_word)mask | count) % WOLFSSL_WORD_SIZE == 0)
          ^
./wolfcrypt/src/misc.c:185:30: error: cast from pointer to integer of different size [-Werror=pointer-to-int-cast]
     if (((wolfssl_word)buf | (wolfssl_word)mask | count) % WOLFSSL_WORD_SIZE == 0)
                              ^
cc1: all warnings being treated as errors
```

Several more are reported for other usages of `xorbuf()`.

dgarske, 2020-01-16:
Hi @tobiasbrunner ,

It looks like we are not properly detecting 64-bit support on that platform. We have code to try and detect the 64-bit support using some build macros here: https://github.com/wolfSSL/wolfssl/blob/master/wolfssl/wolfcrypt/types.h#L105.

I believe the patch to fix would look like this:

```
@@ -104,7 +104,7 @@
     /* These platforms have 64-bit CPU registers.  */
     #if (defined(__alpha__) || defined(__ia64__) || defined(_ARCH_PPC64) || \
          defined(__mips64)  || defined(__x86_64__) || defined(_M_X64)) || \
-         defined(__aarch64__) || defined(__sparc64__) || \
+         defined(__aarch64__) || defined(__sparc64__) || defined(__s390x__ ) || \
         (defined(__riscv_xlen) && (__riscv_xlen == 64))
         typedef word64 wolfssl_word;
         #define WC_64BIT_CPU
```

Is it possible for you to confirm this fix? 

Another option is to specify it via the `SIZEOF_LONG_LONG` macro like this: `./configure CFLAGS="-DSIZEOF_LONG_LONG=8`

Thanks,
David Garske, wolfSSL

tobiasbrunner, 2020-01-16:
Thanks David. The second option didn't work (specified via `C_EXTRA_FLAGS`, though, if that makes a difference), but the patch fixed the issue.

PR #2747 (2020-01-17): Improvements for detection of 64-bit support
* Adds support for IBM s390x. Fixes #2745.
* Improves detection on Windows. Fixes #2600
* Adds check for `HAVE_LIMITS_H` to use limits.h to detect sizeof long.
commit 541bf3e6 (2020-01-17): Improvements for detection of 64-bit support. Adds support for IBM s390x. Improves detection on Windows. Adds new `WC_USE_LIMITS_FOR_SIZEOF` option to use limits.h to detect sizeof long. Fixes #2600 and Fixes #2745.
closed 2020-01-17 by JacobBarthelmeh