wolfSSL/wolfssl #2749: Invalid index access while initialize DecodedCert using GetName function.

ArtBlnd, 2020-01-17:
Seems `GetName` function success to parse `businessCategory`, `jurisdictionC`, `jurisdictionST` but it does not apply to its full name length and `bcIdx`, `jcIdx`, `jsIdx` field in DecodedName.

those index field seems re-calculated on bottom of `GetName` function. and there is lack with `businessCategory`, `jurisdictionC`, `jurisdictionST` tags.

> lack of total length increment
https://github.com/wolfSSL/wolfssl/blob/6b4551c012941ebc407d843c9a6dc835da546a34/wolfcrypt/src/asn.c#L5915-L5937

> lack of index re-calculation
https://github.com/wolfSSL/wolfssl/blob/6b4551c012941ebc407d843c9a6dc835da546a34/wolfcrypt/src/asn.c#L5945-L6057

This can cause serious problem, something like exploit (can cause crash by handshaking with certificate that contains those tags)

also, there is lack of `jurisdiction locality name` tags on `GetName` function.

kaleb-himes, 2020-01-17:
Hi @ArtBlnd,

Would it be possible for you to provide a cert with the elements you noted for testing this?

Regards,

K

ArtBlnd, 2020-01-17:
I used certificates that in [nid.naver.com](url)

simnalamburt, 2020-05-19:
I think this issue is as critical as #2459 or #2555 (CVE-2019-16748 and CVE-2019-18840), since it can cause both Out-of-bounds Read and Out-of-bounds Write on crafted certificates.

Since letting users to use craftied certificate is not that easy, exploitability score is not that high. But still, consequence is too critical if attacker succeeds; heap overflow = information disclosure, DoS, code execution

kaleb-himes, 2020-05-19:
Hi @ArtBlnd, @simnalamburt,

We were never able to reproduce this report in any version of a release nor in checkouts around the time of the report. I will gladly ask one of our devs to review it again and provide analysis feedback.

Configuration: `./configure CC="gcc -fsanitize=address" --enable-opensslextra --enable-certgen --enable-certreq  && make`

```
wolfssl-4.4.0 % ./examples/client/client -h nid.naver.com -p 443 -g -A digicert-certs/DigiCertHighAssuranceEVRootCA.pem
peer's cert info:
 issuer : /C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert ECC Extended Validation Server CA
 subject: /serialNumber=110111-1707178/C=KR/ST=Gyeonggi-do/L=Seongnam-si/O=NAVER Corp./OU=Information Security Team/CN=nid.naver.com
 altname = www.nid.naver.com
 altname = nid.naver.com
 serial number:09:93:40:fb:f1:01:58:a7:32:d9:1b:99:37:b4:97:d8 
SSL version is TLSv1.2
SSL cipher suite is TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384
SSL curve name is SECP256R1
Session timeout set to 500 seconds
Client Random : 9FFD3E808A93B05D05E2AF0E8CF81CFEA426FE7CBD9E64F32E919D0413F69F8F
SSL connect ok, sending GET...
HTTP/1.1 400 Bad Request
Server: nginx
Date: Tue, 19 May 2020 16:50:23 GMT
Content-Type: text/html; charset=iso-8859-1
Cont
ent-Length: 226
Connection: close

<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>400 Bad Request</t
```

```
wolfssl-4.3.0 % ./examples/client/client -h nid.naver.com -p 443 -g -A digicert-certs/DigiCertHighAssuranceEVRootCA.pem
peer's cert info:
 issuer : /C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert ECC Extended Validation Server CA
 subject: /serialNumber=110111-1707178/C=KR/ST=Gyeonggi-do/L=Seongnam-si/O=NAVER Corp./OU=Information Security Team/CN=nid.naver.com
 altname = www.nid.naver.com
 altname = nid.naver.com
 serial number:09:93:40:fb:f1:01:58:a7:32:d9:1b:99:37:b4:97:d8 
SSL version is TLSv1.2
SSL cipher suite is TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384
SSL curve name is SECP256R1
Session timeout set to 500 seconds
Client Random : 12F6EB9751AC32FB3C53842A9313224B3CF603A2A196D68B61CA43E5D2F929EC
SSL connect ok, sending GET...
HTTP/1.1 400 Bad Request
Server: nginx
Date: Tue, 19 May 2020 16:51:36 GMT
Content-Type: text/html; charset=iso-8859-1
Cont
ent-Length: 226
Connection: close

<!DOCTYPE HTML PUBLIC "-//IETF//DTD HTML 2.0//EN">
<html><head>
<title>400 Bad Request</t
```

```
wolfssl-4.2.0 % ./examples/client/client -h nid.naver.com -p 443 -g -A digicert-certs/DigiCertHighAssuranceEVRootCA.pem
peer's cert info:
 issuer : /C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert ECC Extended Validation Server CA
 subject: /serialNumber=110111-1707178/C=KR/ST=Gyeonggi-do/L=Seongnam-si/O=NAVER Corp./OU=Information Security Team/CN=nid.naver.com
 altname = www.nid.naver.com
 altname = nid.naver.com
 serial number:09:93:40:fb:f1:01:58:a7:32:d9:1b:99:37:b4:97:d8 
SSL version is TLSv1.2
SSL cipher suite is TLS_ECDHE_ECDSA_WITH_AES_256_GCM_SHA384
[... 167 lines cut ...]
MEigRqBEhkJodHRwOi8vY3JsNC5kaWdpY2VydC5jb20vRGlnaUNlcnRFQ0NFeHRl
bmRlZFZhbGlkYXRpb25TZXJ2ZXJDQS5jcmwwSwYDVR0gBEQwQjA3BglghkgBhv1s
AgEwKjAoBggrBgEFBQcCARYcaHR0cHM6Ly93d3cuZGlnaWNlcnQuY29tL0NQUzAH
BgVngQwBATCBhwYIKwYBBQUHAQEEezB5MCQGCCsGAQUFBzABhhhodHRwOi8vb2Nz
cC5kaWdpY2VydC5jb20wUQYIKwYBBQUHMAKGRWh0dHA6Ly9jYWNlcnRzLmRpZ2lj
ZXJ0LmNvbS9EaWdpQ2VydEVDQ0V4dGVuZGVkVmFsaWRhdGlvblNlcnZlckNBLmNy
dDAMBgNVHRMBAf8EAjAAMIIBfgYKKwYBBAHWeQIEAgSCAW4EggFqAWgAdgDuS723
dc5guuFCaR+r4Z5mow9+X7By2IMAxHuJeqj9ywAAAWyonV3PAAAEAwBHMEUCIF3g
T8yW3lwnxvFr4vsc9TvOKS5rM+X4esKSH+AT6TCkAiEA475K4Ri5g7xhfxPv6XEB
7t65Zqh1CAS17f2bw98b8G0AdwBWFAaaL9fC7NP14b1Esj7HRna5vJkRXMDvlJhV
1onQ3QAAAWyonV4DAAAEAwBIMEYCIQDlCJ+XkYD9XQkM3Zf38Gbmz7Gj9KYGsL5+
IEA9sj+OgwIhAJtW6p5flvKKiSYwNoJ4Z1VwkegwAw6CXZ4bUzgLOplhAHUAh3W/
51l8+IxDmV+9827/Vo1HVjb/SrVgwbTq/16ggw8AAAFsqJ1fTQAABAMARjBEAiBt
MXCcIJiDyDz51bdHqVmK/DNe7rsYK1+u0otoQgO33wIgCCW58PNHcNSNzprMoEBy
ZEkhOeEDevLHA3p+uoWO3/cwCgYIKoZIzj0EAwMDaAAwZQIwcJ3S05to+FZj82bL
hqZAu49BJNbJJs55uuAthlJKCbmpo5jWAYaOvGx62/ke8MOQAjEAmtAFULFAr9MD
fpF0uSDLIfGZles2F1rsyIbiv6YPlTF3X8ZgMXesJaa64c2yvhoR
-----END CERTIFICATE-----
```

Is there any other proof of concept or evidence that can assist us with further analysis? Does the library need to be configured in a specific way to reproduce this?

Warm Regards,

K

ArtBlnd, 2020-05-19:
You can reproduce this issue by signing certificate using `wolfSSL_X509_sign`. please check explanation code blow.

ArtBlnd, 2020-05-19:
I can provide patch for this issue if you want. @kaleb-himes I already fixed in my own branch.

kaleb-himes, 2020-05-19:
@ArtBlnd,

Thank you so much for clarifying. So this is only when someone were to take a cert with the noted elements and attempt to copy the cert then sign it, so not something one would ever encounter in a TLS connection etc. Is that correct?

I will work to setup a test application to reproduce the issue of duplicating the cert then signing the cert locally. 

Warm Regards,

K

kaleb-himes, 2020-05-19:
@ArtBlnd,

Please don't send a patch we will fix! The steps to reproduce were not clear before, we ran some tests but nothing ever turned up. Thank you for noting this is when copying the cert to a new cert structure and then attempting to sign the cert with a new CA or intermediate CA, not something that happens when just processing the cert in a TLS connection.

Warm Regards,

K

ArtBlnd, 2020-05-19:
@kaleb-himes I think it only affects when signing X509 type certificate. I thought it was serious problem, but not really :)

kaleb-himes, 2020-05-28:
Just an update on this issue it is still being worked on and will result in a new example to the https://github.com/wolfssl/wolfssl-examples.git repo, I'll post a link here when it is finished. The exact issue reported has not yet been reproduced but hopefully the example will help to expose the issue when completed. 

This work might also result in a new and helpful API addition to the library proper, will update this issue with details accordingly.

- KH

kaleb-himes, 2020-05-29:
@ArtBlnd,

After spending a few hours on this I have some questions:

1) When you sign are you re-signing with the same old signer or are you also updating the signer?
2) If updating the signer how are you updating the authority KEY ID?
3) Did this work previously and only fails with the cert noted?
4) What is the purpose of changing out the keys?

I have attached a test app here it works, theres' no crash or issue using normal certs but I need a baseline working case before i can test the fail case. Unfortunately there seem to be functionality gaps, I had to add a function to update the authorityKeyId if I try using a signer other than the signer that originally signed the cert etc. In the attachment you'll find a folder "patches" with ssl.c.patch and ssl.h.patch you'll need to apply to wolfssl/src/ssl.c and wolfssl/wolfssl/ssl.h before you can build this test app.

Can you take a look at the example I've worked up and provide some notes/feedback on how I may have done things differently than you did and let me know if this ever worked in the first place? As of right now it seems any cert re-keyed using the API path you pointed out ends up resulting in a non-usable cert.

Warm Regards,
[tmpdir.zip](https://github.com/wolfSSL/wolfssl/files/4702831/tmpdir.zip)

K

ArtBlnd, 2020-05-30:
1, 2. We updated issuer name using X509_set_issuer_name
3. no, if its EV certification.
4. MITM.

I'll try ASAP.

ArtBlnd, 2020-06-01:
You'll have to check that buffer you got is from UB. writes on buffer is correctly working. but size is incorrect.

kaleb-himes, 2020-06-01:
@ArtBlnd,

As of right now we are unable to achieve a working example without API additions and changes to existing APIs'. From what we can determine this would not result in a working or functioning certificate regardless of the certificate source. Simply updating the subject name is insufficient for a MITM attack, other parts of the cert would also need updated and the whole cert would need to be re-generated. You might be able to accomplish such a proof of concept using DecodedCertificate structures but the X509 path is proving to have many functional gaps and would be considered "Not Supported" at this time.

I have tested the above test application using both the wolfssl-4.4.0/certs/server-cert.pem and the cert that reportedly causes an issue. I have done all testing so far using github master repository and there are no buffer overflows or underflows. I am not able to reproduce the reported issue in the latest development branch so either the issue has already been resolved or something else was incorrect in the original application.

If you can provide an example like the one shared above that fully functions and reproduces the noted issue with the github master sources we would be happy to evaluate it.

Warmest Regards,

K

kaleb-himes, 2020-06-02:
@ArtBlnd,

Will leave open for another 30 days to give ample time for an example that reproduces the issue in case you are busy on other tasks.

- KH

haydenroche5, 2021-01-13:
I'm closing this, since the 30 days Kaleb noted above has lapsed. Feel free to reach out again if there's an issue to discuss. Thanks!

closed 2021-01-13 by haydenroche5