wolfSSL/wolfssl #2751: typedef Timeval still exist in WolfSSL public header, even after #2552 was closed

vaintroub, 2020-01-21:
one of the complaints in #2552 was that wolfssl defines struct Timeval, and this clashes with user defined type.
#2552 was closed as fixed, but the typedef is still there  https://github.com/wolfSSL/wolfssl/blob/master/wolfssl/callbacks.h#L56

Suggestion : use namespace prefix WolfSSL_  for the wolfssl types included into public headers , that can clash with user defined types. The namespace is already used consistently for the exported functions.

JacobBarthelmeh, 2020-01-29:
This PR (https://github.com/wolfSSL/wolfssl/pull/2771) is under review for changing the name. Where it is in a public header the review may resulting in solving it a different way, but we can work towards resolving the name clash one way or another.

JacobBarthelmeh, 2020-02-17:
Adding the define NO_OLD_TIMEVAL_NAME to builds will now make wolfSSL use the structure name WOLFSSL_TIMEVAL instead of Timeval. This keeps backward support by default and allows for users with a name clash to add in the macro in order to not use the name Timeval in wolfSSL.

closed 2020-02-17 by JacobBarthelmeh