wolfSSL/wolfssl #2753: Wrong ASIO error message

RedDwarf69, 2020-01-22:
I have an ASIO client and an ASIO server that have been working fine with openSSL. When trying to use wolfSSL instead in the client the server (still using openSSL) gets a `sslv3 alert bad certificate` error, and the client a `non-blocking socket write buffer full` error.

Looking at Wireshark I see the client has sent a perfectly fine Bad Certificate alert. So the `non-blocking socket write buffer full` error doesn't make sense.

I'm using Boost 1.72 and wolfSSL 4.3.0.

dgarske, 2020-01-24:
Hi @RedDwarf69 , This appears to be an issue with how the Boost/Asio port is handling the non-blocking responses for WANT_WRITE/WANT_READ. @kaleb-himes is assigned to investigate. In the meantime can you tell us a bit more about your project? Feel free to email us directly at support@wolfssl.com Thanks, David Garske

RedDwarf69, 2020-01-24:
It should be easy to reproduce. I was doing a silly client test and wasn't even using `set_verify_mode`(). With openSSL that defaults to verify_none, with wolfSSL it defaults to verify_peer. Since I had created a certificate in the server side that the client had no way to verify the verification failed... and the wrong error message was used.

So you could argue the different verification default is another issue. But you could also argue `verify_peer` is a better default.

In any case there seems to be a long standing issue with the ASIO SSL implementation. It uses way too much memory, independently of the SSL library being used. Which is the reason I was trying wolfSSL to start with.
See https://sourceforge.net/p/asio/mailman/message/21135160/, https://stackoverflow.com/questions/21003761/reducing-memory-usage-per-connection-with-boostasio-ssl-server and https://github.com/boostorg/asio/blob/develop/include/boost/asio/ssl/detail/stream_core.hpp#L39.

Until the ASIO issue is solved I have just ended up using nginx as a reverse proxy taking care of the SSL and removing all the SSL code from my own server code. It needs to handle a lot of connections and ASIO memory usage is a show-stopper.

kaleb-himes, 2020-02-03:
@RedDwarf69,

Thank you for the update. 

> So you could argue the different verification default is another issue. But you could also argue verify_peer is a better default.

Yes we would definitely argue that verify_peer is a better default. A large portion of data loss and successful espionage attempts can be traced back to poor implementation and weak security practices as opposed to actual exploits like heart-bleed. Even if it causes a bit of an initial frustration getting setup wolfSSL will ALWAYS use best security practices as defaults to help protect users against poor implementations. Using a weak default setting for the sake of simplicity is just bad practice and we care about our users and consumers too much to allow for a weak default to be used!

I will go ahead and close this since the root cause was traced back to ASIO overhead but if you end up re-visiting wolfSSL for any future applications please do not hesitate to reach out to support [at] wolfssl [dot] com anytime with any questions.

Regards,

K

closed 2020-02-03 by kaleb-himes