wolfSSL/wolfssl #2758: make check failed (unit test - certificate) version 4.3.0

brymatm, 2020-01-22:
Ubuntu 18.0.4
No errors during make
All tests passed during ./wolfcrypt/test/testwolfcrypt after updated hash.
During make check:
FAIL: tests/unit.test

Some examples from the test-suite.log:

Signature failure test: ECC: Ret -155
Signature failure test: RSA: Ret -155

could not verify the cert: certs/test-pathlen/chainH-ICA1-pathlen0.pem
Error: (-238): ASN CA path length larger than signer error

trying server command line[331]: SuiteTest -a -v 2 -l DHE-RSA-AES256-GCM-SHA384 -H exitWithRet -2 -p 0 
Skip exit() for testing
trying client command line[332]: SuiteTest -a -v 2 -l DHE-RSA-AES256-GCM-SHA384 -H exitWithRet -2 -p 41671 
Skip exit() for testing
SSL_accept error -501, can't match cipher suite
wolfSSL_connect error -308, error state on socket

trying server command line[332]: SuiteTest -v 3 -l ECDHE-RSA-AES128-GCM-SHA256 -c ./certs/test/server-cert-rsa-badsig.pem -2 -p 0 
trying client command line[333]: SuiteTest -v 3 -l ECDHE-RSA-AES128-GCM-SHA256 -j -2 -p 35947 
In verification callback, error = -188, ASN no signer error to confirm failure
	Peer certs: 2
	Subject's domain name at 1 is www.wolfssl.com
	Allowing failed certificate check, testing only (shouldn't do this in production)
SSL version is TLSv1.2
SSL cipher suite is TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256
SSL curve name is FFDHE_2048

trying server command line[362]: SuiteTest -v 3 -l ECDHE-RSA-AES128-GCM-SHA256 -k ./certs/server-key.pem -c ./certs/test/server-goodcn.pem -d -2 -p 0 
trying client command line[363]: SuiteTest -v 3 -l ECDHE-RSA-AES128-GCM-SHA256 -h localhost -A ./certs/test/server-goodcn.pem -m -C -2 -p 32771 
wolfSSL error: getaddrinfo failed
FAIL tests/unit.test (exit status: 1)

Attached tar file contains output of configure, make, make check, and test-suite.log.
[make-check-issue.tar.gz](https://github.com/wolfSSL/wolfssl/files/4100414/make-check-issue.tar.gz)

kaleb-himes, 2020-01-23:
Can you clarify if you are testing fips-ready and what the configure settings were that were being used?

Warm Regards,

K

brymatm, 2020-01-23:
Apologies, I should have mentioned that the configure output in the attachment also contained the command used.  Here it is, though:

 ./configure --enable-fips=v2 --enable-ipv6 --enable-aesccm --enable-aesgcm

After I submitted the issue I ran the process on a different system (also Ubuntu 18.04) and it ran successfully, so it is likely this is an issue with my system's configuration.

kaleb-himes, 2020-02-03:
@brymatm,

It could be. One thing to be aware of is the difference in static and shared linking to an application. Since the FIPS module has to perform a sanity check on itself if it moves in memory the hash will change. What this means is that if you are building/linking statically to two seperate applications, application A might place the module in a different location in memory than application B meaning you would need to:

Step 1) compile Application A, run with the fips callback to get a new hash, update the hash, recompile application A.
Step 2) Compile Application B, run with the fips callback to get a new hash, update the hash, recompile application B.

If using a shared object library instead of static you could compile and link application A and run with the fips callback to get a new hash, update the hash one time and then compile both applications A and B and the hash shouldn't change after that. Not sure if that helps but if your one system is statically linking you will experience frequent updates to the hash since every time you make a change to the app, the module location in memory will also move to allow for the new code in the application, this causes the hash to change throughout the development cycle until no further changes are made to the application. Linking to a shared object can reduce this constant "Change app, compile and run, get new hash, update, compile and run again" cycle.

Warm Regards,

K

kaleb-himes, 2020-04-10:
Marking as resolved.

closed 2020-04-10 by kaleb-himes