wolfSSL/wolfssl #2766: OpenSSL compatibility layer: SSL_CTX_set1_groups_list

miwob, 2020-01-24:
Consider adding SSL_CTX_set1_groups_list to the OpenSSL compatibility layer.

See https://www.openssl.org/docs/man1.1.1/man3/SSL_CTX_set1_groups_list.html

This function has been introduced in 1.1.1.

JacobBarthelmeh, 2020-01-29:
Hi miwob, thanks for the request! We've added that to the feature request list without a set completion time. There is some similar API that gets close like wolfSSL_CTX_set1_curves_list (SSL_CTX_set1_curves_list) and a native API of wolfSSL_CTX_set_groups currently in wolfSSL. Can you tell us some about the project?

miwob, 2020-01-29:
Thanks.

Yes, wolfSSL_CTX_set1_curves_list is what we use currently in the wolfSSL module.

I have added wolfSSL support to ircd-hybrid recently to offer a broader variety of supported TLS implementations. ircd-hybrid is a popular internet relay chat daemon that, for example, is used on OFTC, Rizon, or 2600.net.

Code can be found here https://github.com/ircd-hybrid/ircd-hybrid/blob/master/src/tls_wolfssl.c

JacobBarthelmeh, 2020-01-29:
Cool, thanks for sharing! I'll bring the project up internally, sometimes we bump the feature request to a higher priority on the list for open source projects.

I see you are listed as being in Germany, if you are in the area we will have a booth at "Automotive Ethernet Congress (2/12 - 2/13)" and "Embedded World 2020 (2/25-2/27)", feel free to stop by.

JacobBarthelmeh, 2020-01-29:
@miwob the feature request has been bumped up in priority, one of our engineers will get to the API after finishing up the current paid consulting projects they are on. Also I forgot to mention some of our engineers will be at FOSDEM too if you are able to meet up there? If not we could open an IRC chat to provide some assistance with support and performance optimizations.

julek-wolfssl, 2020-01-30:
Hi @miwob ,
thanks for your interest in wolfSSL! I had some free time today and implemented this API. You can find and test it here: #2776 . Any feedback you would have is greatly appreciated.

miwob, 2020-01-31:
Thank you very much for implementing it that fast. Excellent customer service.
I'll try to test out everything this weekend.

Unfortunately these locations are too far away from here for short drop by. Maybe next time if it's somewhere nearer.

Anyway, your help and support is greatly appreciated!

julek-wolfssl, 2020-02-03:
Hi @miwob, just an update: the new API has been merged into the master branch.

closed 2020-03-05 by JacobBarthelmeh