wolfSSL/wolfssl #2782: Clean away Windows carriage return characters from file

ottok, 2020-02-03:
In certain situations you may end up being unable to reset/commit the repository:
```
$ git status
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   extra/wolfssl/wolfssl/IDE/CRYPTOCELL/main.c
	modified:   extra/wolfssl/wolfssl/IDE/CSBENCH/user_settings.h
	modified:   extra/wolfssl/wolfssl/IDE/Renesas/e2studio/Projects/test/src/key_data.c
	modified:   extra/wolfssl/wolfssl/IDE/Renesas/e2studio/Projects/test/src/key_data.h
	modified:   extra/wolfssl/wolfssl/IDE/VS-AZURE-SPHERE/client/client.h
	modified:   extra/wolfssl/wolfssl/IDE/VS-AZURE-SPHERE/server/server.h
	modified:   extra/wolfssl/wolfssl/mqx/util_lib/Sources/util.h
	modified:   extra/wolfssl/wolfssl/mqx/wolfcrypt_benchmark/Sources/main.c
	modified:   extra/wolfssl/wolfssl/mqx/wolfcrypt_benchmark/Sources/main.h
	modified:   extra/wolfssl/wolfssl/wolfssl/wolfcrypt/port/ti/ti-ccm.h

no changes added to commit (use "git add" and/or "git commit -a")

$ git reset --hard
HEAD is now at 6628eb090 WIP: mysql-test contents

$ git status
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   extra/wolfssl/wolfssl/IDE/CRYPTOCELL/main.c
	modified:   extra/wolfssl/wolfssl/IDE/CSBENCH/user_settings.h
	modified:   extra/wolfssl/wolfssl/IDE/Renesas/e2studio/Projects/test/src/key_data.c
	modified:   extra/wolfssl/wolfssl/IDE/Renesas/e2studio/Projects/test/src/key_data.h
	modified:   extra/wolfssl/wolfssl/IDE/VS-AZURE-SPHERE/client/client.h
	modified:   extra/wolfssl/wolfssl/IDE/VS-AZURE-SPHERE/server/server.h
	modified:   extra/wolfssl/wolfssl/mqx/util_lib/Sources/util.h
	modified:   extra/wolfssl/wolfssl/mqx/wolfcrypt_benchmark/Sources/main.c
	modified:   extra/wolfssl/wolfssl/mqx/wolfcrypt_benchmark/Sources/main.h
	modified:   extra/wolfssl/wolfssl/wolfssl/wolfcrypt/port/ti/ti-ccm.h
```

This is because of Windows line endings in those files, and git having problems reading them when trying to diff the files to understand if there has been any changes.

One way to fix it is to set:
```
$ cat .git/info/attributes 
extra/wolfssl/** binary
```

..but then you will not have any diffs of those files. Best solution would be to remove all CR/LF with a simple LF (the Unix way) everywhere.

kaleb-himes, 2020-02-03:
Please see comment regarding contributions from non employees of wolfSSL on #2785.

Thank you again for the suggested changes!

Regards,

K

haydenroche5, 2021-01-13:
I'm making a note of this issue internally. Closing.

closed 2021-01-13 by haydenroche5