wolfSSL/wolfssl #2794: TLS Client v1.3 connect problem with load google root CA

RyukerLiu, 2020-02-12:
I've configured enable tls v1.3
`$ ./configure --enable-tls13`

I'm using `./examples/client/client` to test connect with client.
But when I want to access www.google.com:443 with root CA download from browser, it encounter fail with tls v1.3.
```
allen@allen-X411UA:~/test_wolfssl/wolfssl$ ./examples/client/client -v 4 -h www.google.com -p 443 -g -A ./certs/www-google-com.pem 
wolfSSL_connect error -188, ASN no signer error to confirm failure
wolfSSL error: wolfSSL_connect failed
```

But It's ok with v1.2
```
allen@allen-X411UA:~/test_wolfssl/wolfssl$ ./examples/client/client -v 3 -h www.google.com -p 443 -g -A ./certs/www-google-com.pem 
SSL version is TLSv1.2
SSL cipher suite is TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256
SSL curve name is SECP256R1
SSL connect ok, sending GET...
HTTP/1.0 404 Not Found
Content-Type: text/html; charset=UTF-8
Referrer-Policy: no-referrer
Content-Length: 1571
Date: Wed, 
12 Feb 2020 07:50:58 GMT
```

And I also try to access www.cloudflare.com:443 with root CA download from browser. It seems ok with v1.3 and v1.2
v1.3
```
allen@allen-X411UA:~/test_wolfssl/wolfssl$ ./examples/client/client -v 4 -h www.cloudflare.com -p 443 -g -A ./certs/cloudflare-com.pem 
SSL version is TLSv1.3
SSL cipher suite is TLS_AES_128_GCM_SHA256
SSL curve name is SECP256R1
SSL connect ok, sending GET...
HTTP/1.1 403 Forbidden
Server: cloudflare
Date: Wed, 12 Feb 2020 07:50:51 GMT
Content-Type: text/html
Content-Length: 151

Connection: close
CF-RAY: 563cf95b8cc9f079-TPE

<html>
<head><title>403 Forbidden</title></head>
<body>
<center><h1>403 F
```

v1.2
```
allen@allen-X411UA:~/test_wolfssl/wolfssl$ ./examples/client/client -v 3 -h www.cloudflare.com -p 443 -g -A ./certs/cloudflare-com.pem 
SSL version is TLSv1.2
SSL cipher suite is TLS_ECDHE_ECDSA_WITH_AES_128_GCM_SHA256
SSL curve name is SECP256R1
SSL connect ok, sending GET...
HTTP/1.1 403 Forbidden
Server: cloudflare
Date: Wed, 12 Feb 2020 07:50:47 GMT
Content-Type: text/html
Content-Length: 151

Connection: close
CF-RAY: 563cf94378c745f4-TPE

<html>
<head><title>403 Forbidden</title></head>
<body>
<center><h1>403 F
```

Could anyone help me with this problem?

kaleb-himes, 2020-02-14:
Hi @RyukerLiu,

I just ran a few tests and I'm getting the same result as you:

```
./examples/client/client -h www.google.com -p 443 -g -A GlobalSign-Root-CA-R2.pem -v 3
SUCCESS
```

```
./examples/client/client -h www.google.com -p 443 -g -A GlobalSign-Root-CA-R2.pem -v 4
wolfSSL_connect error -188, ASN no signer error to confirm failure
wolfSSL error: wolfSSL_connect failed
```

I'll do a little digging and see if I can figure out which cert their TLS 1.3 servers are presenting and why it's a different cert than the TLS 1.2 connections.

- K

kaleb-himes, 2020-02-14:
@RyukerLiu,

It appears there is some sort of front-end when you don't send the server name indication extension that tries to handle the connection on behalf of google.com with TLS 1.3 enabled. I don't know what cert you would need for that front end. I was able to connect by sending the servername indication extension with the -S option in the client (NOTE you have to configure wolfSSL with ```--enable-sni --enable-tls13``` I used the same cert as was used in the TLS 1.2 test (GlobalSign-Root-CA-R2.pem):

```
kalebhimes@kalebs-MBP clean-wolfssl % ./examples/client/client -h www.google.com -p 443 -v 4 -g -S www.google.com -A google-certs/GlobalSign-Root-CA-R2.pem 
SSL version is TLSv1.3
SSL cipher suite is TLS_AES_128_GCM_SHA256
SSL curve name is SECP256R1
SSL connect ok, sending GET...
HTTP/1.0 404 Not Found
Content-Type: text/html; charset=UTF-8
Referrer-Policy: no-referrer
Content-Length: 1571
Date: Fri, 
14 Feb 2020 18:19:11 GMT
Alt-Svc: quic=":443"; ma=2592000; v="46,43",h3-Q050=":443"; ma=2592000,h3-Q049=":443"; ma=2592000,h3-
```

Regards,

K

RyukerLiu, 2020-02-17:
Hi @kaleb-himes ,

Thanks for your quick response.
I tried client with SNI, and it works as you said.
But I'm wondering why we will encounter this front-end issue.
When I use openssl client, it works without SNI.

Openssl client with Google IP
```
allen@allen-X411UA:~/temp/wolfssl$ openssl s_client -connect 172.217.7.132:443 -CAfile www-google-com.pem
CONNECTED(00000005)
depth=2 OU = GlobalSign Root CA - R2, O = GlobalSign, CN = GlobalSign
verify return:1
depth=1 C = US, O = Google Trust Services, CN = GTS CA 1O1
verify return:1
depth=0 C = US, ST = California, L = Mountain View, O = Google LLC, CN = www.google.com
verify return:1
---
Certificate chain
 0 s:C = US, ST = California, L = Mountain View, O = Google LLC, CN = www.google.com
   i:C = US, O = Google Trust Services, CN = GTS CA 1O1
 1 s:C = US, O = Google Trust Services, CN = GTS CA 1O1
   i:OU = GlobalSign Root CA - R2, O = GlobalSign, CN = GlobalSign
---
Server certificate
-----BEGIN CERTIFICATE-----
MIIFizCCBHOgAwIBAgIRAONaPtQ2vFsfAgAAAABXm9MwDQYJKoZIhvcNAQELBQAw
QjELMAkGA1UEBhMCVVMxHjAcBgNVBAoTFUdvb2dsZSBUcnVzdCBTZXJ2aWNlczET
MBEGA1UEAxMKR1RTIENBIDFPMTAeFw0yMDAxMjkyMjAxMzZaFw0yMDA0MjIyMjAx
MzZaMGgxCzAJBgNVBAYTAlVTMRMwEQYDVQQIEwpDYWxpZm9ybmlhMRYwFAYDVQQH
Ew1Nb3VudGFpbiBWaWV3MRMwEQYDVQQKEwpHb29nbGUgTExDMRcwFQYDVQQDEw53
d3cuZ29vZ2xlLmNvbTCCASIwDQYJKoZIhvcNAQEBBQADggEPADCCAQoCggEBAPNt
WwSjdzksVVS6+CS1QRPUZdkPTfqEBmckB9GPIHFU77ZLgbAWputNf646deDs6IXE
1uep+MCPBAuT93DXkf6ytBN8EtExqnc21qlLT2Vy6kRu7EhHEwJgmRSj5drBLPcF
zgrBSpMTdtwLxLGYXh36hEdzA+fVfR8plKYrTj2YgdhqaLPWGxjED3bSiDgQJ+SG
nrW76vumCzolgLaAXyAwbtl8a4hSFPJSUHMCiW7o2WqG79jmJ3sKEF7bXfE139xj
7cz10/8H1xNIh0pS3ei+yemoIc7IuMW1tpl8iyIyEunDPFZ/EGtQExoKSIXzgI8U
mYJyHw3Ln09/HsCGGGECAwEAAaOCAlQwggJQMA4GA1UdDwEB/wQEAwIFoDATBgNV
HSUEDDAKBggrBgEFBQcDATAMBgNVHRMBAf8EAjAAMB0GA1UdDgQWBBQ/C+M2YKqe
mfFFGEkwne5/cjqm5TAfBgNVHSMEGDAWgBSY0fhuEOvPm+xgnxiQG6DrfQn9KzBk
BggrBgEFBQcBAQRYMFYwJwYIKwYBBQUHMAGGG2h0dHA6Ly9vY3NwLnBraS5nb29n
L2d0czFvMTArBggrBgEFBQcwAoYfaHR0cDovL3BraS5nb29nL2dzcjIvR1RTMU8x
LmNydDAZBgNVHREEEjAQgg53d3cuZ29vZ2xlLmNvbTAhBgNVHSAEGjAYMAgGBmeB
DAECAjAMBgorBgEEAdZ5AgUDMC8GA1UdHwQoMCYwJKAioCCGHmh0dHA6Ly9jcmwu
cGtpLmdvb2cvR1RTMU8xLmNybDCCAQQGCisGAQQB1nkCBAIEgfUEgfIA8AB2ALIe
BcyLos2KIE6HZvkruYolIGdr2vpw57JJUy3vi5BeAAABb/OJ37AAAAQDAEcwRQIg
Izy0dm/drQEPcDvUBt1wbBE80CX4u8k0NAZgYAg25WICIQCOdeSNM/Y2WkzqELnH
OZZLkcnBsQbc7LSxtDr90JyklAB2AF6nc/nfVsDntTZIfdBJ4DJ6kZoMhKESEoQY
dZaBcUVYAAABb/OJ39QAAAQDAEcwRQIhAPf98OifKy8SM9hcAeLOI8YsOSYmDLhK
oaXAvIU6ZngzAiA1zrQbaMu2HWp3mymweMjx3gN04uifxnpSnd6NXEAEmTANBgkq
hkiG9w0BAQsFAAOCAQEACFKut+n5j391CdcoSerChshlmp64fQKZATDqejVUPg0q
IeKIduUAn95YrNDq4MH9lHAGivaEzUPSGeik7AdvHFxH3UGUFB2xUzS9XebzOvjA
z20rnlX0IUpTlWcl14UEeYx38kuVZ8biYHfzwdZaRL58GZycP/Y59DyinIriCP3m
lqVVHmN3rREG86PC8dCaHD/SEpaoxLysbsporR23YsXQdNvJDdKIjux+VHVnR9wP
uOlStifAqpZVqxshG4E2P4N0NfMcu3fb3nxD+8WDOTQEdmYgTyPWb9S7ukGSnXMy
C9l1OcauADkuxNo8LEKcc+nHxyjJGFX1kHzQuyL9Hw==
-----END CERTIFICATE-----
subject=C = US, ST = California, L = Mountain View, O = Google LLC, CN = www.google.com

issuer=C = US, O = Google Trust Services, CN = GTS CA 1O1

---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: RSA-PSS
Server Temp Key: X25519, 253 bits
---
SSL handshake has read 3020 bytes and written 395 bytes
Verification: OK
---
[... 55 lines cut ...]
VL6en7KJ5JLIfeoNJbK32dHe2rrE
-----END CERTIFICATE-----
subject=C = US, ST = California, L = Mountain View, O = Google LLC, CN = www.google.com

issuer=C = US, O = Google Trust Services, CN = GTS CA 1O1

---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: ECDSA
Server Temp Key: X25519, 253 bits
---
SSL handshake has read 2634 bytes and written 396 bytes
Verification: OK
---
New, TLSv1.3, Cipher is TLS_AES_256_GCM_SHA384
Server public key is 256 bit
Secure Renegotiation IS NOT supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
Early data was not sent
Verify return code: 0 (ok)
---
```

kaleb-himes, 2020-02-17:
@RyukerLiu,

I suspect OpenSSL is sending the SNI extension by default where wolfSSL it is something that has to be turned on and specified. My theory would be that the below test cases have been run:

OpenSSL + TLS 1.2 (with SNI) - Success
OpenSSL + TLS 1.3 (with SNI) - Success
wolfSSL + TLS 1.2 (no SNI) - Success
wolfSSL + TLS 1.3 (no SNI) - Fail
wolfSSL + TLS 1.3 (with SNI) - Success

You can easily confirm this by capturing a wireshark trace of the connection attempts from OpenSSL and checking for the extension (See attached screenshot)

[image]

kaleb-himes, 2020-02-17:
@RyukerLiu,

Actually just checked it myself, I missed that you had included a test with just the IP (which would in effect not be sending SNI by default!)

I would agree with your underlying suspicion that it is not purely the SNI extension that is causing success in the TLS 1.3 connection tests with OpenSSL. What is very interesting however is that in the wolfSSL test it is using the final TLS 1.3 draft where the OpenSSL test I ran appears to be using one of the draft forms. Do you know if the OpenSSL version you are testing with is using the final TLS1.3 spec or is it also using a draft form of TLS 1.3?

[image]

RyukerLiu, 2020-02-18:
Hi @kaleb-himes,

I'm not sure how to check which draft version my openssl use.
But I try to use wireshark to trace the OpenSSL test and the version seems same as your test.
Does this mean my openssl test also using a draft form of TLS 1.3?
 
[image]

closed 2021-01-13 by kaleb-himes