wolfSSL/wolfssl #2816: OpenSSL PKCS7_SIGNER_INFO API

bgermann, 2020-02-21:
Please add the OpenSSL API to the compatibility layer that is related to PKCS7_SIGNER_INFO.

JacobBarthelmeh, 2020-02-24:
Hi bgermann, thanks for the request! Adding all API related to PKCS7_SIGNER_INFO represents a significant amount of work, can you tell us some about the use case or narrow down the API's requests?

bgermann, 2020-02-24:
This is to resolve https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=952399 but maybe this gets resolved otherwise.

JacobBarthelmeh, 2020-03-05:
Thanks for the feedback on this, we are reviewing the feature request internally. In the mean time it has been added to the feature request list.

bgermann, 2020-10-19:
Debian changed its position on linking with OpenSSL, so the kmod issue is not relevant anymore.

closed 2020-10-19 by bgermann