wolfSSL/wolfssl #2819: Review architecture compatiblity patches written in Debian

ottok, 2020-02-24:
Hello!

I wanted to let you know that there appears to exists a couple of patches in the WolfSSL Debian package. Browse them at https://sources.debian.org/src/wolfssl/4.3.0+dfsg-2/debian/patches/

They all seem to be related to architecture compatibility. Indeed, on Debian they manage to build successfully WolfSSL on all platforms:
[image]
https://buildd.debian.org/status/package.php?p=wolfssl

You might want to incorporate these upstream, perhaps slightly rewritten or something.

kaleb-himes, 2020-02-25:
Thanks @ottok! I sent a note to our developer that works with the debian release team to see if they have asked us to pull in those changes locally or if that is something that they need for say systemX but would break builds on systemY.

Regards,

K

lechner, 2020-02-26:
Hi @ottok @kaleb-himes I am not sure upstream should accept these patches. Over the years, @cconlon and @dgarske accepted many of my patches. This set should probably stay in Debian.

In truth, I would actually like to get rid of `turn-off-fastmath-for-amd64.patch`, which makes the library slower, but enabling `fastmath` vastly complicates the Debian symbol tracking for your shared library because they would then differ by architecture.

Kind regards from the RSA conference,
Felix

kaleb-himes, 2020-02-26:
@lechner, Thank you for your input Felix!

I'll mark this as resolved and we will continue to work and support the Debian team in any way we can. If there is an approach we could take on our end for the fastmath stuff we should go over it! Also the next time you're talking to @cconlon or @dgarske lets' explore the --enable-sp as a potential replacement for --disable-fastmath. The single-precision work we did this last year is even better than fastmath and maybe we can get around some of those issue while providing a more-performant default!

- K

closed 2020-02-26 by kaleb-himes