wolfSSL/wolfssl #2820: build errors with NO_ASN_TIME and OPENSSL_EXTRA

damianslee, 2020-02-24:
I'm using #define NO_ASN_TIME as a temporary work around with time not initialized yet on a embedded system.

I get build errors in ssl.c and asn.c, and have to include <time.h> in user_settings.h to get it to compile

ssl.c(23607,17): error C2079: 'ts' uses undefined struct 'tm'
ssl.c(23622,23): warning C4013: 'XTIME' undefined; assuming extern returning int
ssl.c(23629,5): error C2027: use of undefined type 'tm'
\wolfSSL\wolfssl\wolfcrypt\asn.h(1069): message : see declaration of 'tm'
ssl.c(23635,12): error C2224: left of '.tm_year' must have struct/union type
ssl.c(23636,12): error C2224: left of '.tm_year' must have struct/union type
ssl.c(23637,16): error C2224: left of '.tm_year' must have struct/union type
ssl.c(23638,16): error C2224: left of '.tm_year' must have struct/union type
ssl.c(23644,12): error C2224: left of '.tm_year' must have struct/union type
ssl.c(23645,12): error C2224: left of '.tm_year' must have struct/union type
ssl.c(23646,12): error C2224: left of '.tm_year' must have struct/union type
ssl.c(23647,12): error C2224: left of '.tm_year' must have struct/union type
ssl.c(23648,12): error C2224: left of '.tm_year' must have struct/union type
ssl.c(23656,12): error C2224: left of '.tm_mon' must have struct/union type
ssl.c(23657,12): error C2224: left of '.tm_mon' must have struct/union type
ssl.c(23658,12): error C2224: left of '.tm_mday' must have struct/union type
ssl.c(23659,12): error C2224: left of '.tm_mday' must have struct/union type
ssl.c(23660,12): error C2224: left of '.tm_hour' must have struct/union type
ssl.c(23661,12): error C2224: left of '.tm_hour' must have struct/union type
ssl.c(23662,12): error C2224: left of '.tm_min' must have struct/union type
ssl.c(23663,12): error C2224: left of '.tm_min' must have struct/union type
ssl.c(23664,12): error C2224: left of '.tm_sec' must have struct/union type
ssl.c(23665,12): error C2224: left of '.tm_sec' must have struct/union type
ssl.c(23668,21): warning C4013: 'XGMTIME' undefined; assuming extern returning int
ssl.c(23668,35): warning C4047: '=': 'tm *' differs in levels of indirection from 'int'

dgarske, 2020-02-25:
Hi @damianslee,

Some embedded standard libraries do not support "struct tm". So we have a way to define it ourselves and override it with your own RTC source.

Here is an example for how to build with XTIME overridden manually:
https://github.com/wolfSSL/wolfssl/blob/master/IDE/GCC-ARM/Header/user_settings.h#L449
https://github.com/wolfSSL/wolfssl/blob/master/IDE/GCC-ARM/Source/wolf_main.c#L32

Thanks,
David Garske, wolfSSL

damianslee, 2020-02-25:
i have time support.   I was just trying to disable time support in the short term in wolfssl with NO_ASN_TIME.   

Seems there are parts of OPENSSL_EXTRA x509 support which need some additional #ifndef NO_ASN_TIME ?

dgarske, 2020-02-25:
Hi @damianslee ,

I see, you are trying to disable time support with `OPENSSL_EXTRA` defined. Let me see if I can track that down for you.

Thanks,
David Garske, wolfSSL

PR #2822 (2020-03-05): Fixes for building NO_ASN_TIME with OPENSSL_EXTRA
Tested with `./configure --enable-opensslextra CFLAGS="-DNO_ASN_TIME"`

Fixes #2820
commit 92114fef (2020-03-05): Fixes for building NO_ASN_TIME with OPENSSL_EXTRA. Fixes #2820.
* `./configure --enable-opensslextra CFLAGS="-DNO_ASN_TIME"`
closed 2020-03-05 by toddouska