wolfSSL/wolfssl #2828: Incorrect _fp_exptmod() is passing makes WolfSSL crash on s390x

ottok, 2020-02-27:
While packaing MariaDB 10.4 for Debian, which includes WolfSSL, I noticed mysqld did build on s390x, but always crashed when started. Stacktrace form crash:
```
stack_bottom = 0x0 thread_stack 0x49000
--
/<<PKGBUILDDIR>>/builddir/sql/mysqld(my_print_stacktrace+0x2a)[0x12afc293a]
mysys/stacktrace.c:270(my_print_stacktrace)[0x12a9c7ca4]
[0x3ffed0f3f0e]
sql/signal_handler.cc:209(handle_fatal_signal)[0x12af9dfce]
src/tfm.c:3642(fp_copy)[0x12af9fa8c]
src/tfm.c:1601(_fp_exptmod)[0x12afa044e]
src/tfm.c:2277(fp_exptmod)[0x12afa0aa0]
src/tfm.c:4021(fp_prime_miller_rabin_ex)[0x12afa10f6]
src/tfm.c:4283(mp_prime_is_prime_ex)[0x12af8393a]
src/dh.c:2108(_DhSetKey)[0x12af844b6]
src/dh.c:2154(wc_DhSetCheckKey)[0x12af2dade]
src/ssl.c:1744(wolfSSL_CTX_SetTmpDH)[0x12af5083e]
src/ssl.c:38473(wolfSSL_CTX_set_tmp_dh)[0x12b025eba]
vio/viosslfactories.c:340(new_VioSSLFd)[0x12b0260cc]
vio/viosslfactories.c:413(new_VioSSLAcceptorFd)[0x12a6cab58]
/lib/s390x-linux-gnu/libc.so.6(__libc_start_main+0x10a)[0x3ff88f2a90a]
/<<PKGBUILDDIR>>/builddir/sql/mysqld(+0x5bcae4)[0x12a6bcae4]
[0x0]
```

From Marko Mäkelä, InnoDB lead developer and C/C++ superguru:
> It seems to me that the pointer that _fp_exptmod() is passing as the first parameter of fp_copy() is bad on this architecture for some reason. It seems to me that adding -DWC_NO_CACHE_RESISTANT to the CMAKE_C_FLAGS on this architecture could address the problem.

His analysis was correct, or at least setting `DWC_NO_CACHE_RESISTANT` indeed fixed the issue.

You might want to engineer a proper issue so that WolfSSL always runs correctly on s390x.

dgarske, 2020-02-27:
Hi @ottok ,

Can you provide the `./configure` build options, wolfSSL version used and the DH key size? 

Thanks,
David Garske, wolfSSL

ottok, 2020-02-27:
Here is the full build log where we generated the stack trace: https://launchpadlibrarian.net/466692682/buildlog_ubuntu-focal-s390x.mariadb-10.4_1%3A10.4.12-1~ubuntu20.04.1~1582728811.ab0850b0a_BUILDING.txt.gz

Here is the related bug report: https://jira.mariadb.org/browse/MDEV-21705

WolfSSL version is 4.3.0. I don't know the DH key size, maybe you see it in sources at https://salsa.debian.org/mariadb-team/mariadb-10.4/-/tree/bugfix/s390x-build ?

grooverdan, 2020-03-23:
configure isn't used - there's an explicit CMakefile for MariaDB's importing of WolfSSL - https://salsa.debian.org/mariadb-team/mariadb-10.4/-/blob/bugfix/s390x-build/extra/wolfssl/CMakeLists.txt

wolfssl - seems to be 4.3.0 - https://salsa.debian.org/mariadb-team/mariadb-10.4/-/tree/bugfix/s390x-build/extra/wolfssl/wolfssl

dh key size is 2048 - https://salsa.debian.org/mariadb-team/mariadb-10.4/-/blob/bugfix/s390x-build/vio/viosslfactories.c#L340 

Looks like you should be able to get a s390x VM to test this
https://developer.ibm.com/linuxone/

There's a developer forum for help:
https://community.ibm.com/community/user/ibmz-and-linuxone/groups/community-home?CommunityKey=87042487-940e-49cc-ae6d-5a5b76c5ea7e

Above mention is to gain a bit of attention.

dgarske, 2020-03-25:
Hi @grooverdan ,

Thanks for those links and details. I will review and see what we can do.

Thanks,
David Garske, wolfSSL

haydenroche5, 2021-01-13:
I think this issue may be fixed in our latest release, 4.6.0. Talking with the wolfSSL team, we think this may have been caused by a stack issue or something related to the assembly code in one of our math libraries. Feel free to give the latest release a try. In the meantime, I'm going to close this. You can open a new issue or email us at support@wolfssl.com if there are any lingering issues.

closed 2021-01-13 by haydenroche5

grooverdan, 2021-08-12:
Belated thanks, and feedback:

https://buildbot.mariadb.org/#/builders/309/builds/534/steps/6/logs/stdio was built with wolfssl-4.8.0 and runs through a set of s390x ssl based tests without and issue.