wolfSSL/wolfssl #2832: Why I cannot find v4.3.0-stable git tag in source code?

awsa2ron, 2020-02-29:
Hi,
Maybe this is a silly question. Why I cannot find v4.3.0-stable git tag in source code? And how can I checkout the v4.3.0-stable source code? Because I want to use the new added libwebsockets feature.

Thanks!

embhorn, 2020-03-02:
Hello @awsa2ron 

Thanks for contacting wolfSSL Support. 
Instructions for checking out a tag:

1. clone the wolfSSL repository `git clone https://github.com/wolfSSL/wolfssl.git`
2. `cd wolfssl`
3. Fetch the tags that are not normally pulled down `git fetch && git fetch --tags`
4. Checkout the release tag to a new branch `git checkout -b v4.3.0-stable`

Hope that helps,
Eric Blankenhorn 
wolfSSL Support

awsa2ron, 2020-03-02:
Hi,

I can see v4.3.0-stable after git fetch -–tags
Thank you very much.

closed 2020-03-02 by embhorn