wolfSSL/wolfssl #2860: pkcs12

goldenscale, 2020-03-20:
Use wc_i2d_PKCS12,convert WC_PKCS12 struct to allocated DER buffer, import the p12 file into windows, show password error.

kaleb-himes, 2020-03-25:
@goldenscale,

Thank you for reaching out, can you send over a simple test application (1 - 50 lines) that outputs the PKCS12 which then fails to load in windows so we can reproduce the issue on our end for testing?

Thanks!

- K

dgarske, 2020-03-25:
Hi @goldenscale,

Also you might try using the fixes in this recent PR:
https://github.com/wolfSSL/wolfssl/pull/2868

We discovered a bug that could be the reason for the error.

There is PKCS12 test code here:
https://github.com/wolfSSL/wolfssl-examples/tree/master/crypto/pkcs12

Thanks,
David Garske, wolfSSL

goldenscale, 2020-03-30:
Hi @dgarske, @kaleb-himes,

Thank you for your response!

I used related functions and encountered an error in the processing of the ECC algorithm.  I think there is something wrong with the related length handling.

Some example code likes:
......

	int ret = wc_ecc_init_ex(&genKey, HEAP_HINT, 0);
	if (ret != MP_OKAY) return ret;

	ret = wc_ecc_make_key_ex(&rng, 48, &genKey, ECC_SECP384R1);

	if (ret != MP_OKAY)
	{
		wc_ecc_free(&genKey);
		return ret;
	}
	
......

	WC_PKCS12* pkcs12 = wc_PKCS12_create((char*)passwd, passwdlen, NULL, 
		ecc_prvkey_der, ecc_prvkey_der_len, derCert, derSz, NULL, PBE_SHA1_DES3, PBE_SHA1_DES3, 0,
		/*WC_PKCS12_MAC_DEFAULT*/
		2048, 0, NULL);
	if (NULL == pkcs12)
	{
		wc_ecc_free(&genKey);
		return PARAM_ERR;
	}

	word32 certSz = 0;
	byte *certDer = NULL;

	certSz = wc_i2d_PKCS12(pkcs12, &certDer, NULL);
	if (!((certSz > 0) && (certDer != NULL))) {
		wc_ecc_free(&genKey);
		wc_PKCS12_free(pkcs12);
		if (certDer != NULL) {
			XFREE(certDer, NULL, DYNAMIC_TYPE_PKCS);
		}
		return PARAM_ERR;
	}

......

Actually, I solved the problem using the fixes in this recent PR: [# 2855.](https://github.com/wolfSSL/wolfssl/pull/2855/files?file-filters%5B%5D=)

I changed the code in asn.c, and I used the new pkcs12.c file. Now they work fine.

thank you all! 

goldenscale

closed 2020-03-30 by goldenscale