wolfSSL/wolfssl #2870: comparison of array 'ecc_sets[i].name' not equal to a null pointer is always true [-Werror,-Wtautological-pointer-compare]

vaintroub, 2020-03-25:
I get this when compiling with clang

````
..\extra\wolfssl\wolfssl\src\ssl.c(36383,54): error: comparison of array 'ecc_sets[i].name' not equal to a null pointer is always true [-Werror,-Wtautological-pointer-compare]
    for (i = 0; ecc_sets[i].size != 0 && ecc_sets[i].name != NULL; i++);
                                         ~~~~~~~~~~~~^~~~    ~~~~
````

Indeed, name is not a pointer, it is an array with size 16, so it cannot be NULL.

Also, there is an error trying to use uint8_t, without including <stdint.h>.

..\extra\wolfssl\wolfssl\src\ssl.c(24156,5): error: unknown type name 'uint8_t'; did you mean 'wint_t'?
    uint8_t len = MAX_DESCRIPTION_SZ-1;
    ^~~~~~~

Here is a diff that fixes these, when applied to 26f539400a4568bf551887a2683d13f300250350

````diff
diff --git a/src/ssl.c b/src/ssl.c
index f47759df5..0aa611b53 100644
--- a/src/ssl.c
+++ b/src/ssl.c
@@ -24153,7 +24153,7 @@ int wolfSSL_sk_CIPHER_description(WOLFSSL_CIPHER* cipher)
     const char* name;
     const char *keaStr, *authStr, *encStr, *macStr, *protocol;
     char n[MAX_SEGMENTS][MAX_SEGMENT_SZ] = {{0}};
-    uint8_t len = MAX_DESCRIPTION_SZ-1;
+    unsigned char len = MAX_DESCRIPTION_SZ-1;
     const CipherSuiteInfo* cipher_names;
     ProtocolVersion pv;
     WOLFSSL_ENTER("wolfSSL_sk_CIPHER_description");
@@ -36380,7 +36380,7 @@ size_t wolfSSL_EC_get_builtin_curves(WOLFSSL_EC_BUILTIN_CURVE *r, size_t nitems)
     size_t ecc_sets_count;
     size_t i, min_nitems;
 
-    for (i = 0; ecc_sets[i].size != 0 && ecc_sets[i].name != NULL; i++);
+    for (i = 0; ecc_sets[i].size != 0; i++);
     ecc_sets_count = i;
 
     if (r == NULL || nitems == 0)
diff --git a/wolfcrypt/src/asn.c b/wolfcrypt/src/asn.c
index 079f337e4..e79be6ce2 100644
--- a/wolfcrypt/src/asn.c
+++ b/wolfcrypt/src/asn.c
@@ -5594,7 +5594,7 @@ int wc_OBJ_sn2nid(const char *sn)
     if (XSTRNCMP(sn, "secp384r1", 10) == 0)
         sn = "SECP384R1";
     /* find based on name and return NID */
-    for (i = 0; ecc_sets[i].size != 0 && ecc_sets[i].name != NULL; i++) {
+    for (i = 0; ecc_sets[i].size != 0; i++) {
         if (XSTRNCMP(sn, ecc_sets[i].name, ECC_MAXNAME) == 0) {
             eccEnum = ecc_sets[i].id;
             /* Convert enum value in ecc_curve_id to OpenSSL NID */

kaleb-himes, 2020-03-25:
@vaintroub,

Thank you so much for these suggested changes. You are an approved contributor to wolfSSL inc, would you mind opening a PR with these proposed changes?

Warm Regards,

K

ejohnstown, 2020-10-26:
Thank you for the bug report. This has now been fixed.

closed 2020-10-26 by ejohnstown